atar-axis/xpadneo #219: Problems with 8bitdo SN30 Pro on RetroPie with Bluetooth

ViperAcidZX, 2020-07-08:
I wanted to try and install this on my RetroPie setup with an 8bitdo SN30 Pro (not Pro+) controller because I wanted to be able to play PlayStation 1 games with vibration support for the games that supported the DualShock controller, however, I kept running into issues after pairing my 8bitdo SN30 Pro in XInput mode where I can't map my triggers at all in EmulationStation and sometimes the right analog stick's X-axis wouldn't work properly. I tried to use the `combined_z_axis` setting to try and get around but it wouldn't work at all and it kept telling me it couldn't write some kind of file within my Raspberry, except for the fact that it was actually there already after installing xpadneo and file doesn't want to be configured.

I am using a RaspberryPi Model 3B+ running RetroPie v4.6

**Steps to reproduce**
1. Install RetroPie v4.6 on RaspberryPi Model 3B+
2. Install xpad based on the instructions on the main page
3. Pair an 8bitdo SN30 Pro in XInput mode via Bluetooth settings
4. Reboot operating system
5. Attempt to try and configure the buttons and triggers on EmulationStation
6. Attempt to force single Z-axis setting with xpadneo's configuration tool

**Screenshot**
[image]

You can see that it's complaining about a file that's not in this location to write configurations except for the fact it was already there after installing xpadneo on RetroPie.

kakra, 2020-07-09:
Which version of xpadneo are you using?

ViperAcidZX, 2020-07-09:
I would assume it would be v0.7.5 going by the instructions on how to install xpadneo on RetroPie.

kakra, 2020-07-09:
Thanks, then I'll try to reproduce in Git-master and then backport to v0.7.5. The issue with the axis you're seeing is probably a known problem that you can work around with `jscal -u` and that won't be fixed by combining axes. There's an issue (#203) about it that will be closed soon, take a look. Also, I'm going to add an experimental final fix to this in v0.8 (released in a few days) when we start developing v0.9 (so this fix probably won't go into v0.8 stable unless proven widely-compatible).

If you are familiar with Git, you may want to try checking out Git-master. v0.7.5 is considered final, but your issue with it may be a candidate for another v0.7 version.

kakra, 2020-07-10:
Okay, I think I found the issue: The configure script actually find's two configuration files as candidate and cannot handle it. I created a fix for that. Let's test it so we can merge it.

ViperAcidZX, 2020-07-10:
Okay, I'll see if the workaround can work on my end in the meantime.

kakra, 2020-07-10:
> 1. I downloaded xpadneo using `git clone https://github.com/atar-axis/xpadneo.git -b master`

My fix is not merged to master yet, so you're still seeing these errors. I'll do that in a few hours.

For the wrong axes and `jscal`, please move discussion over to #203. There's a final fix for it in preparation and I'll update #220 when it's done.

closed 2020-07-11 by kakra