atar-axis/xpadneo #224: [Regression] disable_ff=3 no longer disables rumble

ipkpjersi, 2020-07-11:
## Version of xpadneo
0.8

## Severity / Impact
Moderate

* [x] I've read the docs and the bug reporting instructions
* [x] It does not work at all
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the bug
I can no longer disable force feedback/vibration/rumble. I have two scripts (one for enabling, one for disabling) and I have also modified /etc/modprobe.d/99-xpadneo-options.conf from disable_ff=0 to disable_ff=3 but nothing has worked. Before with xpadneo 0.6 when I ran my disable_ff.sh it would not rumble and when I ran enable_ff.sh it would rumble, but now regardless of whether I run disable_ff.sh or enable_ff.sh it always rumbles. Also, when I toggle any of the vibration options in RPCS3, it rumbles whereas before in xpadneo 0.6 it did not rumble:
[image]

## Steps to Reproduce
1. Install xpadneo 0.8
2. Try to disable force feedback

## Expected behavior
I expect force feedback to be disabled when disable_ff=3

## System information

```console
# uname -a
Linux desktop 4.15.0-109-generic #110-Ubuntu SMP Tue Jun 23 02:39:32 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0b 15 00 25 01 75 01 95 0b 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 05 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81  u........$...%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05  ...%.u..............
000000c8: 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07  ..#...%...u.....%.u.
000000dc: 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75  ........!........%.u
000000f0: 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25  .......%.u......p..%
00000104: 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75  du......Pf..U...&..u
00000118: 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00  .........&..u.....e.
0000012c: 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06  U..|..&..u..........
00000140: 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0                    . ..&..u......
3869637708 1558

```

## Controller and Bluetooth information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/4907405/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/4907416/xpadneo-dmesg.txt)

`Bus 003 Device 009: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)`

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/4907422/xpadneo-lsusb.txt)

## Additional context
disable_ff.sh:
```bash
#!/bin/bash
sudo rmmod hid_xpadneo
sudo chmod 777 /etc/modprobe.d/99-xpadneo-bluetooth.conf
sudo chmod 777 /etc/modprobe.d/xpadneo.conf
sudo echo "options hid_xpadneo disable_ff=3" > /etc/modprobe.d/99-xpadneo-bluetooth.conf
sudo echo "options hid_xpadneo disable_ff=3" > /etc/modprobe.d/xpadneo.conf
sudo chmod 777 /etc/modprobe.d/99-xpadneo-bluetooth.conf
sudo chmod 777 /etc/modprobe.d/xpadneo.conf
sudo modprobe hid_xpadneo
```

enable_ff.sh
```bash
#!/bin/bash
sudo rmmod hid_xpadneo
sudo chmod 777 /etc/modprobe.d/99-xpadneo-bluetooth.conf
sudo chmod 777 /etc/modprobe.d/xpadneo.conf
sudo echo "options hid_xpadneo disable_ff=0" > /etc/modprobe.d/99-xpadneo-bluetooth.conf
sudo echo "options hid_xpadneo disable_ff=0" > /etc/modprobe.d/xpadneo.conf
sudo chmod 777 /etc/modprobe.d/99-xpadneo-bluetooth.conf
sudo chmod 777 /etc/modprobe.d/xpadneo.conf
sudo modprobe hid_xpadneo
```

kakra, 2020-07-11:
Yes, the parameter has been partially replaced by `trigger_rumble_mode`. Do you intend to disable rumble completely? I'm going to implement new damping parameters where you could set all rumble to `0%`.

You could also try the new quirks parameter to mask away what makes the controller rumble. Try settings quirks mode to `1`, `4`, or `5`. You'll have to pass the parameter to the module like this:
```
options hid_xpadneo quirks=<MAC>:1
```

where `<MAC>` is the MAC address of your controller (as seen in `bluetoothctl`) and `1` is a sum of the quirk flags (see `modinfo hid_xpadneo`.

I'll conclude we need:

1. Display the MAC address of the controller in `dmesg` for easier building of quirk parameters
2. Add a rumble attenuation parameter (where you could set damping to 100%)
3. Add an extra attenuation parameter for the triggers (which additionally applies onto the general damping)

I may probably move this to a sysfs parameter so we can set those flags per connected controller as you may not want to apply this to every controller you own.

ipkpjersi, 2020-07-11:
Oh, my apologies. I didn't realized 0.7 or 0.8 had breaking changes with 0.6.

I'm not sure of the specific technical details of how disable_ff=3 worked, I just know that it worked perfectly for my requirements of not having any rumble when connecting or playing games.

kakra, 2020-07-11:
I just don't want to have multiple knobs for the same problem. ;-)

I didn't think of your use-case of disabling rumble completely - sorry for that. But since I'm planning to rework the damping mode, we can get that feature back easily through another parameter. I'm working on it and will back-port it to 0.8.1 (master is already on the feature plan to 0.9).

kakra, 2020-07-11:
Please try #225

ipkpjersi, 2020-07-11:
Didn't seem to work, it still rumbles at startup and in RPCS3:

`sudo ./configure.sh -r 100`
```
./configure.sh: Module inserted - writing to /sys/module/hid_xpadneo/parameters
./configure.sh: rumble_attenuation: parameter written to /etc/modprobe.d/99-xpadneo-options.conf
```

Also while I have you here, xpadneo doesn't seem to load the module hid_xpadneo at startup. I am using DKMS on Ubuntu 18.04, and it used to enable at boot/startup but now it doesn't seem to. Do you have any ideas I could quickly check or should I make a new issue for that?

ipkpjersi, 2020-07-11:
On close inspection, it didn't actually write the new parameter to `/etc/modprobe.d/99-xpadneo-options.conf`, I still had `disable_ff=3` which I replaced with `rumble_attenuation=100` and now it no longer rumbles.

I am still curious about the kernel module at system startup, though.

kakra, 2020-07-11:
Yes, a new issue. For that new issue, please disconnect the controller and unload the module. Then run `udevadm monitor` and connect the controller. It'll show a path starting with `/devices` and ends with your controller ID (`0005:...`). Please now run `udevadm test <paste-path-here>`. This should shed some light on this. Apparently, no one reacted to my offer to test this during the development cycle and it seemed to work for me.

For this issue: Yes, I know why that is. The configure script is a little dump when it comes to upgrading. We should maybe document to just delete the config file after upgrades, so it writes out the new defaults. It can only change parameters already existing in the file but cannot add new parameters.

Feel free to close this after you created the new issue.

ipkpjersi, 2020-07-11:
On second closer inspection, it does still rumble when loading the kernel module with `modprobe hid_xpadneo` and this is my configuration file  `/etc/modprobe.d/99-xpadneo-options.conf`:

`cat /etc/modprobe.d/99-xpadneo-options.conf`
`options hid_xpadneo rumble_attenuation=100 trigger_rumble_damping=1 trigger_rumble_mode=2 combined_z_axis=n`

kakra, 2020-07-11:
The rumble attenuation is only for rumble commands by applications. To disable the connect notification, use the new parameter `ff_connect_notify`.

kakra, 2020-07-11:
> Oh, and for the kernel module not loading at startup, it isn't loading for me with 0.6 either - I don't think it's a regression.

My instruction for `udevadm` should still work, please file a new bug report.

ipkpjersi, 2020-07-11:
Excellent, `ff_connect_notify=0` in `/etc/modprobe.d/99-xpadneo-options.conf` worked perfectly. 

I will now create the new bug report for the kernel module and close this one.

closed 2020-07-11 by ipkpjersi

kakra, 2020-07-11:
BTW: Maybe `udevadm control --reload` wasn't run for you to pick the new udev rules up...

ipkpjersi, 2020-07-11:
I might not need a new issue after all, it appears to not load hid_xpadneo until I connect my controller - my controller connected right away this time (I tried multiple times before but had no success, even with rebooting it would not connect) and then as soon as my controller connected, hid_xpadneo was loaded. 

I think I did see the udev rules loading when I first installed 0.6 and when I first installed your patch build.

If my kernel module issue persists I will open a new issue but otherwise it should be all good now. :+1:

kakra, 2020-07-11:
Yes, I converted the udev rules to work plug&play... You no longer need to manually load the module, the new rules just do the right thing without any dirty work-arounds.

ipkpjersi, 2020-07-11:
Excellent. I am looking forward to xpadneo 0.8.1 :)

kakra, 2020-07-11:
I don't want to release two versions on the same day... ;-) Let's leave this for a few days and see if some other bugs qualify for the 0.8 branch.