atar-axis/xpadneo #227: Failed to write ATTR with 8bitdo zero 2 gamepad

terencode, 2020-07-12:
## Version of xpadneo
v0.8.r515.g715c2e9

## Describe the bug

When connecting an 8bitdo zero 2 controller I see the following in the log: 
```
juil. 12 18:43:51 terence-desktop kernel: xpadneo 0005:045E:02E0.001E: controller quirks: 0x0007
juil. 12 18:43:51 terence-desktop kernel: xpadneo 0005:045E:02E0.001E: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [8BitDo Zero 2 gamepad] on d4:d2:52:a9:4f:d3
juil. 12 18:43:51 terence-desktop kernel: input: 8BitDo Zero 2 gamepad as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.001E/input/input59
juil. 12 18:43:51 terence-desktop kernel: xpadneo 0005:045E:02E0.001E: enabling compliance with Linux Gamepad Specification
juil. 12 18:43:51 terence-desktop kernel: xpadneo 0005:045E:02E0.001E: battery detected
juil. 12 18:43:51 terence-desktop kernel: xpadneo 0005:045E:02E0.001E: fixing up report size
juil. 12 18:43:51 terence-desktop systemd-udevd[2509859]: 0005:045E:02E0.001E: /etc/udev/rules.d/98-xpadneo.rules:1 Failed to write ATTR{/sys/devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.001E/[drivers/hid:xpadneo]bind}, ignoring: No such file or directory
```

## Steps to Reproduce

1. Connect an 8bitdo zero 2 controller over bluetooth in X Input Mode (Start + X)
2. Check journalctl

## Expected behavior

No error ^^

## System information

```console
# uname -a
Linux terence-desktop 5.7.8-18-tkg-bmq #1 TKG SMP PREEMPT Fri, 10 Jul 2020 18:00:53 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01  u.................%.
000000b4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
000000c8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
000000dc: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
000000f0: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
00000104: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
00000118: 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75  .u........... ..&..u
0000012c: 08 95 01 81 02 c0                                            ......
3445511648 1458

```

## Controller and Bluetooth information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/4908876/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/4908877/xpadneo-dmesg.txt)

## Image of the controller:

[image]

kakra, 2020-07-12:
Please disconnect the controller and unload the module, then run `udevadm monitor` and connect the device.

terencode, 2020-07-12:
```
KERNEL[66281.968125] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256 (bluetooth)
UDEV  [66281.970103] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256 (bluetooth)
KERNEL[66282.080383] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023 (hid)
KERNEL[66282.080516] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/power_supply/hid-e4:17:d8:aa:02:82-battery/wakeup53 (wakeup)
KERNEL[66282.080557] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/power_supply/hid-e4:17:d8:aa:02:82-battery/hwmon5 (hwmon)
KERNEL[66282.080632] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input64 (input)
KERNEL[66282.080669] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input64/event26 (input)
KERNEL[66282.080688] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input64/js0 (input)
KERNEL[66282.080705] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/hidraw/hidraw7 (hidraw)
KERNEL[66282.080732] bind     /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023 (hid)
KERNEL[66282.081391] remove   /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/power_supply/hid-e4:17:d8:aa:02:82-battery/hwmon5 (hwmon)
KERNEL[66282.081475] remove   /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/power_supply/hid-e4:17:d8:aa:02:82-battery/wakeup53 (wakeup)
KERNEL[66282.081512] remove   /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/power_supply/hid-e4:17:d8:aa:02:82-battery (power_supply)
KERNEL[66282.091071] remove   /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input64/event26 (input)
KERNEL[66282.098070] remove   /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input64/js0 (input)
KERNEL[66282.109096] remove   /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input64 (input)
KERNEL[66282.109122] remove   /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/hidraw/hidraw7 (hidraw)
KERNEL[66282.109139] unbind   /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023 (hid)
KERNEL[66282.110774] add      /module/hid_xpadneo (module)
KERNEL[66282.110953] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input65 (input)
KERNEL[66282.110970] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input65/event26 (input)
KERNEL[66282.111752] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input65/js0 (input)
KERNEL[66282.111770] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/hidraw/hidraw7 (hidraw)
UDEV  [66282.111824] add      /module/hid_xpadneo (module)
KERNEL[66282.771814] bind     /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023 (hid)
KERNEL[66282.771840] add      /bus/hid/drivers/xpadneo (drivers)
UDEV  [66282.771909] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023 (hid)
UDEV  [66282.772358] add      /bus/hid/drivers/xpadneo (drivers)
UDEV  [66282.772896] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/power_supply/hid-e4:17:d8:aa:02:82-battery/wakeup53 (wakeup)
UDEV  [66282.773849] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input64 (input)
UDEV  [66282.773877] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/power_supply/hid-e4:17:d8:aa:02:82-battery/hwmon5 (hwmon)
UDEV  [66282.775021] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input64/js0 (input)
UDEV  [66282.779446] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/hidraw/hidraw7 (hidraw)
UDEV  [66282.789585] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input64/event26 (input)
UDEV  [66282.790390] bind     /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023 (hid)
[... cut ...]
UDEV  [66282.795688] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input65/js0 (input)
UDEV  [66282.798485] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/hidraw/hidraw7 (hidraw)
UDEV  [66282.812173] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023/input/input65/event26 (input)
UDEV  [66282.813417] bind     /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023 (hid)
```

kakra, 2020-07-12:
Please run:
```
udevadm test /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023
```

terencode, 2020-07-12:
```
Load module index
Parsed configuration file /usr/lib/systemd/network/99-default.link
Parsed configuration file /usr/lib/systemd/network/80-iwd.link
Created link configuration context.
Reading rules file: /usr/lib/udev/rules.d/01-md-raid-creating.rules
Reading rules file: /usr/lib/udev/rules.d/10-dm.rules
Reading rules file: /usr/lib/udev/rules.d/11-dm-lvm.rules
Reading rules file: /usr/lib/udev/rules.d/13-dm-disk.rules
Reading rules file: /usr/lib/udev/rules.d/25-com-leapmotion-leap.rules
Reading rules file: /usr/lib/udev/rules.d/39-usbmuxd.rules
Reading rules file: /usr/lib/udev/rules.d/40-gphoto.rules
Reading rules file: /usr/lib/udev/rules.d/40-realtime-privileges.rules
Reading rules file: /usr/lib/udev/rules.d/40-usb-media-players.rules
Reading rules file: /usr/lib/udev/rules.d/49-sane.rules
Reading rules file: /usr/lib/udev/rules.d/50-udev-default.rules
Reading rules file: /etc/udev/rules.d/50-usb_power_save.rules
Reading rules file: /usr/lib/udev/rules.d/51-android.rules
Reading rules file: /usr/lib/udev/rules.d/55-Argyll.rules
Reading rules file: /usr/lib/udev/rules.d/56-hpmud.rules
Reading rules file: /usr/lib/udev/rules.d/60-autosuspend-chromiumos.rules
Reading rules file: /usr/lib/udev/rules.d/60-block.rules
Reading rules file: /usr/lib/udev/rules.d/60-cdrom_id.rules
Reading rules file: /usr/lib/udev/rules.d/60-drm.rules
Reading rules file: /usr/lib/udev/rules.d/60-evdev.rules
Reading rules file: /usr/lib/udev/rules.d/60-ffado.rules
Reading rules file: /usr/lib/udev/rules.d/60-fido-id.rules
Reading rules file: /usr/lib/udev/rules.d/60-input-id.rules
Reading rules file: /usr/lib/udev/rules.d/60-openrgb.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-alsa.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-input.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-storage-tape.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-storage.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-v4l.rules
Reading rules file: /usr/lib/udev/rules.d/60-rfkill.rules
Reading rules file: /etc/udev/rules.d/60-schedulers.rules
Reading rules file: /usr/lib/udev/rules.d/60-sensor.rules
Reading rules file: /usr/lib/udev/rules.d/60-serial.rules
Reading rules file: /usr/lib/udev/rules.d/60-tpm-udev.rules
Reading rules file: /usr/lib/udev/rules.d/60-vboxdrv.rules
Reading rules file: /etc/udev/rules.d/60-wacom.rules
Reading rules file: /usr/lib/udev/rules.d/61-autosuspend-manual.rules
Reading rules file: /usr/lib/udev/rules.d/61-gdm.rules
Reading rules file: /usr/lib/udev/rules.d/61-gnome-settings-daemon-rfkill.rules
Reading rules file: /usr/lib/udev/rules.d/63-md-raid-arrays.rules
Reading rules file: /usr/lib/udev/rules.d/64-btrfs-dm.rules
Reading rules file: /usr/lib/udev/rules.d/64-btrfs.rules
Reading rules file: /usr/lib/udev/rules.d/64-md-raid-assembly.rules
Reading rules file: /usr/lib/udev/rules.d/65-kvm.rules
Reading rules file: /usr/lib/udev/rules.d/65-libwacom.rules
Reading rules file: /usr/lib/udev/rules.d/69-cd-sensors.rules
Reading rules file: /usr/lib/udev/rules.d/69-dm-lvm-metad.rules
Reading rules file: /usr/lib/udev/rules.d/69-libftdi.rules
Reading rules file: /usr/lib/udev/rules.d/69-libmtp.rules
Reading rules file: /usr/lib/udev/rules.d/69-md-clustered-confirm-device.rules
Reading rules file: /usr/lib/udev/rules.d/69-xorg-vmmouse.rules
Reading rules file: /usr/lib/udev/rules.d/70-infrared.rules
Reading rules file: /usr/lib/udev/rules.d/70-joystick.rules
Reading rules file: /usr/lib/udev/rules.d/70-mouse.rules
Reading rules file: /usr/lib/udev/rules.d/70-power-switch.rules
Reading rules file: /usr/lib/udev/rules.d/70-printers.rules
Reading rules file: /usr/lib/udev/rules.d/70-spice-webdavd.rules
Reading rules file: /usr/lib/udev/rules.d/70-steam-input.rules
Reading rules file: /usr/lib/udev/rules.d/70-steam-vr.rules
Reading rules file: /usr/lib/udev/rules.d/70-touchpad.rules
Reading rules file: /usr/lib/udev/rules.d/70-uaccess.rules
Reading rules file: /usr/lib/udev/rules.d/71-seat.rules
Reading rules file: /usr/lib/udev/rules.d/72-joycond.rules
Reading rules file: /usr/lib/udev/rules.d/73-seat-late.rules
Reading rules file: /usr/lib/udev/rules.d/75-net-description.rules
[... 34 lines cut ...]
Reading rules file: /usr/lib/udev/rules.d/90-alsa-restore.rules
Reading rules file: /etc/udev/rules.d/90-android-tethering.rules
Reading rules file: /usr/lib/udev/rules.d/90-bolt.rules
Reading rules file: /usr/lib/udev/rules.d/90-fwupd-devices.rules
Reading rules file: /usr/lib/udev/rules.d/90-i2c-aura.rules
Reading rules file: /usr/lib/udev/rules.d/90-libinput-fuzz-override.rules
Reading rules file: /usr/lib/udev/rules.d/90-pulseaudio.rules
Reading rules file: /usr/lib/udev/rules.d/90-vconsole.rules
Reading rules file: /usr/lib/udev/rules.d/95-cd-devices.rules
Reading rules file: /usr/lib/udev/rules.d/95-dm-notify.rules
Reading rules file: /usr/lib/udev/rules.d/95-upower-csr.rules
Reading rules file: /usr/lib/udev/rules.d/95-upower-hid.rules
Reading rules file: /usr/lib/udev/rules.d/95-upower-hidpp.rules
Reading rules file: /usr/lib/udev/rules.d/95-upower-wup.rules
Reading rules file: /usr/lib/udev/rules.d/96-e2scrub.rules
Configuration file /etc/udev/rules.d/98-xpadneo.rules is marked executable. Please remove executable permission bits. Proceeding anyway.
Reading rules file: /etc/udev/rules.d/98-xpadneo.rules
Reading rules file: /usr/lib/udev/rules.d/99-fuse.rules
Reading rules file: /usr/lib/udev/rules.d/99-fuse3.rules
Reading rules file: /usr/lib/udev/rules.d/99-systemd.rules
Reading rules file: /usr/lib/udev/rules.d/wacom.rules
Failed to open device '/sys/devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0023': No such device
Unload module index
Unloaded link configuration context.
```

kakra, 2020-07-12:
You need to use the path from udevadm monitor which changes on every reconnect.

terencode, 2020-07-12:
Oups, sorry.
I think I got it (this is while the device is still plugged in) : 
```
Configuration file /etc/udev/rules.d/98-xpadneo.rules is marked executable. Please remove executable permission bits. Proceeding anyway.
Reading rules file: /etc/udev/rules.d/98-xpadneo.rules
Reading rules file: /usr/lib/udev/rules.d/99-fuse.rules
Reading rules file: /usr/lib/udev/rules.d/99-fuse3.rules
Reading rules file: /usr/lib/udev/rules.d/99-systemd.rules
Reading rules file: /usr/lib/udev/rules.d/wacom.rules
DEVPATH=/devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0008
DRIVER=xpadneo
HID_ID=0005:0000045E:000002E0
HID_NAME=8BitDo Zero 2 gamepad
HID_PHYS=d4:d2:52:a9:4f:d3
HID_UNIQ=e4:17:d8:aa:02:82
MODALIAS=hid:b0005g0001v0000045Ep000002E0
ACTION=add
SUBSYSTEM=hid
USEC_INITIALIZED=609050293
run: 'kmod load hid:b0005g0001v0000045Ep000002E0'
Unload module index
Unloaded link configuration context.
```

kakra, 2020-07-12:
So the udev rule is parsed correctly: It matches your device with the correct driver, so it managed to load the module and bind the driver. And there's no old udev rule lying around.

The question is what goes wrong during rebind. It actually unbinds and rebinds as can be seen in the monitor, and it only does that unless the driver is already "xpadneo". But I'm seeing a second line with `bind`. I'm not sure if that is supposed to be there. Whatever it is, `systemd-udevd` can resolve `[drivers/hid:xpadneo]/bind` since five years and should resolve it to `/sys/bus/hid/drivers/xpadneo/bind` - does that latter file exist for you? Or is your systemd version just very old?

terencode, 2020-07-12:
`/sys/bus/hid/drivers/xpadneo/bind` exists and I'm on systemd 245.6 (archlinux)

terencode, 2020-07-12:
Weird, I tried it again and this time no error.

terencode, 2020-07-12:
I rebooted in the meanwhile. I'm thinking maybe trying older version of xpadneo to troubleshoot the other issue messed something up and using the latest version and reboot fixed it?

kakra, 2020-07-12:
Then it's probably some artifact of swapping versions around, the udev rules need to be reloaded to work correctly, and when switching between certain versions, that may not always work. You could try if it still happens after a reboot, then we'd have a problem in the first-time execution of the rule.

kakra, 2020-07-12:
If it doesn't show up right after reboot and connecting the controller, I think we can close this?

terencode, 2020-07-12:
Ah! Managed to reproduce by connecting it right after a reboot:
```
juil. 12 20:04:51 terence-desktop kernel: Bluetooth: HIDP (Human Interface Emulation) ver 1.2
juil. 12 20:04:51 terence-desktop kernel: Bluetooth: HIDP socket layer initialized
juil. 12 20:04:51 terence-desktop kernel: hid-generic 0005:045E:02E0.0008: unknown main item tag 0x0
juil. 12 20:04:51 terence-desktop kernel: input: 8BitDo Zero 2 gamepad as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0008/input/input29
juil. 12 20:04:51 terence-desktop kernel: hid-generic 0005:045E:02E0.0008: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [8BitDo Zero 2 gamepad] on d4:d2:52:a9:4f:d3
juil. 12 20:04:51 terence-desktop systemd-udevd[3247]: 0005:045E:02E0.0008: /etc/udev/rules.d/98-xpadneo.rules:1 Failed to write ATTR{/sys/devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0008/[drivers/hid:xpadneo]bind}, ignoring: No such file or directory
juil. 12 20:04:51 terence-desktop kernel: xpadneo 0005:045E:02E0.0008: fixing up report size
juil. 12 20:04:51 terence-desktop kernel: xpadneo 0005:045E:02E0.0008: battery detected
juil. 12 20:04:51 terence-desktop kernel: xpadneo 0005:045E:02E0.0008: enabling compliance with Linux Gamepad Specification
juil. 12 20:04:51 terence-desktop kernel: input: 8BitDo Zero 2 gamepad as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0008/input/input30
juil. 12 20:04:51 terence-desktop kernel: xpadneo 0005:045E:02E0.0008: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [8BitDo Zero 2 gamepad] on d4:d2:52:a9:4f:d3
juil. 12 20:04:51 terence-desktop kernel: xpadneo 0005:045E:02E0.0008: controller quirks: 0x0007
```

kakra, 2020-07-12:
Okay, thanks. I'll look into that later. Seems to not be critical as it works despite that error.

terencode, 2020-07-12:
FYI same thing with the SN30 pro+

kakra, 2020-07-13:
I tested this here, I do not see such an error. So it's probably something with another module grabbing the device first on your system which doesn't happen on mine. I can try to check this again once I upgrade to 5.7 because I'll then try to drop my xpadneo patches from the kernel and see if Bluetooth works OOTB.

terencode, 2020-07-14:
Alright. 
I figured out that I don't need a reboot to trigger this. Just reload the module and it will happen the first time I connect a device.

Here is a list of currently running modules on my system in case it helps:
[modules.txt](https://github.com/atar-axis/xpadneo/files/4917964/modules.txt)

terencode, 2020-07-14:
I think I found it.
Somehow there was `/etc/udev/rules.d/98-xpadneo.rules` which maybe was a leftover from previous version? Removing it and no error now.

terencode, 2020-07-14:
Ah nope this is the file installed by dkms...

kakra, 2020-07-14:
This file is actually the current version, and it has this path that is mentioned in the error code. I don't now yet why it isn't resolved to the proper subsystem path as it should by `systemd-udevd`: It has code that resolves `[drivers/hid:xpadneo]` to the HID subsystem path in `/sys` - and we need to use that as it may be either `/sys/subsystem/...` or `/sys/bus/hid` (whatever exists first) and `systemd-udevd` auto-detects that, and then adds `drivers/xpadneo` to it. We also cannot use a relative path because at that point in time `xpadneo` is not yet the responsible driver, this is only after writing the device name to `bind`. I can only imagine that during module auto-load, we are a bit too early and `/sys/bus/hid/drivers/xpadneo/bind` does not exist yet but a few microseconds later.

kakra, 2021-01-31:
Do have multiple versions of xpadneo installed (e.g., through different kernel versions)?

Also, please post the contents of `/etc/udev/rules.d` (each file content).

terencode, 2021-01-31:
I'm using a vanilla archlinux kernel and a heavily customized one so dkms has two version installed.

### 60-schedulers.rules

```
# set scheduler for NVMe
ACTION=="add|change", KERNEL=="nvme[0-9]*", ATTR{queue/scheduler}="none"
# set scheduler for SSD and eMMC
ACTION=="add|change", KERNEL=="sd[a-z]|mmcblk[0-9]*", ATTR{queue/rotational}=="0", ATTR{queue/scheduler}="bfq"
# set scheduler for rotating disks
ACTION=="add|change", KERNEL=="sd[a-z]", ATTR{queue/rotational}=="1", ATTR{queue/scheduler}="bfq"
```

###  60-wacom.rules 

```
# udev rules for wacom tablets.

KERNEL!="event[0-9]*", GOTO="wacom_end"

# Multiple interface support for stylus and touch devices.
DRIVERS=="wacom", ATTRS{bInterfaceNumber}=="00", ENV{WACOM_TYPE}="stylus"
DRIVERS=="wacom", ATTRS{bInterfaceNumber}=="01", ENV{WACOM_TYPE}="touch"

# Convenience links for the common case of a single tablet.  We could do just this:
#ATTRS{idVendor}=="056a", SYMLINK+="input/wacom-$env{WACOM_TYPE}"
# but for legacy reasons, we keep the input/wacom link as the generic stylus device.
ATTRS{idVendor}=="056a", ENV{WACOM_TYPE}!="touch", SYMLINK+="input/wacom"
ATTRS{idVendor}=="056a", ENV{WACOM_TYPE}=="touch", SYMLINK+="input/wacom-touch"

# Check and repossess the device if a module other than the wacom one
# is already bound to it.
ATTRS{idVendor}=="056a", ACTION=="add", RUN+="check_driver wacom $devpath $env{ID_BUS}"

LABEL="wacom_end"
```

### 90-android-tethering.rules

```
# Execute pairing program when appropriate
ACTION=="add|remove", SUBSYSTEM=="net", ATTR{idVendor}=="18d1" ENV{ID_USB_DRIVER}=="rndis_host", SYMLINK+="android", RUN+="/usr/bin/systemctl restart systemd-networkd.service"
```

### 98-xpadneo.rules 

```
ACTION=="add", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*|0005:045E:0B05.*", SUBSYSTEM=="hid", DRIVER!="xpadneo", ATTR{driver/unbind}="%k", ATTR{[drivers/hid:xpadneo]bind}="%k"
ACTION=="add", DRIVERS=="xpadneo", SUBSYSTEM=="input", ENV{ID_INPUT_JOYSTICK}=="1", TAG+="uaccess", MODE="0664", ENV{LIBINPUT_IGNORE_DEVICE}="1"
```

kakra, 2021-01-31:
The `98-xpadneo.rules` file is old but you're probably not using the latest git version anyways?

https://github.com/atar-axis/xpadneo/blob/master/hid-xpadneo/etc-udev-rules.d/98-xpadneo.rules:
```
ACTION=="add", KERNEL=="0005:045E:02FD.*|0005:045E:02E0.*|0005:045E:0B05.*|0005:045E:0B13.*", SUBSYSTEM=="hid", DRIVER!="xpadneo", ATTR{driver/unbind}="%k", ATTR{[drivers/hid:xpadneo]bind}="%k"
ACTION=="add", DRIVERS=="xpadneo", SUBSYSTEM=="input", ENV{ID_INPUT_JOYSTICK}=="1", TAG+="uaccess", MODE="0664", ENV{LIBINPUT_IGNORE_DEVICE}="1"
```
It doesn't matter if you're not using the latest Xbox Series X|S model (that one with the share button). And there are no conflicting files, that's good already.

Please turn the controller off, then run `sudo udevadm monitor -p`, connect the controller and post the output. Maybe we can see  there what's going on.

terencode, 2021-01-31:
> The 98-xpadneo.rules file is old but you're probably not using the latest git version anyways?

I'm always using the latest from git so that's weird but I think found the issue:

Your dkms install hook targets /etc/udev/rules.d but the aur packager changed it to /usr/lib/udev/rules.d.
I'm guessing I still had one in /etc/ because I was trying to debug something and was installing directly from your repo

So my old comment (https://github.com/atar-axis/xpadneo/issues/227#issuecomment-658100948) was right.

I'm going to reboot and connect it again to be sure  but I think that was the issue.

terencode, 2021-01-31:
> Which version of systemd are you running?

systemd 247 (247.2-1-arch)

> does the error message go away when you run modprobe hid-xpadneo before connecting the controller?

No, only a warning about the udev config file being marked as executable.

> Please turn the controller off, then run sudo udevadm monitor -p, connect the controller and post the output. Maybe we can see there what's going on.

Connecting after a reboot  (error shown): [bad.txt](https://github.com/atar-axis/xpadneo/files/5901074/bad.txt)
If it helps here is after disconnecting and connecting again (no error shown): [good.txt](https://github.com/atar-axis/xpadneo/files/5901076/good.txt)

kakra, 2021-02-01:
> ~~Your dkms install hook targets /etc/udev/rules.d but the aur packager changed it to /usr/lib/udev/rules.d.~~

Yes, that's expected. Packages should install to `/usr/lib` - that's fine but it's up to you to handle it properly if you mix an installation from repository with an installation from a package maintainer. But it should not be executable - no idea why that is.

Connect after reboot: Yeah, the module is not loaded at that point, thus there's no driver directory yet. The second time, the module is already loaded and the directory mentioned in the udev rule exists. This is what I was suspecting.

The first time after reboot, there's an unbind action which removes the device from the standard hid driver, then there's a bind action to the xpadneo driver. So despite the error, it works. When trying a second time, there's two bind actions in the end - that's probably okay.

Did the maintainer (or you) install the modprobe.d file which contains the modaliases? It should be in `/etc/modprobe.d` or `/usr/lib/modprobe.d` (or `/lib/modprobe.d`, see the man page `man modprobe.d`). It should contain an alias `hid:b0005g*v0000045Ep000002E0` for your controller which should load the module just when the controller appears. Maybe there's a race condition and the module initializes a little later after udev has already been run, but still the change is picked up and udev finally binds the device correctly.
https://github.com/atar-axis/xpadneo/blob/master/hid-xpadneo/etc-modprobe.d/xpadneo.conf

terencode, 2021-02-01:
> Did the maintainer (or you) install the modprobe.d file which contains the modaliases? It should be in /etc/modprobe.d

Maintainer didn't touch this part so it's installed in `/etc/modprobe.d`. with the same content. It's also set as executable...

kakra, 2021-02-01:
Well, it's not `chmod +x` in the repository, so there's probably a bogus `chmod` in the package of your distribution.

terencode, 2021-02-01:
> Well, it's not `chmod +x` in the repository, so there's probably a bogus `chmod` in the package of your distribution.

In your `dkms.post_install` you don't pass the permissions to `install` so it uses `rwxr-xr-x` by default.

From the man page:
```
       -m, --mode=MODE
              set permission mode (as in chmod), instead of rwxr-xr-x
```

Is there anything else I can provide for the original issue or you have a good idea of the problem now?

kakra, 2021-02-01:
> In your `dkms.post_install` you don't pass the permissions to `install` so it uses `rwxr-xr-x` by default.

Oops, okay, I should fix that.

I think you should maybe just ignore that error message, seems be cosmetic as ultimately the module gets loaded. Leave this open, I'll see if I can work around that.

kakra, 2021-03-20:
I'm inclined to closing this as a minor and purely cosmetic issue. The permissions problems of the post install rules is fixed in my patch queue and will be backported to v0.9.

I'll give it another look but we can probably safely ignore the warning. Maybe also retry with latest master and see if it still exists.

terencode, 2021-03-25:
Was waiting for your latest code to check but still the same unfortunately :

```
0005:045E:02E0.000C: /usr/lib/udev/rules.d/60-xpadneo.rules:1 Failed to write ATTR{/sys/devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.000C/[drivers/hid:xpadneo]bind}, ignoring: No such file or directory
```

Yeah I guess the original issue didn't cause any problem except the red error message in the log.

kakra, 2021-03-25:
Well, could you reproduce with `udevadm monitor -p` running? It may give a clue what does wrong. I suspect that we need to split the rule or limit it to a specific udev action. But if that doesn't give a clue, I'm out of ideas. If you want to give it a try, please re-open with the requested log added.

terencode, 2021-03-25:
[udevadmn.txt](https://github.com/atar-axis/xpadneo/files/6206638/udevadmn.txt)

kakra, 2021-03-25:
The problematic trigger is probably here:
```
KERNEL[82.393202] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.000C (hid)
ACTION=add
DEVPATH=/devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.000C
SUBSYSTEM=hid
HID_ID=0005:0000045E:000002E0
HID_NAME=8BitDo SN30 Pro+
HID_PHYS=8c:55:4a:fd:3b:9d
HID_UNIQ=e4:17:d8:37:9d:61
MODALIAS=hid:b0005g0001v0000045Ep000002E0
SEQNUM=4498
```

It matches the udev rule which unbinds the driver - except that this block absolutely has no DRIVER line (which is obviously unequal to "xpadneo").

But we can find this block later:
```
KERNEL[82.393687] bind     /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.000C (hid)
ACTION=bind
DEVPATH=/devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.000C
SUBSYSTEM=hid
DRIVER=hid-generic
HID_ID=0005:0000045E:000002E0
HID_NAME=8BitDo SN30 Pro+
HID_PHYS=8c:55:4a:fd:3b:9d
HID_UNIQ=e4:17:d8:37:9d:61
MODALIAS=hid:b0005g0001v0000045Ep000002E0
SEQNUM=4505
```

So what happens if you change the first line in the udev rule `ACTION=="add"` into `ACTION=="bind"`? Does it work better then?

kakra, 2021-03-25:
BTW: The original intention of this rule was to match on kernels that actually add the device to a different driver than xpadneo, it seems modern kernels auto-rebind the driver as soon as an alternative driver appears, so it still works for you although the udev rule does not apply. I didn't think of a situation where `DRIVER` would actually be empty and thus yet undecided and unbound. But maybe it works with the bind action and everything will be fine.

kakra, 2021-03-25:
And then comes this block which actually does the unbind job for you:
```
UDEV  [83.117544] add      /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.000C (hid)
ACTION=add
DEVPATH=/devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.000C
SUBSYSTEM=hid
HID_ID=0005:0000045E:000002E0
HID_NAME=8BitDo SN30 Pro+
HID_PHYS=8c:55:4a:fd:3b:9d
HID_UNIQ=e4:17:d8:37:9d:61
MODALIAS=hid:b0005g0001v0000045Ep000002E0
SEQNUM=4498
USEC_INITIALIZED=82451598
DRIVER=hid-generic
```

So if `ACTION=="bind"` doesn't work, maybe `ACTION=="add", DRIVER!="", DRIVER!="xpadneo", ...` works.

terencode, 2021-03-26:
I don't have good knowledge of udev at all so I trust you :D 

Unfortunately none your suggestions worked but maybe the log with those variants can tell you more?

`udevadm monitor -p` for `ACTION=="bind"` :

[udevadm.txt](https://github.com/atar-axis/xpadneo/files/6208590/udevadm.txt)

and for `ACTION=="add", DRIVER!="", DRIVER!="xpadneo", ...` :

[udevadm.txt](https://github.com/atar-axis/xpadneo/files/6208592/udevadm.txt)

kakra, 2021-03-26:
If you edit the rules, you need to reboot or run `udevadm control -R` to reload the ruleset.

terencode, 2021-03-26:
The archwiki states :
> udev automatically detects changes to rules files, so changes take effect immediately without requiring udev to be restarted

So I thought I was good?

kakra, 2021-03-26:
It doesn't do that for me. It may be an Arch-specific hack where the systemd udev services watches the rules directory for changes - tho I think that's a bad idea. You could look at `systemctl cat systemd-udevd.service` to see if it does that.

Actually, that sentence doesn't make sense to me: That's a two-phase step: You need to reload the rules which doesn't apply anything on it's own. If you actually want to apply changes to devices already detected, you need to also run `udevadm trigger`.

`man udevadm` state nothing about automatic reloads but it mentions that even on reload, changes aren't applied to existing devices, which is why there's also trigger.

So the wiki may just be wrong, or refers the non-systemd udevd, or refers an Arch-specific hack.

Try this: Create a rule file with a syntax error and save it: Does udev log the error to the journal automatically? If not, does it do so when you use `udevadm control -R`?

kakra, 2021-11-17:
I think we should close this, it's purely a cosmetic error or may be caused by Arch-specific hacks (which even look like a bad idea if that's actually the case).

closed 2021-12-17 by no-response[bot]