atar-axis/xpadneo #228: 'Y' and 'R' button not detected by SDL applications

terencode, 2020-07-12:
## Version of xpadneo
v0.8.r515.g715c2e9

## Describe the bug
When connecting an 8bitdo controller (tested with both the zero 2 and SN30 pro+) in X Input Mode over bluetooth,
the keys labeled 'Y' and 'R' are not registered in sdl controllermap and antimicrox.

## Steps to Reproduce

1. Connect an 8bitdo zero 2 controller over bluetooth in X Input Mode (Start + X)
2. Run jstest/jstest-gtk and notice all buttons are correctly registered
3. Run antimicrox or sdl controllermap and notice the 'Y' and 'R' buttons do not do anything

## Expected behavior

All the buttons should be detected.

## System information

```console
# uname -a
Linux terence-desktop 5.7.8-18-tkg-bmq #1 TKG SMP PREEMPT Fri, 10 Jul 2020 18:00:53 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01  u.................%.
000000b4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
000000c8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
000000dc: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
000000f0: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
00000104: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
00000118: 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75  .u........... ..&..u
0000012c: 08 95 01 81 02 c0                                            ......
3445511648 1458

```

## Controller and Bluetooth information (for the zero2)

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/4908876/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/4908877/xpadneo-dmesg.txt)

kakra, 2020-07-12:
```
[63032.114975] xpadneo: hello there!
[63040.284204] xpadneo 0005:045E:02E0.001D: unknown main item tag 0x0
[63040.284343] input: 8BitDo Zero 2 gamepad as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.001D/input/input57
[63040.284451] xpadneo 0005:045E:02E0.001D: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [8BitDo Zero 2 gamepad] on d4:d2:52:a9:4f:d3
[63492.176904] xpadneo: goodbye!
```

The log seems strange. Did you mix different xpadneo versions? `hello there!` should no longer be present in current versions.

terencode, 2020-07-12:
Probably bad copy paste, I was trying to know if it's a regression but it doesn't same to be the case.
Let me know if you want me to redo something with latest driver but I can certify it's the same with the latest version.

kakra, 2020-07-12:
Did you manually make some SDL mappings in the past? E.g., look in `$HOME/.local/share/gamecontrollerdb.local.txt`. Also, do these buttons show up correctly in `evtest`?
`

terencode, 2020-07-12:
I actually did make one for the SN30 pro+ but not the zero 2 using the controllermap program.
Here are the mappings (this is also what shows up in env):
```
"050000005e040000e002000001000000,8BitDo SN30 Pro+,platform:Linux,a:b0,b:b1,x:b2,y:b3,back:b6,guide:b8,start:b7,leftstick:b9,rightstick:b10,leftshoulder:b4,rightshoulder:b5,dpup:h0.1,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,leftx:a0,lefty:a1,rightx:a3,righty:a4,lefttrigger:a2,righttrigger:a5,"
```
The buttons show up in evtest:
pressing 'y' button: 
```
Event: time 1594576476.189905, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1594576476.189905, type 1 (EV_KEY), code 307 (BTN_NORTH), value 1
Event: time 1594576476.189905, -------------- SYN_REPORT ------------
Event: time 1594576476.314848, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1594576476.314848, type 1 (EV_KEY), code 307 (BTN_NORTH), value 0
Event: time 1594576476.314848, -------------- SYN_REPORT ------------
```

pressing 'R' button:

```
Event: time 1594576499.643199, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90006
Event: time 1594576499.643199, type 1 (EV_KEY), code 311 (BTN_TR), value 1
Event: time 1594576499.643199, -------------- SYN_REPORT ------------
Event: time 1594576499.718180, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90006
Event: time 1594576499.718180, type 1 (EV_KEY), code 311 (BTN_TR), value 0
Event: time 1594576499.718180, -------------- SYN_REPORT ------------
```

terencode, 2020-07-12:
I unset the `SDL_GAMECONTROLLERCONFIG` and tried the programs again and same behaviour.

kakra, 2020-07-12:
Oh, funny... Your controller has X and Y swapped, and also A and B. Did you remap with the controllermap program for this reason? Should be fix this in the driver?

kakra, 2020-07-12:
```
"050000005e040000e002000001000000
```

This ID will also match your new controller and apply the same remapping to it. That may be the problem. During v0.8 we will add a remapping capability right into the driver in a way that it will detect the exact device. It will even work correctly with two identical models and still see them as different devices with different mappings.

terencode, 2020-07-12:
> This ID will also match your new controller and apply the same remapping to it

Hmm that's weird I tried it again and while the ID for both the SN30pro+ and the zero2 are the same, they are now different from what they were before (030000005e040000e002000000006800,Xbox One S Controller)

> Your controller has X and Y swapped, and also A and B. Did you remap with the controllermap program for this reason?

I think so. I guess now that you implemented quirks for these devices it makes sense to fix this within the driver?

> Do all 8Bitdo controllers use Nintendo mappings?

I would think so as they are first intended to be used on nintendo games.

kakra, 2020-07-12:
SDL matches by ID only. The name is just what is presented to the application, you can rename your controller there if you'd want to. The first part `0300` is the bus number. `0500` is for Bluetooth. You probably connected by USB now. Actually, the order of bytes is swapped: `0300` should be read as `0003`, `5e04` should be read as  `045e` (which is the PID for Microsoft).

If you confirm that all buttons work correctly (aside from being swapped) after you removed the mappings, I'll implement a quirk handler for that.

terencode, 2020-07-12:
I rebooted after removing SDL_GAMECONTROLLERCONFIG and confirmed it with the absence of anything SDL related within `env` but the 'R', 'Y', left and right joystick click (only with SN30 pro+) are still not working in controllermap and antimicrox.

kakra, 2020-07-12:
Please try if https://github.com/kakra/xpadneo/tree/queue/for-0.9 makes a difference.

For the rest, I need `btmon` logs: Capture the `a1 01` packets and tell me which button you pressed and what logs were generated for it. I'll have to go now, will look at it later.

terencode, 2020-07-12:
> Please try if https://github.com/kakra/xpadneo/tree/queue/for-0.9 makes a difference. 

No difference unfortunately, previously mentioned buttons not registered. However the 'X' button seems to correspond but the A and B buttons are still swapped.

> For the rest, I need btmon logs: Capture the a1 01 packets and tell me which button you pressed and what logs were generated for it. I'll have to go now, will look at it later.

I'm not sure how to do that. 
Here I used `sudo btmon`, pressed A, B , Y, X, L, R in this order (Y and R being the ones not working in SDL apps) and redirected the output to a file:
[btmon.txt](https://github.com/atar-axis/xpadneo/files/4909259/btmon.txt)

kakra, 2020-07-12:
Thanks, I've extracted the button data from the packets sent by the controller. It matches exactly with the quirks settings I've applied:
```
ID -- -- -- -- -- -- -- -- -- -- -- -- Buttons
01 00 80 00 80 00 80 00 80 00 00 00 00 00 02 00 - A 0000 0010 = 90002
01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00 - A 0000 0010
01 00 80 00 80 00 80 00 80 00 00 00 00 00 01 00 - B 0000 0001 = 90001
01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00 - B 0000 0001
01 00 80 00 80 00 80 00 80 00 00 00 00 00 04 00 - Y 0000 0100 = 90003
01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00 - Y 0000 0100
01 00 80 00 80 00 80 00 80 00 00 00 00 00 08 00 - X 0000 1000 = 90004
01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00 - X 0000 1000
01 00 80 00 80 00 80 00 80 00 00 00 00 00 10 00 - L 0001 0000 = 90005 (shoulder)
01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00 - L 0001 0000
01 00 80 00 80 00 80 00 80 00 00 00 00 00 20 00 - R 0010 0000 = 90006 (shoulder)
01 00 80 00 80 00 80 00 80 00 00 00 00 00 00 00 - R 0010 0000
```

The bits set and reset in the `btmon` packets match exactly with the usage map in the source code.

So what does it tell us? I think the problem is that we fixup the buttons of the reports from Linux to Windows mode but your controller already reports in Windows mode. I need to carry a flag to fix this. Hang on...

kakra, 2020-07-12:
Do you know the MAC addresses of both your controllers?

terencode, 2020-07-12:
SN30 pro+ is E4:17:D8:37:9D:61 and zero 2 is E4:17:D8:AA:02:82

kakra, 2020-07-12:
Please test #229.

kakra, 2020-07-12:
Please confirm that A, B, X, Y are in the correct order in `jstest`, and that `evtest` shows the correct directions east, south, north, west.

kakra, 2020-07-12:
Also, please show the dmesg output of this new driver version.

terencode, 2020-07-12:
I'm not sure how to make sure A, B, X, Y are in correct order in `jstest` but in `evtest` X and Y are swapped (Y says NORTH and X WEST).

In antimicrox, the same buttons still do not work and B and A are swapped.

dmesg output:
```
[11992.109220] xpadneo 0005:045E:02E0.000E: fixing up report size
[11992.109370] xpadneo 0005:045E:02E0.000E: battery detected
[11992.109371] xpadneo 0005:045E:02E0.000E: enabling compliance with Linux Gamepad Specification
[11992.109413] input: 8BitDo Zero 2 gamepad as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.000E/input/input40
[11992.109644] xpadneo 0005:045E:02E0.000E: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [8BitDo Zero 2 gamepad] on d4:d2:52:a9:4f:d3
[11992.109649] xpadneo 0005:045E:02E0.000E: controller quirks: 0x0027
[11992.770288] xpadneo 0005:045E:02E0.000E: 8BitDo Zero 2 gamepad [e4:17:d8:aa:02:82] connected
```

kakra, 2020-07-12:
`jstest` shows button 0, 1, 2, 3... This should match in order with A, B, X, Y.

terencode, 2020-07-12:
Ok so with jstest, A=1, B=0, X=3, Y=2

kakra, 2020-07-12:
Okay, this matches with `evtest` at least and all buttons respond now. Now let's see why it didn't swap A with B, and X with Y... Whatever this is, you should now see all buttons in antimicro and in SDL applications.

terencode, 2020-07-12:
All buttons were always responding in evtest and jstest, but not in antimicrox, `controllermap`, cemu (wine) and probably more.

kakra, 2020-07-12:
Could you send another `evtest` output and press all the buttons in order A, B, X, Y, L, R?

terencode, 2020-07-12:
```
Event: time 1594594606.536562, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1594594606.536562, type 1 (EV_KEY), code 305 (BTN_EAST), value 1
Event: time 1594594606.536562, -------------- SYN_REPORT ------------
Event: time 1594594606.687586, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1594594606.687586, type 1 (EV_KEY), code 305 (BTN_EAST), value 0
Event: time 1594594606.687586, -------------- SYN_REPORT ------------
Event: time 1594594608.790414, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1594594608.790414, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
Event: time 1594594608.790414, -------------- SYN_REPORT ------------
Event: time 1594594608.966446, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1594594608.966446, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
Event: time 1594594608.966446, -------------- SYN_REPORT ------------
Event: time 1594594612.267866, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90004
Event: time 1594594612.267866, type 1 (EV_KEY), code 308 (BTN_WEST), value 1
Event: time 1594594612.267866, -------------- SYN_REPORT ------------
Event: time 1594594612.376455, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90004
Event: time 1594594612.376455, type 1 (EV_KEY), code 308 (BTN_WEST), value 0
Event: time 1594594612.376455, -------------- SYN_REPORT ------------
Event: time 1594594612.755355, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1594594612.755355, type 1 (EV_KEY), code 307 (BTN_NORTH), value 1
Event: time 1594594612.755355, -------------- SYN_REPORT ------------
Event: time 1594594612.876366, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1594594612.876366, type 1 (EV_KEY), code 307 (BTN_NORTH), value 0
Event: time 1594594612.876366, -------------- SYN_REPORT ------------
Event: time 1594594613.881484, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90005
Event: time 1594594613.881484, type 1 (EV_KEY), code 310 (BTN_TL), value 1
Event: time 1594594613.881484, -------------- SYN_REPORT ------------
Event: time 1594594614.035520, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90005
Event: time 1594594614.035520, type 1 (EV_KEY), code 310 (BTN_TL), value 0
Event: time 1594594614.035520, -------------- SYN_REPORT ------------
Event: time 1594594614.342706, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90006
Event: time 1594594614.342706, type 1 (EV_KEY), code 311 (BTN_TR), value 1
Event: time 1594594614.342706, -------------- SYN_REPORT ------------
Event: time 1594594614.471567, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90006
Event: time 1594594614.471567, type 1 (EV_KEY), code 311 (BTN_TR), value 0
Event: time 1594594614.471567, -------------- SYN_REPORT ------------
```

kakra, 2020-07-12:
> I double checked and I have nothing related to SDL in `env`.

Please check that there is no file `$HOME/.local/share/gamecontrollerdb.local.txt` which is equivalent to having the SDL environment set. Also, please keep in mind that this file may be taken from `/root` if you are testing as root, and from another directory when running games under some user.

terencode, 2020-07-12:
I checked both $HOME/.local/share/ and /root/.local/share/ and no `gamecontrollerdb.local.txt` there.

kakra, 2020-07-12:
I may have found the problem, please test #229 again.

The Linux spec says A = south (mismatch), B = east (mismatch), X = north (match), Y = west (match), so A and B where still swapped.

terencode, 2020-07-12:
Weird :sweat_smile: unfortunately X and Y are still swapped with evtest.
With jstest: B=0, A=1, X=3, Y=2 (same as before).

kakra, 2020-07-12:
Could you show the output of `jscal -q`?

terencode, 2020-07-12:
`jscal -u 8,0,1,2,3,4,5,16,17,10,304,305,307,308,310,311,314,315,317,318 /dev/input/js0`

kakra, 2020-07-12:
BTW: `evtest` shows cardinal directions, don't rely on that. `evtest` should match the following: A = south, B = east, X = north, Y = west. Games are not using cardinal directions, they go for A, B, X, Y, so you need to correct that in your mind.

kakra, 2020-07-12:
So if you press A, B, X, Y in that order, `evtest` should show south, east, north, west in that order. Can you confirm?

terencode, 2020-07-12:
> BTW: evtest shows cardinal directions, don't rely on that

Damn that's really confusing..

> So if you press A, B, X, Y in that order, evtest should show south, east, north, west in that order. Can you confirm?
So, here, when I press A I get EAST; B is SOUTH; Y is NORTH, X is WEST.

kakra, 2020-07-12:
Yes... Okay, let's start different...
```bash
evtest | sed 's/SOUTH/A/;s/EAST/B/;s/NORTH/X/;s/WEST/Y/'
```

Should reduce headache...

terencode, 2020-07-12:
With that, A = B, B = 1, Y = X and X = Y.

kakra, 2020-07-12:
Probably B = A not 1 ;-)

terencode, 2020-07-12:
Yeah this is messing me up :laughing: I was actually editing it to be easier to parse for the eyes...

kakra, 2020-07-12:
I shared a lot more hours with this mess, don't ask me... ;-)

With the above sed included, please press A, B, X, Y in that order again and post the output.

terencode, 2020-07-12:
```
Event: time 1594596847.144665, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1594596847.144665, type 1 (EV_KEY), code 305 (BTN_B), value 1
Event: time 1594596847.144665, -------------- SYN_REPORT ------------
Event: time 1594596847.269687, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1594596847.269687, type 1 (EV_KEY), code 305 (BTN_B), value 0
Event: time 1594596847.269687, -------------- SYN_REPORT ------------
Event: time 1594596847.801173, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1594596847.801173, type 1 (EV_KEY), code 304 (BTN_A), value 1
Event: time 1594596847.801173, -------------- SYN_REPORT ------------
Event: time 1594596847.929770, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1594596847.929770, type 1 (EV_KEY), code 304 (BTN_A), value 0
Event: time 1594596847.929770, -------------- SYN_REPORT ------------
Event: time 1594596848.761296, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90004
Event: time 1594596848.761296, type 1 (EV_KEY), code 308 (BTN_Y), value 1
Event: time 1594596848.761296, -------------- SYN_REPORT ------------
Event: time 1594596848.915891, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90004
Event: time 1594596848.915891, type 1 (EV_KEY), code 308 (BTN_Y), value 0
Event: time 1594596848.915891, -------------- SYN_REPORT ------------
Event: time 1594596849.197351, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1594596849.197351, type 1 (EV_KEY), code 307 (BTN_X), value 1
Event: time 1594596849.197351, -------------- SYN_REPORT ------------
Event: time 1594596849.338520, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1594596849.338520, type 1 (EV_KEY), code 307 (BTN_X), value 0
Event: time 1594596849.338520, -------------- SYN_REPORT ------------
```

kakra, 2020-07-12:
A is still sending scan code 90002 (which is the original scan code from `btmon` but it should have matched a quirk mapping). I guess my quirk mapping isn't applied. Your quirk code is `0x27` which says "nintendo mapping"... Give me a few minutes.

kakra, 2020-07-12:
Please try #229 again, reconnect the controller and show the dmesg output...

terencode, 2020-07-12:
```
[15466.632922] hid-generic 0005:045E:02E0.0019: unknown main item tag 0x0
[15466.633028] input: 8BitDo Zero 2 gamepad as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0019/input/input54
[15466.633101] hid-generic 0005:045E:02E0.0019: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [8BitDo Zero 2 gamepad] on d4:d2:52:a9:4f:d3
[15466.667641] xpadneo 0005:045E:02E0.0019: fixing up report size
[15466.667761] xpadneo 0005:045E:02E0.0019: mapping hid 0x90001 quirks 0x0000 mask 0x0000 map 0x90001
[15466.667762] xpadneo 0005:045E:02E0.0019: mapping hid 0x90002 quirks 0x0000 mask 0x0000 map 0x90002
[15466.667763] xpadneo 0005:045E:02E0.0019: mapping hid 0x90003 quirks 0x0000 mask 0x0000 map 0x90003
[15466.667763] xpadneo 0005:045E:02E0.0019: mapping hid 0x90004 quirks 0x0000 mask 0x0000 map 0x90004
[15466.667764] xpadneo 0005:045E:02E0.0019: mapping hid 0x90005 quirks 0x0000 mask 0x0000 map 0x90005
[15466.667765] xpadneo 0005:045E:02E0.0019: mapping hid 0x90006 quirks 0x0000 mask 0x0000 map 0x90006
[15466.667765] xpadneo 0005:045E:02E0.0019: mapping hid 0x90007 quirks 0x0000 mask 0x0000 map 0x90007
[15466.667766] xpadneo 0005:045E:02E0.0019: mapping hid 0x90008 quirks 0x0000 mask 0x0000 map 0x90008
[15466.667766] xpadneo 0005:045E:02E0.0019: mapping hid 0x90009 quirks 0x0000 mask 0x0000 map 0x90009
[15466.667767] xpadneo 0005:045E:02E0.0019: mapping hid 0x9000a quirks 0x0000 mask 0x0000 map 0x9000a
[15466.667768] xpadneo 0005:045E:02E0.0019: mapping hid 0x10085 quirks 0x0000 mask 0x0000 map 0x10085
[15466.667768] xpadneo 0005:045E:02E0.0019: battery detected
[15466.667769] xpadneo 0005:045E:02E0.0019: enabling compliance with Linux Gamepad Specification
[15466.667798] input: 8BitDo Zero 2 gamepad as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0019/input/input55
[15466.667866] xpadneo 0005:045E:02E0.0019: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [8BitDo Zero 2 gamepad] on d4:d2:52:a9:4f:d3
[15466.667876] xpadneo 0005:045E:02E0.0019: controller quirks: 0x00000027
[15467.328285] xpadneo 0005:045E:02E0.0019: 8BitDo Zero 2 gamepad [e4:17:d8:aa:02:82] connected
```

kakra, 2020-07-12:
Okay, so we have two bugs here... One is: The nintendo quirk is applied too late, otherwise you should see `quirks 0x0020` in the debug log. The second is: The quirk matching is somehow ignored, it's not showing the expected result on my side, too.

kakra, 2020-07-12:
It's late... There was a brainbug in my head... ;-) Now the bugs in the debug output should be gone and the quirks flag is applied more early. Please pull #229 again and show dmesg.

terencode, 2020-07-12:
Let's see...

```
[16291.789356] hid-generic 0005:045E:02E0.001A: unknown main item tag 0x0
[16291.789456] input: 8BitDo Zero 2 gamepad as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.001A/input/input56
[16291.789540] hid-generic 0005:045E:02E0.001A: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [8BitDo Zero 2 gamepad] on d4:d2:52:a9:4f:d3
[16291.815039] xpadneo 0005:045E:02E0.001A: fixing up report size
[16291.815155] xpadneo 0005:045E:02E0.001A: mapping hid 0x90001 quirks 0x0000 mask 0x0000 map (1,0x130)
[16291.815157] xpadneo 0005:045E:02E0.001A: mapping hid 0x90002 quirks 0x0000 mask 0x0000 map (1,0x131)
[16291.815158] xpadneo 0005:045E:02E0.001A: mapping hid 0x90003 quirks 0x0000 mask 0x0000 map (1,0x133)
[16291.815159] xpadneo 0005:045E:02E0.001A: mapping hid 0x90004 quirks 0x0000 mask 0x0000 map (1,0x134)
[16291.815160] xpadneo 0005:045E:02E0.001A: mapping hid 0x90005 quirks 0x0000 mask 0x0000 map (1,0x136)
[16291.815160] xpadneo 0005:045E:02E0.001A: mapping hid 0x90006 quirks 0x0000 mask 0x0000 map (1,0x137)
[16291.815161] xpadneo 0005:045E:02E0.001A: mapping hid 0x90007 quirks 0x0000 mask 0x0000 map (1,0x13a)
[16291.815162] xpadneo 0005:045E:02E0.001A: mapping hid 0x90008 quirks 0x0000 mask 0x0000 map (1,0x13b)
[16291.815163] xpadneo 0005:045E:02E0.001A: mapping hid 0x90009 quirks 0x0000 mask 0x0000 map (1,0x13d)
[16291.815164] xpadneo 0005:045E:02E0.001A: mapping hid 0x9000a quirks 0x0000 mask 0x0000 map (1,0x13e)
[16291.815165] xpadneo 0005:045E:02E0.001A: mapping hid 0x10085 quirks 0x0000 mask 0x0000 map (1,0xac)
[16291.815166] xpadneo 0005:045E:02E0.001A: battery detected
[16291.815167] xpadneo 0005:045E:02E0.001A: enabling compliance with Linux Gamepad Specification
[16291.815204] input: 8BitDo Zero 2 gamepad as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.001A/input/input57
[16291.815371] xpadneo 0005:045E:02E0.001A: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [8BitDo Zero 2 gamepad] on d4:d2:52:a9:4f:d3
[16291.815373] xpadneo 0005:045E:02E0.001A: controller quirks: 0x00000027
[16292.475761] xpadneo 0005:045E:02E0.001A: 8BitDo Zero 2 gamepad [e4:17:d8:aa:02:82] connected
```

BTW: Let me know if you prefer to keep going later, it's not a problem ;)

terencode, 2020-07-13:
When running evtest, Y becomes BTN_C ??

kakra, 2020-07-13:
Okay, this looks better but clearly the early quirk flag is still missing for you... It should say 0x0020. I'll need to dig into the code a little. Meanwhile, try to load the module with `modprobe hid-xpadneo quirks=e4:17:d8:aa:02:82:0x27` and show dmesg (don't forget to rmmod first).

terencode, 2020-07-13:
```
[16668.085290] xpadneo 0005:045E:02E0.001B: fixing up report size
[16668.085391] xpadneo 0005:045E:02E0.001B: mapping hid 0x90001 quirks 0x0000 mask 0x0000 map (1,0x130)
[16668.085392] xpadneo 0005:045E:02E0.001B: mapping hid 0x90002 quirks 0x0000 mask 0x0000 map (1,0x131)
[16668.085393] xpadneo 0005:045E:02E0.001B: mapping hid 0x90003 quirks 0x0000 mask 0x0000 map (1,0x133)
[16668.085394] xpadneo 0005:045E:02E0.001B: mapping hid 0x90004 quirks 0x0000 mask 0x0000 map (1,0x134)
[16668.085395] xpadneo 0005:045E:02E0.001B: mapping hid 0x90005 quirks 0x0000 mask 0x0000 map (1,0x136)
[16668.085396] xpadneo 0005:045E:02E0.001B: mapping hid 0x90006 quirks 0x0000 mask 0x0000 map (1,0x137)
[16668.085397] xpadneo 0005:045E:02E0.001B: mapping hid 0x90007 quirks 0x0000 mask 0x0000 map (1,0x13a)
[16668.085398] xpadneo 0005:045E:02E0.001B: mapping hid 0x90008 quirks 0x0000 mask 0x0000 map (1,0x13b)
[16668.085399] xpadneo 0005:045E:02E0.001B: mapping hid 0x90009 quirks 0x0000 mask 0x0000 map (1,0x13d)
[16668.085400] xpadneo 0005:045E:02E0.001B: mapping hid 0x9000a quirks 0x0000 mask 0x0000 map (1,0x13e)
[16668.085401] xpadneo 0005:045E:02E0.001B: mapping hid 0x10085 quirks 0x0000 mask 0x0000 map (1,0xac)
[16668.085402] xpadneo 0005:045E:02E0.001B: battery detected
[16668.085403] xpadneo 0005:045E:02E0.001B: enabling compliance with Linux Gamepad Specification
[16668.085452] input: 8BitDo Zero 2 gamepad as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-2/1-2:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.001B/input/input59
[16668.085525] xpadneo 0005:045E:02E0.001B: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [8BitDo Zero 2 gamepad] on d4:d2:52:a9:4f:d3
[16668.085528] xpadneo 0005:045E:02E0.001B: quirks override: e4:17:d8:aa:02:82
[16668.085529] xpadneo 0005:045E:02E0.001B: controller quirks: 0x00000027
[16668.746394] xpadneo 0005:045E:02E0.001B: 8BitDo Zero 2 gamepad [e4:17:d8:aa:02:82] connected
```

kakra, 2020-07-13:
> When running evtest, Y becomes BTN_C ??

According to your dmesg output, there's no way to see `BTN_C` because that would be code `0x132` an that is not mapped anywhere... Strange.

terencode, 2020-07-13:
> According to your dmesg output, there's no way to see BTN_C because that would be code 0x132 and that is not mapped anywhere... Strange.

And I can't reproduce anymore...

terencode, 2020-07-13:
Ah, there you go: 
```
Event: time 1594598893.612479, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1594598893.612479, type 1 (EV_KEY), code 306 (BTN_C), value 1
Event: time 1594598893.612479, -------------- SYN_REPORT ------------
Event: time 1594598893.837524, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1594598893.837524, type 1 (EV_KEY), code 306 (BTN_C), value 0
Event: time 1594598893.837524, -------------- SYN_REPORT ------------
```

No idea how to reproduce it consistently.

kakra, 2020-07-13:
If this happens, xpadneo is not loaded... 90003 to BTN_C is the standard Linux mapping.

terencode, 2020-07-13:
You're right, oups... 
Going to bed now, can't delay anymore :D
Hopefuly we'll be done with this mess soon!

kakra, 2020-07-13:
Okay, I've come to the conclusion that we cannot map by quirk matching because the mapping is done before we can set our quirks. So I'll rewrite the code to swap the bits instead in the report. I don't like that idea too much as this messes with the physical positions games may expect but it should finally fix this.

kakra, 2020-07-13:
I think we got this, try #229 again.

terencode, 2020-07-13:
Running `evtest | sed 's/SOUTH/A/;s/EAST/B/;s/NORTH/X/;s/WEST/Y/'` gives me:

- Pressing A gives B
- Pressing B gives A
- Pressing X gives Y
- Pressing Y gives X

controllermap and antimicrox still does not detect Y, R left/right joystick click.

kakra, 2020-07-13:
Now I'm a bit lost because when I set quirks 0x27, I have my buttons swapped on the Xbox controller (which is has a different layout to yours), and when I set quirks 0x07, I have normal buttons. PS: Actually, my quirks are a little different, it's the "2" that makes the difference.

kakra, 2020-07-13:
That is, until it doesn't... I tested this by manually setting the quirks within the source code. Putting it on the cmdline doesn't work for some reason. :-(

kakra, 2020-07-13:
Okay, I think I found it. Your reports are 1 byte shorter than the original Xbox controller reports: only 16 bytes instead of 17, so it wasn't applying the button swap. Should work now, try #229 again and also post dmesg if it didn't work.

[33 more replies]

closed 2020-07-25 by kakra