atar-axis/xpadneo #233: Turning off the controller

Liorst4, 2020-07-18:
After you disconnect the bluetooth connection to the controller, the controller is left turned on, looking for the host.
This drains battery, there should be a way to turn the controller off. 

Possible ways to solve the issue:
* Turn the controller off when you disconnect from it.
* Add a sysfs file / an ioctl command to turn the controller off

kakra, 2020-07-18:
The Bluetooth protocol does not allow us to turn the controller off. The controller will turn off after around 30-60 seconds after being disconnected from the Bluetooth connection. Even if we could turn of the controller from the driver, how should that be possible if the connection is already gone?

You can press and hold the Xbox logo button to turn the controller off, and the driver can notice that (will be included in v0.9) and write a message to dmesg. But that's about all we can do, there's no HID report to send to the controller for turning it off.

Liorst4, 2020-07-18:
> Even if we could turn the controller off from the driver, how should that be possible if the connection is already gone?

1. Send the power off command
2. Abort the connection in the kernel

[From my understanding, the Xbox One console support this behavior](https://www.lifewire.com/thmb/ErrGme0wv7ENObUhDcZHhvsva74=/1920x1080/filters:no_upscale():max_bytes(150000):strip_icc()/vlcsnap-2018-12-02-15h48m04s891-5c0453aa46e0fb0001cf95bc.png) so there must be a command to turn off the controller via bluetooth

kakra, 2020-07-18:
1. There is no such command defined in the Bluetooth protocol of the controller. The available commands are listed in the descriptor docs of the docs folder. Also, there's no infrastructure in the kernel that could tell us that something wants to disconnect a Bluetooth device, it will simply go away, and only then the driver knows that the controller is gone.
2. It works the other way: The Bluetooth stack disconnects our device, then the kernel will tear down our driver. The driver has no knowledge about the disconnect until the device is already missing. So if anything could send a turn-off command before stopping the connection, it would be the Bluetooth stack. But there's simply no infrastructure for that by which drivers could register for a pre-disconnect hook. We cannot even instruct Bluetooth to initiate a disconnect from the kernel (which also makes no sense when the device is turning off anyways).

So whatever you want to accomplish: It's not possible with Bluetooth.

The Xbox One console does not use Bluetooth, it uses a proprietary protocol which supports this command.

Also, I don't see how 30-60s of the controller trying to recover a disconnected Bluetooth connection before it turns off would drain the battery. It's not like you do that the whole day repeating and repeating, you usually play 2-3 hours or even longer, and then put the controller away. I always let it turn off automatically after use: If the controller is idle for like 10 or 15 minutes, it will turn off. Trying to reconnect for 30-60s because you turned off your Bluetooth connection is nothing to worry about. Battery life is great for me with 40+ hours (using disposable batteries, rechargeable batteries are probably in the range of 30 hours).

Liorst4, 2020-07-18:
> Also, I don't see how 30-60s of the controller trying to recover a disconnected Bluetooth connection before it turns off would drain the battery.

Yeah, its not really a major issue, just something annoyed me a bit. 

> The Xbox One console does not use Bluetooth, it uses a proprietary protocol which supports this command.

damn, that sucks.

closed 2020-07-18 by Liorst4

kakra, 2020-07-18:
xow may support this because it uses the proprietary protocol.

Marukesu, 2020-07-19:
You can ever hold the Xbox button for 30s and the controller will shutdown.

kakra, 2020-07-19:
> You can ever hold the Xbox button for 30s and the controller will shutdown.

It doesn't even need 30s... It's more like 5-7.