atar-axis/xpadneo #234: takes long to detect disconnect

ramilmsh, 2020-07-29:
**Is your feature request related to a problem? Please describe.**
I am using xbox one wireless controller over bluetooth with ROS, to control a moving robot, however, if there is a connection loss (power/signal failure), it takes about 20-30 seconds for linux to detect that and all that time it is assumed that the last signal received is the desired command (so a robot collides with obstacles)

**Describe the solution you'd like**
I'd like to know if it would be possible to introduce some sort of polling for the connection (maybe with a rate parameter), to detect that.

**Describe alternatives you've considered**
I have tried setting the /sys/kernel/debug/bluetooth/hci0/supevisor_timeout, /sys/kernel/debug/bluetooth/hci0/conn_min_interval and /sys/kernel/debug/bluetooth/hci0/conn_max_interval to smaller values to no effect. I have also tried disabling ble, with the same result
 
**Additional context**
To be fair, it is a problem with all of the drivers I've tried, and I am not even sure if this is possible/reasonable

kakra, 2020-07-29:
The problem is that xpadneo cannot control the Bluetooth connection: There's simply no way to re-initialize the connection from the driver side, neither we can poll anything (it's a one way communication from Bluetooth to our driver), and we don't get notified about intermittent connection issues. When the connection breaks after timeout, the driver is torn down by Bluetooth, and then announces a new controller after reconnect.

The range is limited. I'm not sure if you've put the Bluetooth connection directly on the robot. It's probably better to have the Bluetooth dongle always nearby the controller, and use wifi to pass the receiver via usb-over-ip to the robot control board. That way you could even bypass Bluetooth completely and plug the controller into the usb-over-ip server directly, thus the control board could see the USB-attached Xbox controller directly in its usb-over-ip client.

Also, your design is wrong: The robot needs to handle connection losses. You need to watch the connection from there and it should stop moving if the connection appears to be broken. You could write a little daemon that watches the output of `btmon` to look for retransmissions and timeouts. If a loss is detected, or multiple losses in sequence, the robot should stop whatever it is doing after a short timeout.

ramilmsh, 2020-07-29:
@kakra thank you for your answer,  i will reconsider the design.

but it seems neither can i poll the bluetooth connection, since both btmon and js0 seem to think it's connected...

the problem is, it is impossible to distinguish between a person holding the stick in a certain position and the battery being dead or disconnected from looking at the data alone. 

I am gonna close this issue now, but thanks for the advise

closed 2020-07-29 by ramilmsh

kakra, 2020-07-29:
When the battery dies, you're going to see a message in dmesg from xpadneo. The current driver can detect that.

But I'd strongly suggest to go with a custom client-server-daemon to send data. It could detect from dmesg when the controller turns off and signal it to the robot. In all other cases, you should send keep-alive packets and the robots watches for these and acts accordingly. What you're asking for (directly maintaining status and stability with only the driver) is not possible with how the driver stack in Linux is designed, you *will* need a custom protocol implemented in user-space.

kakra, 2020-07-29:
If you create such a project on Github which attaches to xpadneo on the local side, implements a IP-bases protocol to communicate with the robot, and runs a daemon on the robot, I'd happily link that here in xpadneo for others to help and discover your project. I could also mentor you for some of the design decisions.

kakra, 2020-07-29:
I think for designing a protocol, you should start with what information you can get:

1. From the driver, you get input reports - none more or less. (producer side)
2. It can be possible to get a turn-off signal from the driver. (producer side)
3. We get no connection state, so you need to send keep-alive packets. (producer side)

So what can the protocol implementation in the robot expect now?

1. We will get input reports for controlling the robot that have been forwarded to your protocol implementation (producer-1)
2. We know when the controller turned off (producer-2)
3. We know when the protocol lost connection (producer-3, keep-alive)
4. We DO NOT know when the controller lost its Bluetooth connection

To work around 4, connect the controller via USB. Apparently, the controller doesn't send us keep-alive data over Bluetooth, so we have no early detection for a timeout. All we get is a battery report once in a while, but those could be minutes apart, so it's no usable. Also, we only get input data if the sticks are moved.

So it may be possible to pretend a disconnect if no more input data was received for some time. But that may not be practical: If you just hold the stick steadily into one direction, you would expect the robot to move but you won't get any new input reports. It may work if you can make some assumptions about how the robot is used, e.g. you could say: The sticks must be moved at most within 10 seconds again - then you could assume: If there's no input data within 10 seconds, the controller was disconnected.

kakra, 2020-07-29:
Also, I'd follow the OSI separation of layers: You robot should not directly attach to the protocol implemented by the driver because it needs additional information not provided by the driver itself. So you have to implement your own protocol layer on top of the driver layer which can then pull and poll additional information.