atar-axis/xpadneo #235: [Regression] Start button no longer works as of 0.8.2 in any games

ipkpjersi, 2020-08-02:
## Version of xpadneo
0.8.2

## Severity / Impact
Moderate

* [x] I've read the docs and the bug reporting instructions
* [x] It does not work at all
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the bug
I can no longer use the start button in any games. The button does not do anything anymore. It worked fine in 0.8.1 and of course 0.6 as well.

## Steps to Reproduce
1. Install xpadneo 0.8.2
2. Try to use the start button in any games

## Expected behavior
I expect force feedback to be disabled when disable_ff=3

## System information

```console
# uname -a
Linux desktop 4.15.0-109-generic #110-Ubuntu SMP Tue Jun 23 02:39:32 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0b 15 00 25 01 75 01 95 0b 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 05 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81  u........$...%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05  ...%.u..............
000000c8: 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07  ..#...%...u.....%.u.
000000dc: 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75  ........!........%.u
000000f0: 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25  .......%.u......p..%
00000104: 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75  du......Pf..U...&..u
00000118: 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00  .........&..u.....e.
0000012c: 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 85 04 05 06  U..|..&..u..........
00000140: 09 20 15 00 26 ff 00 75 08 95 01 81 02 c0                    . ..&..u......
3869637708 1558

```

## Controller and Bluetooth information
Honestly, most of this info is from before, if you need me to re-create it I will:

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/4907405/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/4907416/xpadneo-dmesg.txt)

`Bus 003 Device 009: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)`

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/4907422/xpadneo-lsusb.txt)

## Additional context

kakra, 2020-08-03:
Is this a copy of the "disable_ff" report?

Besides that: Did you remap buttons with jscal? Do you use custom SDL mappings? Do the thumb stick buttons still work correctly? This looks similar to a problem I was having with my controller after I upgraded the firmware. I then changed the driver to patch HID reports instead of remapping in the input layer, and it worked again.

kakra, 2020-08-03:
I suspect this may be a result of https://github.com/atar-axis/xpadneo/commit/0b4ff6972d81bb14ba98ef3b20175ca5e8769a06 - but that means you probably applied SDL mappings before

kakra, 2020-08-03:
BTW: It looks like your Bluetooth reports are one byte shorter than mine... I'm guessing you never upgraded the firmware of the controller?

ipkpjersi, 2020-08-03:
Hi,

Yes, this is a copy of the template of the disable_ff report. Hopefully the information from that post is still valid.

I did not remap buttons with jscal, I do not use custom SDL mappings as far as I am aware, I believe the thumbstick buttons still work properly - it seems like it's just the start button that doesn't work. It works in 0.8.1 and earlier, it does not work in 0.8.2

I am not sure if I upgraded the firmware of the controller - how would I check which version I have?

Thanks.

kakra, 2020-08-03:
> Yes, this is a copy of the template of the disable_ff report. Hopefully the information from that post is still valid.

If the descriptor is also from the older driver/post, this is pretty useless to me.

Please connect your controller, then run `btmon` and press the start button. Post the resulting log, it should hopefully only be a few lines long.

ipkpjersi, 2020-08-03:
Well it does appear to be more than a few lines long but hopefully this log is helpful: 
[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/5015766/xpadneo-btmon.txt)

Also, I noticed that the RB now functions as the start button, I've attached that log too - it seems this log is even bigger, not sure why:
[xpadneo-btmon-rb.txt](https://github.com/atar-axis/xpadneo/files/5015777/xpadneo-btmon-rb.txt)

kakra, 2020-08-03:
Without looking at the logs: Yes that is why I asked for the thumb buttons: I assume your left thumb is mapped to the right thumb, right thumb became mapped to start, and start is probably left thumb instead. So these buttons are shifted in sort of a triangle.

The other log is longer because you are not able to press the thumb button without moving the stick, so it creates additional logs.

Let's start different: You'll notice when clicking buttons, that the last bytes of the `a1 01` lines change, usually they show values like `01`, `02`, `04` or sums of that if you press multiple buttons at once. Could you work out which button produces which byte pattern - combinations do not matter, I only need single button events? Please note that the line is broken into two lines, so also look at the line just below `a1 01`. Then copy a line corresponding to a button and post the line with the name of the button you did press.

You can use `btmon | grep -A1 'a1 01'` to get an easier view.

ipkpjersi, 2020-08-03:
I have now created several more log files - each with the grep you requested.

For each of the log files, I spammed a specific button to get the pattern to repeat. I have also titled the log files to include which button was pressed - for example, btmon-spamming-rightstick-click would be me clicking the right stick button, btmon-spamming-rightstick-left would be me going left on the right stick, etc.

[btmon-spamming-rightstick-click.txt](https://github.com/atar-axis/xpadneo/files/5015867/btmon-spamming-rightstick-click.txt)
[btmon-spamming-leftstick-click.txt](https://github.com/atar-axis/xpadneo/files/5015868/btmon-spamming-leftstick-click.txt)
[btmon-spamming-rightstick-left.txt](https://github.com/atar-axis/xpadneo/files/5015869/btmon-spamming-rightstick-left.txt)
[btmon-spamming-leftstick-left.txt](https://github.com/atar-axis/xpadneo/files/5015870/btmon-spamming-leftstick-left.txt)
[btmon-spamming-rb.txt](https://github.com/atar-axis/xpadneo/files/5015871/btmon-spamming-rb.txt)
[btmon-spamming-start.txt](https://github.com/atar-axis/xpadneo/files/5015872/btmon-spamming-start.txt)

kakra, 2020-08-03:
Do you use Arch Linux and updated SDL lately?

ipkpjersi, 2020-08-03:
I am using Ubuntu 18.04 LTS with kernel 4.15.0-112-generic x86_64, I believe these are my SDL versions:

```
libsdl-image1.2/bionic-updates,bionic-security,now 1.2.12-8ubuntu0.1 amd64 [installed,automatic]
libsdl1.2-dev/bionic-updates,bionic-security,now 1.2.15+dfsg2-0.1ubuntu0.1 amd64 [installed]
libsdl1.2debian/bionic-updates,bionic-security,now 1.2.15+dfsg2-0.1ubuntu0.1 amd64 [installed,automatic]
libsdl2-2.0-0/bionic,now 2.0.10+dfsg1-3 amd64 [installed]
libsdl2-dev/bionic,now 2.0.10+dfsg1-3 amd64 [installed]
libsdl2-image-2.0-0/bionic,now 2.0.3+dfsg1-1 amd64 [installed]
libsdl2-mixer-2.0-0/bionic,now 2.0.2+dfsg1-2 amd64 [installed]
```

vishnugopal, 2020-08-10:
I have a similar issue as well, both my Start and Select buttons do not come up when I test for events, instead the shoulder buttons are mapped to it somehow:

```
➜  xpadneo-0.8.2 perl -0777 -l -ne 'print "/dev/input/$1\n" if /Name="Xbox Wireless Controller".*Handlers.*(event[0-9]+)/s' /proc/bus/input/devices | xargs evtest

Input driver version is 1.0.1
Input device ID: bus 0x5 vendor 0x45e product 0x2e0 version 0x903
Input device name: "Xbox Wireless Controller"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 172 (KEY_HOMEPAGE)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   1325
      Min   -32768
      Max    32767
      Fuzz      32
      Flat    3072
    Event code 1 (ABS_Y)
      Value    963
      Min   -32768
      Max    32767
      Fuzz      32
      Flat    3072
    Event code 2 (ABS_Z)
      Value      0
      Min        0
      Max     1023
      Fuzz       4
    Event code 3 (ABS_RX)
      Value  -1169
      Min   -32768
      Max    32767
      Fuzz      32
      Flat    3072
    Event code 4 (ABS_RY)
      Value   4161
      Min   -32768
      Max    32767
      Fuzz      32
      Flat    3072
    Event code 5 (ABS_RZ)
      Value      0
      Min        0
      Max     1023
      Fuzz       4
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
  Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
    Event code 81 (FF_PERIODIC)
[... 177 lines cut ...]
> ACL Data RX: Handle 71 flags 0x02 dlen 22                 #8 [hci0] 14.280708
      Channel: 65 len 18 [PSM 0 mode Basic (0x00)] {chan 65535}
        a1 01 2d 85 d4 81 6f 7b 41 90 00 00 00 00 00 00  ..-...o{A.......
        00 01                                            ..              
> ACL Data RX: Handle 71 flags 0x02 dlen 22                 #9 [hci0] 14.438304
      Channel: 65 len 18 [PSM 0 mode Basic (0x00)] {chan 65535}
        a1 01 2d 85 d4 81 6f 7b 41 90 00 00 00 00 00 00  ..-...o{A.......
        00 00                                            ..              
> ACL Data RX: Handle 71 flags 0x02 dlen 22                #10 [hci0] 14.445724
      Channel: 65 len 18 [PSM 0 mode Basic (0x00)] {chan 65535}
        a1 01 2d 85 d4 81 6f 7b 41 90 00 00 00 00 00 00  ..-...o{A.......
        00 00                                            ..              
> ACL Data RX: Handle 71 flags 0x02 dlen 22                #11 [hci0] 14.453192
      Channel: 65 len 18 [PSM 0 mode Basic (0x00)] {chan 65535}
        a1 01 2d 85 d4 81 6f 7b 41 90 00 00 00 00 00 00  ..-...o{A.......
        00 00                                            ..              
> ACL Data RX: Handle 71 flags 0x02 dlen 22                #12 [hci0] 15.353502
      Channel: 65 len 18 [PSM 0 mode Basic (0x00)] {chan 65535}
        a1 01 2d 85 d4 81 6f 7b 41 90 00 00 00 00 00 00  ..-...o{A.......
        00 01                                            ..              
> ACL Data RX: Handle 71 flags 0x02 dlen 22                #13 [hci0] 15.360908

```

I'll now try to install an older version to see if that fixes things!

vishnugopal, 2020-08-10:
Confirming that the controller works fine in 0.8.1.

Here's my 0.8.1 events if that helps:

```
➜  ~ perl -0777 -l -ne 'print "/dev/input/$1\n" if /Name="Xbox Wireless Controller".*Handlers.*(event[0-9]+)/s' /proc/bus/input/devices | xargs evtest

Input driver version is 1.0.1
Input device ID: bus 0x5 vendor 0x45e product 0x2e0 version 0x903
Input device name: "Xbox Wireless Controller"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 172 (KEY_HOMEPAGE)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    856
      Min   -32768
      Max    32767
      Fuzz     255
      Flat    4095
    Event code 1 (ABS_Y)
      Value   -904
      Min   -32768
      Max    32767
      Fuzz     255
      Flat    4095
    Event code 2 (ABS_Z)
      Value      0
      Min        0
      Max     1023
      Fuzz       3
      Flat      63
    Event code 3 (ABS_RX)
      Value  -1027
      Min   -32768
      Max    32767
      Fuzz     255
      Flat    4095
    Event code 4 (ABS_RY)
      Value   4078
      Min   -32768
      Max    32767
      Fuzz     255
      Flat    4095
    Event code 5 (ABS_RZ)
      Value      0
      Min        0
      Max     1023
      Fuzz       3
      Flat      63
    Event code 16 (ABS_HAT0X)
      Value      0
      Min       -1
      Max        1
    Event code 17 (ABS_HAT0Y)
      Value      0
      Min       -1
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
  Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
    Event code 81 (FF_PERIODIC)
    Event code 88 (FF_SQUARE)
    Event code 89 (FF_TRIANGLE)
    Event code 90 (FF_SINE)
    Event code 96 (FF_GAIN)
Properties:
Testing ... (interrupt to exit)
Event: time 1597056093.263418, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90008
Event: time 1597056093.263418, type 1 (EV_KEY), code 315 (BTN_START), value 1
Event: time 1597056093.263418, -------------- SYN_REPORT ------------
Event: time 1597056093.333610, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90008
Event: time 1597056093.333610, type 1 (EV_KEY), code 315 (BTN_START), value 0
Event: time 1597056093.333610, -------------- SYN_REPORT ------------
Event: time 1597056093.356469, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90007
Event: time 1597056093.356469, type 1 (EV_KEY), code 314 (BTN_SELECT), value 1
Event: time 1597056093.356469, -------------- SYN_REPORT ------------
Event: time 1597056093.506747, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90007
Event: time 1597056093.506747, type 1 (EV_KEY), code 314 (BTN_SELECT), value 0
Event: time 1597056093.506747, -------------- SYN_REPORT ------------

```

alexideavila, 2020-09-12:
Im having a similar issue. My controller used to work fine until now, maybe it was the update? I am using a normal wireless Xbox One Controller. Tried using the steam overlay and no luck. The games I tried were Skullgirls, Street Fighter IV and Dark Souls III. It seems my start, X and RB button are dead and some of them are remapped to other controls.

kakra, 2020-09-12:
> Confirming that the controller works fine in 0.8.1.

@vishnugopal Does the same test show different results in 0.8.2 for `evtest`?

If yes, could you run `git bisect start v0.8.2 v0.8.1`, the build the driver, do the test and run `git bisect good` or `git bisect bad` after each test to mark the commit good or bad? After a few loops it should tell you which commit is the bad one. Then maybe post the output of `git bisect log` and stop the bisect with `git bisect reset`.

Maybe you could repeat the same for git master: `git bisect start master v0.8` to see if it pins the same commit as bad.

Apparently, I don't see the issue here.

Also, we are only looking into `evtest` results here. If SDL shows mapping problems, that's a different known problem with latest SDL versions bypassing the driver in a strange way.

kakra, 2020-09-12:
**Update:** I actually CAN reproduce this now in v0.8.2 - this was different some weeks ago.
```
0b4ff6972d81bb14ba98ef3b20175ca5e8769a06 is the first bad commit
commit 0b4ff6972d81bb14ba98ef3b20175ca5e8769a06
Author: Kai Krakow <kai@kaishome.de>
Date:   Sun Jul 12 23:52:03 2020 +0200

    hid-xpadneo, quirks: We need to carry a quirk for Linux button mappings
    
    There are controllers which are not 100% compatible with Xbox One S
    because they are not switching to Linux mode when connected to Linux.
    We can detect such devices at HID parse time and set a flag when we
    detect Linux mode.
    
    With this commit, we no longer shift the button bits when we detect a
    controller in native Windows mode.
    
    Fixes: https://github.com/atar-axis/xpadneo/issues/228
    Signed-off-by: Kai Krakow <kai@kaishome.de>

 hid-xpadneo/src/hid-xpadneo.c | 9 +++++++--
 1 file changed, 7 insertions(+), 2 deletions(-)
```

@vishnugopal Can you confirm that result?

kakra, 2020-09-12:
Does the above commit fix the problem?

SamuelWAnderson45, 2020-09-18:
Not OP, but that last commit fixed button mapping / start button for me.

kakra, 2020-09-18:
Fixed in v0.8.3

closed 2020-09-18 by kakra

ipkpjersi, 2020-09-18:
Sorry for my late response, I can confirm that the 0.8.3 hotfix does fix this regression. Excellent work! :+1: