atar-axis/xpadneo #236: 8BitDo Layout

cakesale, 2020-08-05:
**Is your feature request related to a problem? Please describe.**
"Nintendo mode" shouldn't be the default configuration set for 8BitDo controllers.

**Describe the solution you'd like**
Leave controller layout as expected for X-input mode.

**Describe alternatives you've considered**
-

**Additional context**
The controller in X-input mode is to play on a PC and therefore they should respect the Xbox controller layout as all PC games use this layout/config. The default settings for xpadneo changes the layout to "Nintendo mode", this should be optional instead of the other way around as it currently is.

kakra, 2020-08-06:
I think we can revert that for the 0.8 branch but I'd keep this in the master branch because we will add profile switching to 0.9 which will allow for switchable profiles for button mappings.

cakesale, 2020-08-06:
I think it's a good option for those who want that profile but default should be set to Xbox controller layout.

As mentioned as well there was no mention of how to change layout back to normal. I am not sure if I did it the correct way but I removed the following from hid-xpadneo.c before building (manually went through all files before finding this):
	/* swap button A with B and X with Y for Nintendo style controllers */
	if ((xdata->quirks & XPADNEO_QUIRK_NINTENDO) && report->id == 1 && reportsize >= 15) {
		data[14] = SWAP_BITS(data[14], 0, 1);
		data[14] = SWAP_BITS(data[14], 2, 3);
	}

Sanaki, 2020-08-06:
Yeah, I'd like to know the -correct- way to disable this. I despise the Nintendo layout. It also breaks compatibility with anything that expects the standard layout, such as Retroarch.

kakra, 2020-08-06:
You cannot change the quirks at runtime, it is a load-time-only option so it must be passed as a modparam. It doesn't exist in `configure.sh` because I'm too lazy to adjust it - that script needs a complete rework, it duplicates too many options all over the code.

We should probably just not set the flag by default for 8BitDo controllers. This can be done somewhere in the beginning of the source in the OUI table: The correct way would be to remove the flag there.

PRs are welcome for both issues: The default flag and the configure.sh script.

But to be clear: The Xbox controller violates the Linux gamepad standard in the following way: Y should be `BTN_NORTH`, X west, A south, B east. The gamepad API expects buttons to be mapped by position. But this driver maps the buttons by symbol name, A is south (correct), B is east (correct), X is west (but it should be north), Y is north (but it should be west). So we are mapping by symbols, not by position. The quirk flag does just the same: It maps the buttons by symbol for 8BitDo controllers instead of by position. So what you're asking for is neither the one nor the other, it's hybrid. This is turn makes me believe that we SHOULD set the flag by default, so it matches with the default behavior of the driver mapping buttons by symbols instead of by position.

Sanaki, 2020-08-06:
> You cannot change the quirks at runtime (well, you could but it doesn't apply to currently connected controllers)
<p />Yeah, I forgot to mention that you need to power cycle the controller for the method I mentioned to take effect. Whoops.

For what it's worth, I agree not setting it by default would be preferable. For one thing, not every 8bitdo controller uses that layout. For example, the M30 is the six-button Genesis controller. That one maps ABXY in the standard xbox ABXY positions, with C mapped to RB, Z mapped to LB, R mapped to RT, and L mapped to LT.

cakesale, 2020-08-06:
Not sure if related but when I tried to set controller config in Steam
options today I noticed the triggers are set to be the same axis. Which is
weird... Luckily I've got config set and saved from before I installed
xpadneo. The only reason or need I have for this driver is to use rumble
with bluetooth on 8BitDo SN30 pro+ but considering just plugging in
controller when I want to use it considering all of this.

Sanaki, 2020-08-06:
Triggers are exposed in two ways. LT is both Axis 2 full range (-32767 - 32767) and Axis 8 negative range (0 - -32767), while RT is Axis 5 full range and Axis 8 positive range. Pretty sure this is intentional and done for compatibility purposes, since some applications can't handle full range axes for triggers and will only respond to half the pull.

kakra, 2020-08-07:
Triggers are exposed with a range of 0..1023, it's joydev which rescales this to -32767..32767. Axis 8 is a combined axis of -1023..1023 which simply adds both triggers which will also be rescaled to -32767..32767 by joydev. This is meant as a rudder or roll axis for simulators, it has no actual compatibility purpose, it's just there for simulators which cannot combine both triggers on their own. That axis will actually have its idle position at value 0, in contrast to each trigger on its own (which actually DO have their idle position at 0 until joydev rescaled it). Joydev is not really made for supporting triggers.

cakesale, 2020-08-07:
Maybe I'm the wrong type of user that you're expecting to be using this project but asking a general user to submit pull requests etc is a bit far-fetched. I'll probably just plug in a USB cable instead of using this as all that I am really using this for is to expose rumble while using bluetooth. Besides rumble over bluetooth the controller works perfectly, this project just seems to make unnecessary changes for 8BitDo controllers.

kakra, 2020-09-12:
@cakesale I didn't ask for submitting a pull request. I just said it would we welcome. It's not required that *you* do it, nor any one else. But there are people out there willing to help (because they can and want to), and those people may take this opportunity. That is, unless I took it first. But some options are low on my priority list - either because I don't use it or I don't have the hardware. So I'm asking for help. This is currently more or less a one mans work.

FWIW, these "unnecessary" changes have actually been asked by users. And most of this hasn't been hard-coded into the project, it can be configured. But I'm not willing to switch features back and forth just on every occasion. A decision was made some time ago to expose buttons by symbols, not by position. So this is what we've set as default for 8BitDo controllers. Otherwise, X and Y would be swapped on Xbox controllers (because Linux thinks Y should be west). OTOH, Linux also thinks A should be south which would mess up just the other pair of buttons on a Nintendo controller. So this decision hasn't been made up out of the blue, the reason was it should actually match what the button symbol says.

AFAIK, 8BitDo controllers support different modes. An Xinput mode that is supposed to match Xbox controllers, and another one which probably should match PS4 controllers. I wonder if it has a third mode. Maybe there would be an opportunity to support all modes in this driver, and than match the button positions depending on the mode.

Also, I'm hesitating to change too much now because I'm in the process of making switchable controller layouts possible. The Xbox Elite 2 controller has a handy hardware switch for that, current master already supports that, and even supports an emulation: Hold the guide button while clicking A, B, X, or Y, to switch between four profiles. It'll show up in dmesg. Currently it doesn't do much more. Issue reports like this make me think about plans to pre-populate such four profiles with sensible settings. So in the end, you could possibly have your cake and eat it too. Wouldn't this be fantastic? But this is all only due to "unnecessary" changing - otherwise probably nobody would've even thought of this.

Also, "unnecessary" is quite subjective: The button swap has actually been done to support Nintendo emulators better and have the button in the proper location. So it may seem unnecessary to you but that actually has been a purposed and founded change request by another user.

So, keep calm, contribute ideas. Thanks. :-)

But I got that you're probably the wrong person to ask for helping on figuring out how the driver could work with the other 8BitDo modes. OTOH, maybe you could suggest improvements on the documentation (requires no programming, you could even edit files directly in Github to suggest a PR). Or you could add ideas of how to make these settings more accessible: I agree that some aspects are not optimal currently.

I have maybe a few hours to dedicate to this driver in my weekends (and only some of them), nobody pays me a beer or pizza for this, and there are - including me - only a tiny few people testing the changes which - for my part - would be the games and functions I am interested in. Most "this or that does not work" reports come in for stable releases which is a bit suboptimal. I wish we had some more testing of the master branch before releasing a version or back-porting patches. So, for v0.9 I will be taking more time.

> Not sure if related but when I tried to set controller config in Steam options today I noticed the triggers are set to be the same axis. Which is weird...

Thanks for pointing this out. I didn't test the combined controller axis with the Steam controller options because I disabled that feature in Steam (it made my controller feel awkward, sluggish, or laggy when using it). I wasn't expecting configuration software picking that axis up because it's in the end of the list. But there's also no issue report for that currently.

orbea, 2021-01-13:
Removing the nintendo mappings is not enough. The positions of the `A`/`B` and `X`/`Y` buttons are switched as compared with a xpad gamepad.

> I wonder if it has a third mode.

The manual for the sn30 pro+ advertises four modes, switch, android (dinput), windows (xinput) and macos. I tested the dinput and xinput modes, but it seems xpadneo is only used for xinput.

Not all of the 8bitdo gamepads have the same modes.

closed 2021-03-02 by kakra