atar-axis/xpadneo #237: Guide Button not exposed on 8bitdo SN30 Pro+

Sanaki, 2020-08-06:
## Version of xpadneo

xpadneo-v0.8-26-g041e8d8 (041e8d8cc9a1ec3184a12fe057b8d7d568e72f23)

## Severity / Impact

* [X] I've read the docs and the bug reporting instructions
* [ ] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the bug

Similar to #162, the 8bitdo SN30 Pro+ (in xinput mode) doesn't expose the guide button when used via xpadneo. jstest-gtk confirms the guide button is simply missing. Since I haven't used it previously, I can't confirm if it worked in the past.

## Steps to Reproduce

1. Connect 8bitdo SN30 Pro+ via bluetooth
2. Attempt to bind or use the guide button in any application that makes use of it

## Expected behavior

The guide button should be exposed as a button, same as any xinput device.

## System information

```console
# uname -a
Linux NVC-CC 5.4.0-42-generic #46~18.04.1-Ubuntu SMP Fri Jul 10 07:21:24 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01  u.................%.
000000b4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
000000c8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
000000dc: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
000000f0: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
00000104: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
00000118: 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75  .u........... ..&..u
0000012c: 08 95 01 81 02 c0                                            ......
3445511648 1458
```

## Controller and Bluetooth information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/5034481/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/5034482/xpadneo-dmesg.txt)

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/5034484/xpadneo-lsusb.txt)

## Additional context

kakra, 2020-08-06:
According to xinput, the guide button is not part of the gamepad button map, it's exposed as a separate feature. If we expose this button as a gamepad button, this confuses Wine games. So we are currently sending this button as a keyboard event. Native Windows goes a similar route, xinput does not see the button. Later versions of proton have a special handler which removes an 11th button from the input mapping.

So not seeing this button in jstest is by design. It is currently mapped to `KEY_HOMEPAGE`, that way it adheres the xinput standard with 10 buttons, and still works to bring up the Steam client and Steam overlay. It cannot be used for game input, tho.

Sanaki, 2020-08-06:
Well, there's some info I never knew. I did verify it works fine with steam, so I'll be closing this issue. I'll see if we can get that handled properly in Retroarch instead, since that's where this is distinctly relevant to me. That said, it may be a desirable candidate for a quirk (though as with the Nintendo layout, it's more of a preference/usage matter than an actual trait of the controller).

kakra, 2020-08-07:
If you think there's a use case for having the button exposed, you should leave this open. I may consider it during remapping feature integration.

But it should be renamed: it's not specific to 8BitDo.

DaiKaiser93, 2020-08-09:
It also affects Retroach, I'm unable to map the menu to the guide button using xpadneo.

kakra, 2020-09-12:
I think we can safely put the guide button back as button 11 if Proton properly strips that button away in the Wine layer before exposing the mapping to games. I'll consider that after the v0.9 release. This needs some testing. You may want to evaluate `KEY_HOMEPAGE` in Retroarch anyways as long as the guide button serves this purpose: Bringing the user back to the OS / platform menu... Thus, the homepage key on a keyboard would bring you back to that menu which is probably what I would expect from such a key.

msdos, 2020-10-17:
> According to xinput, the guide button is not part of the gamepad button map, it's *exposed as a separate feature*. If we expose this button as a gamepad button, this confuses Wine games. So we are currently sending this button as a keyboard event.

> I'll see if we can get that handled properly in Retroarch instead, since that's where this is distinctly relevant to me. 

Don't know if this can be related to a problem I'm facing right now.

I've installed the driver in a RPi3B+, with retropie installed (a frontend to retroarch). Configured the input, and can play the games flawlessly.

The problem is... when playing a game, even after configuring Xbox button as a hotkey in the configuration section, it doesn't work _inside_ the game (I've searched the whole internet and just one person has this same problem https://forum.batocera.org/d/1990-xbox-one-controller). If I press the hotkey... nothing happens.

So, after seeing this issue, don't know if this could be somehow related. Are you familiar with retropie, retroarch and the hotkey concept?

kakra, 2020-10-17:
I know nothing about Retroarch and its components or hardware it is running on. But I could give this clue: The keyboard event that we send when you press the button is part of the same input device that represents the game input - and actually, that input device is registered in udev and libinput as a game input device, not as a keyboard. So for some applications that directly work with some APIs, that keyboard event may come unexpected and they simply discard it.

I'm planning to change that in one of the next versions as we are going to implement full support of the XBE2 controller (which has been donated to me by a cool community member of this project). This controller internally registers itself as a full keyboard and game input in two separate devices, and the guide button as a third device. We currently ignore that in xpadneo and combine it into one device and discard the keyboard part. But that made me think it may be better to expose the guide button as a separate device in xpadneo, too. Such a device would then be a keyboard or - in case of the guide button - a consumer control device. To applications the guide button would then appear to come from a different device class than the game input itself and it should work better. But well, I don't know how Retroarch works internally. It's just speculation that it may work around the problem.

So future versions of xpadneo will register a controller as three separate devices, with clearly distinctive types of input events (game input, keyboard, consumer control). This is needed because XBE2 can be programmed to generate other events for buttons than just their intended game input event, you can upload button mapping profiles to the device. As I don't know how to do that currently, a first implementation will do that as an emulation in the driver. The cool thing about this is: We're going to get this function for the XB1S controller, too. The current driver already implements a profile switcher which does nothing more than update an internal driver flag: If you switch the XBE2 controller with its profile switcher button, it will register that and print to dmesg. This is already emulated for XB1S by holding the guide button, then pressing A, B, X, or Y and releasing it again, and it will print to dmesg that the profile has been changed. Under the hoods, it will become a bit messy I already know. So this is probably going to be scheduled for v0.11, laying the groundwork in v0.10 which is supposed to cleanly split the driver into three modules.

orbea, 2021-01-13:
Following a suggestion in another issue I decided to try the sn30 pro+, the home button is seen with both `xev` and `ppsspp` so I think retroarch could be doing more here.

kakra, 2021-03-26:
Current master splits out keyboard events to a separate input device. It may be worth retesting it with Retroarch.

paul-ri, 2021-03-27:
Tested master today (58f234ab387f82456f30afa38cf5d0af3dc85553) with Retroarch and an Xbox one S controller and it behaves as @kakra described above.

Unfortunately, it breaks my use case which is likely similar to the original issue author:
* In Retroarch, I use the guide button as a hotkey to press in combination with other keys to enable emulator options. e.g. `guide+right shoulder` would load the game save state etc.

A few more details below that might influence your future decisions. These are specific to Retroarch, which doesn't look like they handle gamepads with advanced features like profiles or multiple inputs from one gamepad.

> With the profile switcher now in the driver, the guide button is only sent as an event when you release it. It actually synthesizes a button down and up event as two separate events then.

That prevents button combinations with the guide button, as highlighted above like `guide+R1`

> xpadneo will register a controller as three separate devices, with clearly distinctive types of input events (game input, keyboard, consumer control)

I see that and it breaks another use case in Retroarch: They allow assigning two _"hotkey"_ shortcuts, one from the keyboard and one from the gamepad.  
With the guide button being mapped as a keyboard event, that means I cannot use both the guide button and my keyboard key as a hotkey. The keyboard one helps stopping the game when something goes bad with the controller.

With that in mind, it'd be great if there was a way / quirk flag to keep the guide button as a button of the same controller. Only today did I learn about controller profiles for Xbox controllers; being able to disable that behavior would be helpful for me.

Many thanks for the detailed answers above!

kakra, 2021-03-27:
I'm currently working on an idea to add an Xbox360 emulation mode where the driver exposes the controller as a USB-connected original Xbox360 controller and no advanced features.

Thanks for pointing out that Retroarch uses a similar hotkey combination as I reserved for xpadneo by using the Guide button as a hotkey modifier. This is certainly a conflict with how Retroarch expects this to work. Apparently, this is the only button we can use for such features as it is usually a system key and cannot be controlled by the games itself. So there's probably no way around it other than disabling advanced features completely.

And yes, this will probably become a quirk flag so you could apply it to specific controller MACs only - in case you use multiple controllers and prefer specific ones for some games.

kakra, 2021-03-27:
In my patch queue, I've added a module parameter `xpad_emulation=1` which may fix this for you. To properly use that, use `git remote add kakra https://github.com/kakra/xpadneo && git remote update` and checkout my queue branch. Do not clone that to a separate repository because it will break versioning. You may add a worktree, tho, after adding the remote: `git worktree add ../xpadneo-kakra kakra/queue/for-0.10` should do the trick. But that's a rebase-branch, you cannot properly pull from it for future updates (use `pull --rebase` instead).

Apparently, it is not currently possible to add it as a quirk flag per controller due to dependencies between when the hardware initializes, when the quirk flags are parsed, and when the quirk flags are evaluated. I may fix that some time later, that's why I will keep this in the queue for a while.

paul-ri, 2021-03-28:
> I've added a module parameter xpad_emulation=1 which may fix this for you

Yep! I've just tried it on two machines and it works like a charm. Amazing! The guide button is reported as button no.8 on `jstest` and Retroarch is happy :+1: 

That controller has been difficult to use over the years, it's now perfect for me. Now I wish there was a way to disable the _"hold guide button for 6 seconds to turn off"_ and that' be ideal but I don't see that possible when briefly looking at the code.

I've also left a comment on your commit [here](https://github.com/kakra/xpadneo/commit/14c9f54a4d9b0c9cbbc2d6943daa09123ea9d822#r48808740).

good stuff, thanks!

kakra, 2021-03-28:
Yeah, it's not possible: The hold-to-turn-off is implemented in the hardware, we have no control over that. The last thing we get is a battery report saying goodbye.

kakra, 2021-03-28:
I moved this to a pull request so you can better follow the updates or easily extract a patch for local package installation:

* https://github.com/atar-axis/xpadneo/pull/283
* https://github.com/atar-axis/xpadneo/pull/283.patch

kakra, 2021-06-23:
Closing as queued in #283

closed 2021-06-23 by kakra