atar-axis/xpadneo #240: Controller isn't detected by bluetoothctl

PopeRigby, 2020-08-19:
## Version of xpadneo
0.8.2-1

## Severity / Impact

* [x] I've read the docs and the bug reporting instructions
* [x] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the bug
At first the controller refused to connect. I could `pair` and `trust` it when using `bluetoothctl`, but it always failed when I tried to `connect`. Now `bluetoothctl` won't even detect it.

## Steps to Reproduce
1. `sudo bluetoothctl`
2. `scan on`
3. Nothing happens

## Expected behavior
The controller is detected and connects.

## System information

```console
# uname -a
Linux snowy 5.8.1-arch1-1 #1 SMP PREEMPT Wed, 12 Aug 2020 18:50:43 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0

```

## Controller and Bluetooth information

I can't get this because my controller won't connect.

kakra, 2020-08-19:
Please run `bluetoothctl devices` to see if it is listed. If it is, remove it with `remove GAMEPAD-MAC`. Then try again. If it still doesn't work, stop your Bluetooth services, purge `/var/lib/bluetooth` clean, and reboot. Important: This clears the trust of all your Bluetooth devices, don't purge the directory if you're a using a BT mouse or keyboard with no alternative.

PopeRigby, 2020-08-19:
Ok. It detects it now, but fails to pair

```
Failed to pair: org.bluez.Error.ConnectionAttemptFailed
```

PopeRigby, 2020-08-24:
Yeah, I did.

kakra, 2020-08-24:
Maybe you could try following the steps from https://github.com/atar-axis/xpadneo/issues/239 to see if it makes you a successful connection?

PopeRigby, 2020-08-24:
I tried all those suggestions and it said it was connected, but the light kept blinking.

marcospampi, 2020-11-03:
> Please run `bluetoothctl devices` to see if it is listed. If it is, remove it with `remove GAMEPAD-MAC`. Then try again. If it still doesn't work, stop your Bluetooth services, purge [...]

I had the same issue of @PopeRigby , I updated xpadneo to 0.8 and this was the first time too I attemped this via cli,
following @kakra make the day

kakra, 2020-11-03:
Story time: I got a brand new controller, and it connected flawlessly to my Linux box. I then did some more experiments with it, connecting it to a Windows box, pairing it to another system, ... Now, that controller is no longer pairing automatically and I need to do the "send `bluetoothctl connect` at the right time" thing to get it to connect, and sometimes it even disconnects again after 30-60s. I'm quite convinced by now that there's something wrong in the controller itself. It worked flawless when fresh out of the box, and now that it's used for a while and has been paired to different systems, it shows the same problem as everyone is seeing. I suspect some internal state or buffer to be messed up. I'm not sure if it's possible to purge the internal memory back to factory fresh state - but if it is, I suspect it would pair and connect instantly again.

kakra, 2022-09-17:
Closing old Bluetooth issues, please report to the bluez project if the problem persists.

closed 2022-09-17 by kakra