atar-axis/xpadneo #246: Gamepad connects (light stays on) but not rumble and is not recognized by steam games (previously working).

gtsop, 2020-10-20:
## Version of xpadneo

xpadneo: v0.8-29-gd55e6d4!

## Severity / Impact

* [x] I've read the docs and the bug reporting instructions
* [x] It does not work at all
* [x] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the bug

Using the wireless xbox one controller, was successfully using xpadneo for about 2 months until a recent system upgrade.

The device seems to be connected (light stays turned on and the bluetooth panel shows the device is connected), however there is no rumble at the time of connection and the games don't actually seem to pick it up. Checking it in steam.

## Steps to Reproduce

I go through my usual connection process:
- Fire up bluetooth on the laptop
- Press the xbox button on the controller
- Wait for the light to stay on and feel the rumble.
- Light stays on but there is no rumble, device shows to be connected

## Expected behavior

I expect to feel the rumble when the device is connected and see the device picked up in steam and in game.

## Screenshots/Gifs

## System information

```console
# uname -a
OS: Linux gtsop 5.8.16-2-MANJARO #1 SMP PREEMPT Mon Oct 19 11:33:03 UTC 2020 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0
```

## Controller and Bluetooth information

```
Bluetooth monitor ver 5.55
= Note: Linux version 5.8.16-2-MANJARO (x86_64)                        0.793991
= Note: Bluetooth subsystem version 2.22                               0.793995
= New Index: A8:7E:EA:BE:37:EF (Primary,USB,hci0)               [hci0] 0.793996
= Open Index: A8:7E:EA:BE:37:EF                                 [hci0] 0.793998
= Index Info: A8:7E:EA:BE:37:EF (Intel Corp.)                   [hci0] 0.793999
@ MGMT Open: bluetoothd (privileged) version 1.17             {0x0001} 0.794001
> HCI Event: Connect Request (0x04) plen 10                  #1 [hci0] 7.246584
        Address: 9C:AA:1B:25:4C:D1 (Microsoft Corporation)
        Class: 0x000508
          Major class: Peripheral (mouse, joystick, keyboards)
          Minor class: 0x02
        Link type: ACL (0x01)
< HCI Command: Accept Connection Req.. (0x01|0x0009) plen 7  #2 [hci0] 7.246700
        Address: 9C:AA:1B:25:4C:D1 (Microsoft Corporation)
        Role: Slave (0x01)
> HCI Event: Command Status (0x0f) plen 4                    #3 [hci0] 7.249390
[... 578 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x003f  1x 63 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x003f  1x 63 bytes
        bInterval               1
```

## Additional context

I hope this is not a duplicate, the tickets I saw already report the game pad is not connected (light stays pulsing) or gets disconnected right after being connected.

gtsop, 2020-10-20:
Not exactly sure how the problem was caused, I resolved this problem by:
- Upgrading to kernel-5.9
- Uninstalling xpadneo (make sure to remove the dkms module)
- Installing xpadneo again

Now the controller works again as expected

closed 2020-10-20 by gtsop

kakra, 2020-10-20:
Yeah, there seem to have been some Bluetooth updates in the kernel for 5.9, and then some fixes were applied on top and only those fixes were backported to older kernels. Two of them handle a fix in establishing the encrypted channel, and that seems to be one of the problematic/weak areas of the Xbox controller.

So we may conclude that the last kernel updates to 5.4 and 5.8 may have broken Xbox Controller compatibility. It still works for me in 5.4, tho.