atar-axis/xpadneo #249: install script is path dependend

atar-axis, 2020-11-06:
The install script is depending on the path from which it is invoked. 

## Describe the bug
`cp: cannot stat 'hid-xpadneo': No such file or directory`

## Steps to Reproduce
1. clone repo
2. invoke `install.sh` from ../

atar-axis, 2020-11-06:
I'll fix that ASAP

kakra, 2020-11-07:
I'd say we fix this in master, v0.9 is going to land soon. Unless other bugs are found, I probably won't backport it to v0.8 because there's an easy work-around: Just don't run it from the parent directory.

The install/configure scripts itself use a lot of kludges like relying on a VERSION file in the repository, duplicating very similar code in multiple places, rewriting files inside the repository itself, etc. We should rather fix that. I already fixed a lot of that by outsourcing code to a library. But in the long term, I'd prefer a way to install through a makefile, with the master branch using no VERSION file at all.

atar-axis, 2020-11-07:
There was a Makefile once upon a time before I switched to dkms. But you are right, a Makefile would be a widely accepted way to go, let's do it this way instead.

kakra, 2020-11-07:
Yes...

There's still a makefile to build without dkms: `make -C hid-xpapneo reinstall`

kakra, 2020-11-07:
Or at least remove the version logic from the install script... It doesn't seem to belong there. The sed command that writes the version to the include file should go away, too. Instead, the build script itself should have a rule to create a `version.h` from scratch.

closed 2020-11-15 by kakra