atar-axis/xpadneo #256: trigger rumble mode setting not reflected in conf

k9bbzJavftZaqAqSW3dY, 2020-11-11:
Content of my /etc/modprobe.d/99-xpadneo-options.conf
`options hid_xpadneo disable_ff=0 disable_deadzones=0 rumble_attenuation=0 trigger_rumble_mode=0 combined_z_axis=n`

What it should be:
`options hid_xpadneo **trigger_rumble_mode**=0 disable_deadzones=0 rumble_attenuation=0 trigger_rumble_mode=0 combined_z_axis=n`

kakra, 2020-11-11:
Yeah, I'm currently too ~~lazy~~busy to patch that, and I don't like the code duplication in the script too much anyways. Feel free to submit a PR, or wait until I got some time to redo the configuration script. Thanks. 😊

ipkpjersi, 2021-03-20:
Uhh.. correct me if I'm mistaken @k9bbzJavftZaqAqSW3dY but shouldn't disable_ff just be removed outright? There's already `trigger_rumble_mode` in the two examples you posted so now you're just displaying it twice.

k9bbzJavftZaqAqSW3dY, 2021-03-22:
@ipkpjersi You're right there, it's a mistake I made posting this bug report, as I fixed it for myself already but thought I'd drop a line here, and I got the copy pasting wrong. I'm fixing it in the OP

closed 2021-03-24 by kakra