atar-axis/xpadneo #257: Can't Run Install Script

cow-killer, 2020-11-11:
## Version of xpadneo
The latest commit from kakra's fork: https://github.com/kakra/xpadneo

## Severity / Impact
* [X] I've read the docs and the bug reporting instructions
* [X] It does not work at all
* [X] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the bug
Was able to successfully clone kakra's repository with `git clone https://github.com/kakra/xpadneo.git`, but getting this when running the install script as sudo:
`sed: can't read ./lib/../VERSION: No such file or directory`

## Steps to Reproduce
* Clone the repository
* Run `install.sh` as root

## Expected behavior
Installs just fine with atar's repository, but not kakra's

## System information

```console
# uname -a
Linux pop-os 5.8.16-050816-lowlatency #202010170731 SMP PREEMPT Sat Oct 17 07:50:28 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

gort818, 2020-11-12:
It is not working since that repo doesn't have any git tags.

You can cd into the git repo and `git tag 0.9-beta`  or something similar.

kakra, 2020-11-12:
> It is not working since that repo doesn't have any git tags.

In this case, run `git fetch --tags`.

Also, try `sudo ./install.sh --verbose` and post the output.

kakra, 2020-11-12:
> also if you try to build the default branch you will get a compilation error `S16MAX` is undefined looks to be a typo probably should be `S16_MAX`

Whoops... Didn't show up in my test builds. Strange. I'll fix that.

kakra, 2020-11-12:
Instead of cloning from a different repo, you can add a remote:

1. Start from a git clone of atar-axis
2. Run `git add remote kakra https://github.com/kakra/xpadneo`
3. Update all remotes: `git remote update`
4. Checkout a worktree: `git worktree add ../kakra-xpadneo NAME-OF-A-BRANCH`

This is how I work with multiple repositories: I clone just the original, and then only add remotes per fork. That's why my repo doesn't have any tags.

cow-killer, 2020-11-12:
Getting this when running the install script with verbose:
```* verbose mode enabled
++ dirname ./install.sh
+ source ./lib/installer.sh
++ '[' 0 -ne 0 ']'
+++ git rev-parse --show-toplevel
++ GIT_ROOT=/home/cowkiller/Programs/xpadneo
+++ __version
+++ git describe --tags --dirty
++++ dirname ./lib/installer.sh
+++ sed '-res/v?(.*)/v\1/' ./lib/../VERSION
sed: can't read ./lib/../VERSION: No such file or directory
++ VERSION=```

> Run git add remote kakra https://github.com/kakra/xpadneo

Got this when trying to run that:
`fatal: pathspec 'remote' did not match any files`

kakra, 2020-11-12:
> > Run `git add remote kakra https://github.com/kakra/xpadneo`
> 
> Got this when trying to run that:
> `fatal: pathspec 'remote' did not match any files`

This happens if one is in a hurry...

It should be `git remote add ...`

kakra, 2020-11-12:
> +++ sed '-res/v?(.*)/v\1/' ./lib/../VERSION

Yeah, I think I need to fix that line or guard it better. I'll be looking into it.

cow-killer, 2020-11-12:
Alright, so good news is I was able to install after adding the remote repository. Bad news is, my Series X pad is still not connecting. At this rate I'm pretty certain my internal Bluetooth card doesn't support low energy. I ordered a low energy adapter yesterday, should be getting it in the mail tomorrow. I'll let you know the results then.

kakra, 2020-11-12:
Is this about the XBXS? If yes, we should probably put a note about BLE in the docs, and also compile a list of adapters that work.

kakra, 2020-11-12:
Nevermind, the info is here (just not in the top post):
> Bad news is, my Series X pad is still not connecting

So I'll be adding more info to the docs.

kakra, 2020-11-12:
> also if you try to build the default branch you will get a compilation error `S16MAX` is undefined looks to be a typo probably should be `S16_MAX` :)

This is from the branch `queue/wip-patches` - do not use this branch: Those commits are unfinished, may crash, may break several things, or are not working at all.

cow-killer, 2020-11-13:
So I just got [this](https://www.amazon.com/gp/product/B06XHY5VXF/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) adapter today. When trying to pair with the Series X pad, the computer says it's connected, but the Xbox button keeps glowing and `jstest` won't detect it. Secure Boot is disabled, tried both USB 2.0 and 3.0, same result.

I followed the instructions from [this](https://github.com/atar-axis/xpadneo/issues/257#issuecomment-726101442) comment, changed into the `kakra/xpadneo` directory, ran the install script from there, and restarted the machine.

ugly95, 2020-11-13:
> So I just got [this](https://www.amazon.com/gp/product/B06XHY5VXF/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) adapter today. When trying to pair with the Series X pad, the computer says it's [...]

I don't know if it helps, but I actually have that exact same adapter.  I haven't used that one in a while, but that particular Sabrent adapter did work for me with the XBox One S controller.  Like most other CSR-based USB adapters that I've tried, it does have difficulty making the initial connection, but works well once the connection is established.  I usually have to manually send the 'connect' command in `bluetoothctl` to get it to connect.

kakra, 2020-11-14:
This problem occurred due to checking out a repository without tags, more exactly: My [own fork](https://github.com/kakra/xpadneo) is not to be used outside of the context of https://github.com/atar-axis/xpadneo. But support for the XBXS controller has been merged now, so I'm going to close this.

closed 2020-11-14 by kakra