atar-axis/xpadneo #259: xbox series x controller connected but not working

vvdveen, 2020-11-22:
## Version of xpadneo
v0.9-1-g84cabe9

## Severity / Impact
Critical

* [x] I've read (most of the) the docs and the bug reporting instructions
* [x] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [x] I can code and I want to help

## Describe the bug
Connecting the new xbox series x controller fails. Pairing seems successful (the gamepad rumbles), but the X button on the controller continues to blink. Both dmesg and bluetoothctl claim the device is connected. The gamepad is also visible in jspad (there is a /dev/js0), but I am unable to get any interaction with the device.

Disconnecting and reconnecting brings the device in a reconnect loop.

I am on Ubuntu 20.04.1 (LTS). I tried with both my onboard Intel Dual Band Wireless-AC 3168 and a CSR8510 A10 Bluetooth dongle (same results). Today, I updated the controller's firmware to whatever is the latest version. The controller connects perfectly to my Android device.

## Steps to Reproduce
See xpadneo-bluetoothctl.txt below. Basically just following the README.

## Expected behavior
The device should connect (it seems it does?) and I should be able to use it.

## System information
```
# uname -a
Kernel version: Linux ryzen 5.4.0-54-generic #60-Ubuntu SMP Fri Nov 6 10:37:59 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0
# ls /sys/module/hid_xpadneo/drivers/hid:xpadneo/
bind  module  new_id  uevent  unbind
# maybe I missed something?
```

## Controller and Bluetooth information
- [xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/5580089/xpadneo-btmon.txt)
- [xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/5580090/xpadneo-dmesg.txt)
- [xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/5580091/xpadneo-lsusb.txt)
- [xpadneo-bluetoothctl.txt](https://github.com/atar-axis/xpadneo/files/5580092/xpadneo-bluetoothctl.txt)

## Additional context
Happy to help debugging this issue further!

dk89, 2020-11-22:
I'd like to attach to this issue because I have the same problem with my Series X Controller and my Raspberry PI 4 (Rasbian buster / RetroPie). Some time it says connected but the X keeps blinking and sometimes it's in a connection loop and keeps trying to connect. 

## System information

```console
# uname -a
Linux retropie 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

```

## Controller and Bluetooth information
[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/5580214/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/5580216/xpadneo-dmesg.txt)

vvdveen, 2020-11-22:
Good to see I am not the only one :)

A different machine with an Intel Wireless 7260 chipset gives me similar issues.

I did manage to get the report_descriptor:
```
➜  0005:045E:0B13.0003 xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363
```

kakra, 2020-11-22:
This is not an xpadneo issue, apparently. Secretly, I wish it would, then we could do something about it. But I found myself in a similar situation lately. But I was able to get it back to a working state by repeating the following steps until the link key appeared successfully in `/var/lib/HOST-MAC/DEVICE-MAC/info` using `bluetoothctl`:

1. Use `remove DEVICE-MAC`
2. Put the controller into pairing mode
3. Wait for it to announce itself, then use `pair DEVICE-MAC`
4. When it paired, immediately run `connect DEVICE-MAC`
5. Check if the link key was written to the info file, otherwise restart from step 1
6. Reboot

I needed to repeat the steps for 5-10 times until the controllers successfully connected again, and after reboot, it would even auto-connect again now without waiting 30-60s. It looks like repeating the steps often enough clears some internal broken state in the controller, or it depends on the timing you're doing the steps at.

Sometimes, it also works using the KDE GUI to pair the device by going through the "new device wizard" instead of using connect to select one from the devices shown. Whatever I did, it was important to remove the device first with `bluetoothctl`.

vvdveen, 2020-11-23:
Thanks! I tried your suggestion many times now, but no link key shows up.

Do you think that the the missing link key is also causing the connect/disconnect later? The output of btmon makes me think it might be related (missing PIN or Key missing, so encryption fails, so the device disconnects):
```
...
< HCI Command: LE Start Encryption (0x08|0x0019) plen 28    #54 [hci1] 4.818729
        Handle: 68
        Random number: 0x6001800500180660
        Encrypted diversifier: 0x001e
        Long term key: 80016006180005800160061800058001
> HCI Event: Command Status (0x0f) plen 4                   #55 [hci1] 4.824574
      LE Start Encryption (0x08|0x0019) ncmd 1
        Status: Success (0x00)
> HCI Event: Encryption Change (0x08) plen 4                #56 [hci1] 4.920600
        Status: PIN or Key Missing (0x06)
        Handle: 68
        Encryption: Disabled (0x00)
< HCI Command: Disconnect (0x01|0x0006) plen 3              #57 [hci1] 4.920637
        Handle: 68
        Reason: Authentication Failure (0x05)
...
```
What if we get a working link key from when the device is connected to Windows? Would that work? Also, what piece of software do you think is failing here? The bluetoothctl backend? The bluetooth driver? Something else?

kakra, 2020-11-23:
Yes, I think it's possible to get the link key from the Windows registry and put it into the Bluetooth info file - but I'm not sure if the format needs to be converted. But I think it's possible according to the very early development stages of this driver. Also, I do not remember where it is stored.

Did you turn off ETRM? Otherwise, it won't work at all because the device disconnects due to invalid configuration profiles requested (which will end in a similar cycle).

kakra, 2020-11-23:
Also, there have been firmware updates fixing Bluetooth pairing and connection stability for the older models, maybe check if your model already has an update available. Windows will install firmware updates to the controller only in USB or GIP-dongle mode, not in Bluetooth mode.

albertzdic, 2020-11-23:
> Connecting the new xbox series x controller fails. Pairing seems successful (the gamepad rumbles), but the X button on the controller continues to blink. Both dmesg and bluetoothctl claim the device is connected.

Same here. I have ERTM disabled.
[btmon.txt](https://github.com/atar-axis/xpadneo/files/5581570/btmon.txt)
[dmsg.txt](https://github.com/atar-axis/xpadneo/files/5581571/dmsg.txt)
[lsbus.txt](https://github.com/atar-axis/xpadneo/files/5581572/lsbus.txt)
[uname.txt](https://github.com/atar-axis/xpadneo/files/5581573/uname.txt)

kakra, 2020-11-23:
> Also, what piece of software do you think is failing here? The bluetoothctl backend? The bluetooth driver? Something else?

I'm not sure what exactly fails here. It's in the Bluetooth stack for sure - but if that's the kernel or the bluez daemon, I'm not sure. If we look at Android which uses the Linux kernel but a different Bluetooth stack (fluoride), and seeing that the controller pairs without problems there, it looks like the problem is in the bluez daemon. OTOH, the controller seems to behave differently based on the Bluetooth stack it sees, and after all seeing that the L2CAP patch works around one of the problems points to the kernel. It may be a bad combination of all three: The kernel, the bluez daemon, and the firmware.

If you manage to pair the controller by transferring the link key from Windows to Linux, it may be interesting to see if the controller presents us a different HID descriptor - and thus may have a "Windows mode".

Maybe we should check if the Linux kernel of Android uses some Bluetooth patches to support its fluoride daemon better. I haven't yet been able to compile fluoride in Linux.

kakra, 2020-11-23:
> Maybe we should check if the Linux kernel of Android uses some Bluetooth patches to support its fluoride daemon better. I haven't yet been able to compile fluoride in Linux.

Skimming through the commits to `net/bluetooth` and `drivers/bluetooth` in the Android kernel didn't reveal any Android-specific patches, so I think they are using the upstream kernel there. So the problem may be in Bluez. If someone manages to compile [fluoride](https://android.googlesource.com/platform/system/bt/) for a Linux distribution, it would be interesting to see if it fixes the connection problems, maybe even the rumble-associated disconnects.

albertzdic, 2020-11-24:
I tried to compile fluoride but without success, i get error using gn as a build system.

albertzdic, 2020-11-25:
Now my controller works.

kakra, 2020-11-25:
> Now my controller works.

What did you do?

albertzdic, 2020-11-25:
I upgraded the firmware of the controller with the windows app in a virtual machine. The upgrade was unsuccessful but now in linux bluetooth works.

kakra, 2020-11-25:
Yeah, there may a problem with updating in a virtual machine in that the controller reboots into a flash mode, and thus disconnects from the VM. You'd need to immediately re-connect it in that case for the firmware update to be successful.

albertzdic, 2020-11-25:
Only a little thing is changed:
in bluetooth the share button is mapped to select (the left button) and on usb select is mapped correctly.

albertzdic, 2020-11-25:
> Yeah, there may a problem with updating in a virtual machine in that the controller reboots into a flash mode, and thus disconnects from the VM. You'd need to immediately re-connect it in that case for the firmware update to be successful.

ooops, i hope every thing is ok

kakra, 2020-11-25:
> ooops, i hope everything is ok

Should be, if no firmware is being sent in that mode, nothing happens and the device will boot back normally after re-plugging it.

> when the controller is connected by bluetooth the share button is mapped to select (the left button)

Looks like MS likes to shuffle buttons around again on firmware update. We should inspect that and fix the button, and if we cannot properly detect one or the other firmware, I'd prefer going with what the newer firmware provides.

albertzdic, 2020-11-25:
also X and Y are inverted

RustyStriker, 2020-11-27:
Also had this problem, booted into my windows 10, updated the controller's firmware, fixed all issues... It seems the xbox sx require the update in order to function properly, as before the update it would connect and dc multiple times(not always rumbling).
Axises and buttons didnt change for me

japgolly, 2020-11-27:
I booted into Win 10 and updated the firmware too but unfortunately I still can't get mine to work in Linux :(

johnhoran, 2020-11-28:
Similiar issue, on first pairing I noticed 
```
(II) config/udev: Adding input device Xbox Wireless Controller (/dev/input/js0)
(II) No input driver specified, ignoring this device.
(II) This device may have been added with another device file.
(II) config/udev: Adding input device Xbox Wireless Controller (/dev/input/event25)
(**) Xbox Wireless Controller: Applying InputClass "libinput keyboard catchall"
(II) Using input driver 'libinput' for 'Xbox Wireless Controller'
(II) systemd-logind: got fd for /dev/input/event25 13:89 fd 64 paused 0
(**) Xbox Wireless Controller: always reports core events
(**) Option "Device" "/dev/input/event25"
(**) Option "_source" "server/udev"
(II) event25 - Xbox Wireless Controller: is tagged by udev as: Keyboard Joystick
(II) event25 - Xbox Wireless Controller: device is a keyboard
(II) event25 - Xbox Wireless Controller: device removed
(**) Option "config_info" "udev:/sys/devices/virtual/misc/uhid/0005:045E:0B13.0008/input/input29/event25"
(II) XINPUT: Adding extended input device "Xbox Wireless Controller" (type: KEYBOARD, id 18)
(II) event25 - Xbox Wireless Controller: is tagged by udev as: Keyboard Joystick
(II) event25 - Xbox Wireless Controller: device is a keyboard
```
in journalctl, so I tried adding 0B13 entries to udev and modprobe.  My controller now vibrates but the pairing light never stops flashing and it never appears in steam.

kakra, 2020-11-29:
> when the controller is connected by bluetooth the share button is mapped to select (the left button)

Should be fixed in my queued commits. I was seeing a similar problem for XBE2, and that resulted from an SDL2 upgrade that put the controller into its database.

keidax, 2020-11-29:
Having the same issue, I upgraded the firmware in Windows but that didn't make any difference.
After applying @kakra's changes to udev and modprobe, I see these two errors in `journalctl` output:
```
Nov 29 00:14:36 titan kernel: Bluetooth: hci0: link tx timeout
Nov 29 00:14:38 titan kernel: input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0009/input/input34
Nov 29 00:14:38 titan kernel: hid-generic 0005:045E:0B13.0009: input,hidraw6: BLUETOOTH HID v5.05 Gamepad [Xbox Wireless Controller] on 8c:c6:81:ce:51:d9
Nov 29 00:14:38 titan systemd-udevd[10333]: 0005:045E:0B13.0009: /etc/udev/rules.d/98-xpadneo.rules:3 Failed to write ATTR{/sys/devices/virtual/misc/uhid/0005:045E:0B13.0009/[drivers/hid:xpadneo]bind}, ignoring: No such file or directory
```

kakra, 2020-11-29:
I'd say you are in the reconnect loop then currently. The udev error is there because the device is already gone, our driver wasn't even loaded yet. Do you see the link key in the Bluetooth database?

vvdveen, 2020-11-29:
Found some time to look at this again. I extracted the key information from the windows registry and added a LinkKey section to the `/var/lib/bluetooth/<mac>/<mac>/info` file. Connecting with bluetoothctl didn't work (again a loop). I then removed the configuration files and tried again from bluetoothctl. This time it did work and it seems to be pretty stable also. It is weird, because I tried this many times before with no luck. The only difference now is that I paired the controller in Windows, which maybe removed some internal state in the controller? Or I got lucky...

There is no LinkKey section in the bluetooth info file.

kakra, 2020-11-30:
I'm suspecting that there is some internal state that gets cleared by luck sometimes, otherwise I cannot explain the seemingly random behaviors. The link key section shouldn't be missing, tho. You might by lucky to get it when trying to pair again.

ShadowNinja, 2020-11-30:
I can confirm that booting into Windows, pairing the controller, and then unpairing it while it's connected fixes the failure to pair.

Steps I followed:
  * Unpaired the controller in Linux.
  * Rebooted into Windows and paired the controller
  * Rebooted into Linux and tried to pair the controller unsuccessfully (controller buzzes, and it shows up as a controller in an SDL app (antimicrox), but the pair light on the controller doesn't stop blinking and inputs don't register in antimicrox).
  * Rebooted into Windows.  I noticed that Windows also seemed to also be stuck in a disconnect/reconnect loop this time.  I unpaired the controller, repaired it, and then unpaired it while it was still connected.
  * Rebooted into Linux.  This time I was able to pair the controller successfully (other than the back/share/guide and L/R stick click mappings being wrong, but that's a seperate issue).  I'm able to disconnect and reconnect successfully.

Shatur95, 2020-11-30:
I had the same problem. Controller worked only via USB.
I downloaded the latest version of Windows, updated the controller firmware via Xbox Acessories and paired. Then rebooted into Linux and now it works. But yes, with wrong button mappings. Via USB controller still have correct mappings.

kakra, 2020-11-30:
> But yes, with wrong button mappings. Via USB controller still have correct mappings.

I have a fix queued. I'll create a PR for it for you to test. I don't have this controller so I cannot test myself but it seems a similar issue like with the Elite 2 controller in SDL.

Shatur95, 2020-11-30:
> I'll create a PR for it for you to test.

Great, just ping me.

keidax, 2020-12-01:
I'm not exactly sure what the "link key" looks like. After the controller rumbles but the light is still blinking, I can see these sections in `/var/lib/HOST-MAC/DEVICE-MAC/info`:
```
[General]
...
[IdentityResolvingKey]
...
[RemoteSignatureKey]
...
[LocalSignatureKey]
...
[LongTermKey]
...
[SlaveLongTermKey]
...
[DeviceID]
...
```
Does that look correct, or should there be something else called link?

I tried pairing and unpairing in Windows several times, which didn't seem to make a difference. But I'm using Windows on a separate laptop, not in a dual-boot situation. Maybe it only works once already paired with the same Bluetooth controller?

kakra, 2020-12-01:
The sections are different for BLE... Bluetooth up to Xbox Elite 2 have only one key section.

dk89, 2020-12-01:
My controller works now. First of all I deactivated my internal Bluetoothcontroller an bought an external USB Bluetooth Controller, because the internal had a pretty bad range. 

After that I got the same problems like before. But now I connected the controller with my Windows-PC and copied the BLE-paring-keys from the registry to the infofile in Linux. After restarting the Bluetooth service the Controller immediately connected to my Linux.

kakra, 2020-12-01:
So one of the failure points seems to be the controller and the Linux Bluetooth stack negotiating the keys properly. It may work in maybe 1 of 20 tries properly but the others just fail. Copying the keys from Windows seems to always work.

keidax, 2020-12-02:
After [following these instructions](https://bbs.archlinux.org/viewtopic.php?pid=1678405#p1678405), I was able to connect to the controller in Linux.

kolomansell, 2020-12-04:
Don't know if it helps somehow but for me worked following:

kakra, 2020-12-05:
Yeah, it often looks like pairing the controller to any other host which successfully pairs the controller (Android, Windows) usually mostly fixes a following pairing in Linux. I think the Linux Bluetooth stack is incomplete from the view of the Xbox controller firmware and fails to properly initialize. Pairing to a different system may clear some invalid states. I wish it would be possible to somehow firmware-reset the controller, so it starts fresh. But I know of no such reset method.

L0gi, 2020-12-05:
> After [following these instructions](https://bbs.archlinux.org/viewtopic.php?pid=1678405#p1678405), I was able to connect to the controller in Linux.

can confirm that copying keys from windows worked for me as well.  At least as far as fixing the pairing and connecting issue (" PIN or Key Missing" Connect/Reconnect loop).

kolomansell, 2020-12-05:
> can confirm that copying keys from windows worked for me as well. At least as far as fixing the pairing and connecting issue (" PIN or Key Missing" Connect/Reconnect loop).

How is the controller working for you ? Didn't look much into the issue but using mine via bluetooth has a noticable lag. USB connection is flawless.

L0gi, 2020-12-06:
> How is the controller working for you ? Didn't look much into the issue but using mine via bluetooth has a noticable lag. USB connection is flawless.

I'm running a osmc rpi3b+ setup trying to get things to work with steamlink (Xbox Wireless controller model 1914 - firmware 5.5.2641.0). 
Controlling the Kodi UI in osmc seems to be ok. But I'm running into problems with steamlink so couldn't test it in games yet. 

The issue with steamlink seems to be that it's for whatever reason doubling over the left analog stick to serve as a super sensitive mouse input. Switching "controller - mouse" (or whatever it's called) in the steamlink overlay doesn't seem to do anything. 
But this appears to be a steamlink/osmc issue not related to xpadneo as steamlink does appear to discover the controller as an XInputDevice even with xpadneo removed. Interestingly it doesn't recognize the controller when it's not paired via bluetooth but plugged in through usb directly...

rubencabrera, 2020-12-10:
I didn't need to copy the keys from Windows, after it connected on a W10 virtual machine, I could connect to a the host with a 1914 model controller. It's the second gamepad I connect and now none of them shows up for the games I've tested (supertux and a game in steam). Guess that's a different issue, as the connected/disconnected loop that I was experiencing is gone.

kakra, 2020-12-11:
@rubencabrera Does it show up in `jstest` and `evtest`? If yes, then that's a different issue.

kakra, 2020-12-11:
So it seems BT is okay. Then maybe the driver is not bound properly. Run `dmesg -w` in a second terminal and connect the device, watch what happens in the second terminal.

rubencabrera, 2020-12-11:
Doing it just for one controller, the model 1708, which used to work fine until the 1914 connected.

```
[<  549,396284>] hid-generic 0005:045E:02E0.001C: unknown main item tag 0x0
[<    0,000326>] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.001C/input/input45
[<    0,000504>] hid-generic 0005:045E:02E0.001C: input,hidraw5: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on ac:ed:5c:5d:5e:93
```

kakra, 2020-12-11:
It doesn't seem to trigger loading/binding xpadneo. Could you check if DKMS actually reinstalled xpadneo after the latest kernel update? I have at least one user report that found DKMS didn't rebuild the driver after kernel update. The cause for this is unknown but you can fix it by reinstalling the driver.

rubencabrera, 2020-12-12:
`dkms status` yields: `hid-xpadneo, v0.9-4-gbf8a3c3, 4.15.0-124-generic, x86_64: installed` but `uname -a` shows `Linux myhostname 4.15.0-126-generic`. I don't remember updating the kernel between the moment the second Xbox controller could connect and when none of the 2 were detected, these 2 events were almost immediate, IIRC. Not sure if there's a correlation here or I just messed with something that made this more confusing. 

Well, another kernel update became available, so I just installed.

`dkms status` showed

```
hid-xpadneo, v0.9-4-gbf8a3c3, 4.15.0-124-generic, x86_64: installed
hid-xpadneo, v0.9-4-gbf8a3c3, 4.15.0-128-generic, x86_64: installed
```
 but nothing changed.

Uninstalled xpadneo and installed again. When connecting the 1708 controller it rumbled (which it didn't do before even though it was working quite well in every other aspect). The 1914 doesn't rumble but it is now detected and usable in games (haven't checked all inputs but good enough to win a race in SuperTux :) )

I'm going to give it a test run this weekend, thanks a lot for your help!

In case it helps anyone who doesn't have a dual boot or a second windows device (connecting the 1914 controller to an android device didn't do the trick for me):

- Get a Windows 10 dev VM image from [here](https://developer.microsoft.com/en-us/windows/downloads/virtual-machines/)
- If you choose Virtualbox, don't install the one in the repos, you might get an older version which won't be compatible, look for 6.1 in [Oracle's downloads](https://www.oracle.com/virtualization/technologies/vm/downloads/virtualbox-downloads.html).
- Don't forget to install the [extensions](https://www.oracle.com/virtualization/technologies/vm/downloads/virtualbox-downloads.html#extpack) for VBox if that's what you're using.
- Import the VM, run it and install the guest extensions there too (you'll get a notification about the CD or you'll be able to insert it from the "Devices" menu).
- Make sure the user you use to run VBox is in the vboxusers group (check with `groups` on your terminal). If not, add it with `sudo adduser $USER vboxusers`, this is quite important for the guest VM to see the bluetooth controller.
- You might have to reboot the Guest (and VBox) for the next steps to work, I did but I'm not sure if needed.
- On the bottom bar of the guest window of VBox, click on the USB icon and you'll see some devices available, you should have identified which is your Bluetooth one by running `lsusb -v` and looking for the name of the device with `Bluetooth` next to `bDeviceProtocol` (that's how I found mine, yours might differ).  Select the correct device. In my case VBox took hold of the controller and Bluetooth was not available on the host anymore, if this doesn't happen automatically, try to disable BT on the host first.
- Windows won't install the driver, so you'll need to go to your manufacturer's site and download the drivers for the right one. 
- Once that's done, you should be able to connect your controller, light should stay on and after that you should be safe to disable the BT controller on the guest, thus recovering BT on the host, and connecting your XBox controller. It should just work (TM). 

I used [this askubuntu answer](https://askubuntu.com/questions/963424/cant-access-in-built-bluetooth-for-windows-in-virtual-box) as help, what I wrote here adds some of my experience but I found the answer very useful in case you want to check.

kakra, 2020-12-12:
If you did the `dmesg -w` thing again, you should see hid-xpadneo taking over the device this time. I'm not sure if this worked for the second controller: If it doesn't rumble on connect, it's usually not been taken over by xpadneo.

rubencabrera, 2020-12-12:
`dmesg -w` shows when the 1708 connects:
```
[<   40,686001>] hid-generic 0005:045E:02E0.000B: unknown main item tag 0x0
[<    0,000205>] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.000B/input/input28
[<    0,000474>] hid-generic 0005:045E:02E0.000B: input,hidraw5: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on ac:ed:5c:5d:5e:93
```

1914 controller then enters the connected/disconnected loop as before connecting it to Windows 10 (I'm now stuck at getting the guest VM to see the bluetooth again, for some reason my steps above don't work now). No messages are shown in `dmesg -w` output.

rubencabrera, 2020-12-12:
VirtualBox udev rules had disappeared, I reinstalled virtualbox 6.1 and in the Widows 10 VM I had the controller light flashing, so had to remove it and pair it again with Windows 10. That succeeded. I went back to the host and now, again, both controllers connect but xpadneo is not used, I have the same output in dmesg as before. Uninstalling xpadneo and installing again.

Dmesg shows:

First controller (1708):
```
[<  289,449072>] hid-generic 0005:045E:02E0.0008: unknown main item tag 0x0
[<    0,000202>] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0008/input/input25
[<    0,000549>] hid-generic 0005:045E:02E0.0008: input,hidraw5: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on ac:ed:5c:5d:5e:93
[<    0,095636>] hid_xpadneo: unknown parameter 'disable_ff' ignored
[<    0,000258>] xpadneo 0005:045E:0B13.0006: fixing up Rx axis
[<    0,000004>] xpadneo 0005:045E:0B13.0006: fixing up Ry axis
[<    0,000004>] xpadneo 0005:045E:0B13.0006: fixing up Z axis
[<    0,000004>] xpadneo 0005:045E:0B13.0006: fixing up Rz axis
[<    0,000004>] xpadneo 0005:045E:0B13.0006: fixing up button mapping
[<    0,000491>] xpadneo 0005:045E:0B13.0006: working around wrong SDL2 mappings (changed version from 0x00000501 to 0x00000501)
[<    0,000004>] xpadneo 0005:045E:0B13.0006: enabling compliance with Linux Gamepad Specification
[<    0,000160>] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0006/input/input26
[<    0,000424>] xpadneo 0005:045E:0B13.0006: input,hidraw5: BLUETOOTH HID v5.01 Gamepad [Xbox Wireless Controller] on AC:ED:5C:5D:5E:93
[<    0,000010>] xpadneo 0005:045E:0B13.0006: controller quirks: 0x00000050
[<    0,980188>] xpadneo 0005:045E:0B13.0006: Xbox Wireless Controller [44:16:22:6B:C3:1A] connected
[<    0,000088>] xpadneo 0005:045E:02E0.0008: fixing up report size
[<    0,000211>] xpadneo 0005:045E:02E0.0008: battery detected
[<    0,000002>] xpadneo 0005:045E:02E0.0008: working around wrong SDL2 mappings (changed version from 0x00000903 to 0x00000903)
[<    0,000000>] xpadneo 0005:045E:02E0.0008: enabling compliance with Linux Gamepad Specification
[<    0,000061>] input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/bluetooth/hci0/hci0:256/0005:045E:02E0.0008/input/input27
[<    0,000231>] xpadneo 0005:045E:02E0.0008: input,hidraw6: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on ac:ed:5c:5d:5e:93
[<    0,000002>] xpadneo 0005:045E:02E0.0008: controller quirks: 0x00000040
[<    0,979578>] xpadneo 0005:045E:02E0.0008: Xbox Wireless Controller [44:16:22:0a:d3:b8] connected
[<    0,798706>] xpadneo 0005:045E:02E0.0008: battery registered
```

Nothing shows up when I connect the second controller, but what you see in the output above is also related to it (its the MAC ending in `1A`).

Both controllers are usable not from games, but 1914 controller is having the half a second lag. 

I'll leave things as they are till tomorrow and be sure to not install anything or do any upgrades. I want to see if the same situations repeats itself after a reboot. Let me know if there's any other test of logs that might be useful.

T-Zeros, 2020-12-16:
I guess I found a solution to the same problem, I [commented in another issue](https://github.com/atar-axis/xpadneo/issues/166#issuecomment-747010866).
But for completeness: just add "Privacy=device" to /etc/bluetooth/main.conf and restart your bluetooth service

nonooo, 2020-12-25:
I can confirm that the solution with the Privacy=device works...
Take care that in debian you need to fill with space:

# Default privacy setting.
# Enables use of private address.
# Possible values: "off", "device", "network"
# "network" option not supported currently
# Defaults to "off"
Privacy = device

asukiaaa, 2021-01-01:
@T-Zeros Thank you.
Adding `Privacy=device` to General section of `/etc/bluetooth/main.conf` solves problem on my PC (Ubuntu 20.04).

kakra, 2021-01-01:
I've upgraded to kernel 5.10 (from 5.4) today and found some interesting bits which seem:

The kernel Bluetooth stack solved a bug for BLE devices in privacy mode: https://bugzilla.kernel.org/show_bug.cgi?id=209745, seen since 5.9, solved since 5.10.4, not sure if it has been backported yet.

They also added support for Microsoft HCI extensions in the Bluetooth stack (probably in 5.6) although this seems to be a driver thing and is implemented only for two drivers so far: https://docs.microsoft.com/en-us/windows-hardware/drivers/bluetooth/microsoft-defined-bluetooth-hci-commands-and-events - it's probably not related to any stability with the Xbox controller (do not confuse "controller" in the mentioned document with the Xbox controller). It seems to be mainly about setting proximity conditions in the driver when a device should be detected by the driver and be allowed to connect (RSSI thresholds). But it's still an example how MS implements their own "inventions" in a vendor-specific way without going for some standards. Who knows what "vendor specific extensions" the Xbox controller may use.

kakra, 2021-01-01:
I've added the Bluetooth stack related information to our kernel bug tracking issue: https://github.com/atar-axis/xpadneo/issues/198

ShadowNinja, 2021-01-05:
Enabling privacy was initially causing issues for me (after enabling privacy I couldn't start a bluetooth scan on either 5.9 or 5.4) but it worked on second computer (running 5.4).  However, I've updated to 5.10.4 and the issue on the first machine seems to be fixed now.

So setting `Privacy=device` in `bluetooth.conf` (or temporarily enabling it by entering `power off` `privacy on` `power on` in `btmgmt` as root) seems to resolve any issues as long as the driver for your hardware isn't bugged like mine was before 5.10 (Intel AX200 8087:0029).

ivyl, 2021-01-10:
I've just got my hands on model 1914 and I had the same issue with the controller being stuck in the reconnecting loop with my `8087:0029 Intel Corp. AX200 Bluetooth`. I've updated the firmware to the latest using Windows VM but that didn't help. I've then redirected the bluetooth controller to the VM, paired the controller there (I got prompted about the fast proximity-based pairing thingy which I have used), unpaired it and then next pairing attempt on the Linux side worked just fine and the controller is still working.

If you want me to test me anything let me know. I am even willing to disassemble the controller. It would be nice to get this running with less hassle for the others.

kakra, 2021-01-10:
By disassembling you mean "physically taking it apart" or "decompiling its firmware"? The latter case would be interesting because we could maybe fix the Bluetooth stability bug in the firmware. There seem to be actually two bugs: Sending rumble commands faster then 10ms per cycle crashes the Bluetooth connection (Windows uses the same work-around for this as we do), and sending certain (yet unknown) rumble patterns also crash it (bug exists also in Windows). I quite sure both these bugs are completely unrelated to the Bluetooth implementation on the OS side.

rubencabrera, 2021-01-10:
@ivyl Do you have another controller to test connecting both at the same time? I'm experiencing a lag on the 1914 that makes gaming quite frustrating. I've just connected the 1914 and with it being the only ones I can still notice it, but it's not so bad that feels broken.

kakra, 2021-01-10:
If I'm seeing lag in Proton games, it's usually due to the game itself and how it uses CPU cores. That can probably be fixed by using a Wine version with thread priority patches (Proton experimental has them, and when I made my own Proton version back in the 3.x and 4.x times, I also included and enhanced those patches to fix it). In that case, most likely you also need rtkit and some rlimit adjustments.

Using the controller directly from CLI, I'm seeing no lag, using models XB1S and XBE2. There's a very high lag in KDE system settings but that's present with any joystick so I'd attribute that to KDE itself.

ivyl, 2021-01-10:
@kakra I've meant taking it apart in hopes of finding a physical way of resetting the state without resorting to Windows, but looking at the upgrade process, the source of the blobs and maybe trying to bring that to Linux may also be interesting. I'll probably play with that a bit as time allows but no promises :-)

[20 more replies]

closed 2022-09-17 by kakra