atar-axis/xpadneo #261: Xbox One Bluetooth controller no longer recognized in RPCS3 (Natively) or Cemu (Wine)

ipkpjersi, 2020-12-05:
## Version of xpadneo

## Severity / Impact

* [x] I've read the docs and the bug reporting instructions
* [x] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the bug

My Xbox One Bluetooth controller is no longer recognized by any games or programs in Ubuntu 18.04 after upgrading from xpadneo 0.8.3 to xpadneo 0.9 and no buttons work on the controller. The controller does power on and stay powered on/connected via Bluetooth but no buttons work.

## Steps to Reproduce

1. Install xpadneo 0.9 on Ubuntu 18.04
2. Launch any game

## Expected behavior

I expect the controller to work as it did 

## Screenshots/Gifs

As you can see, the device is not even detected here in RPCS3:
[image]

## System information

```console
# uname -a
Linux desktop 4.15.0-126-generic #129-Ubuntu SMP Mon Nov 23 18:53:38 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0

```

## Controller and Bluetooth information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/5646513/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/5646514/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/5646515/xpadneo-lsusb.txt)

kakra, 2020-12-05:
This looks like the driver hasn't even been loaded. This may be an effect of using kernel 4.15 still, and I'm tempted to say we no longer support this kernel. However, did the udev rules install properly?

Please try the following:

1. Disconnect the controller (turn it off)
2. Run as root: `udevadm monitor`
3. Turn the controller on
4. Post the result

You could also try auto-loading the xpadneo driver on boot.

ipkpjersi, 2020-12-05:
Here is the logs you asked for:
[udev-logs.txt](https://github.com/atar-axis/xpadneo/files/5647383/udev-logs.txt)

However, with some quick testing by me, just like that I've found the problem. It's installed for my kernel version 4.15.0-124 but not 4.15.0-126. I am not sure why it didn't install to my newer kernel version - perhaps the DKMS automatic rebuilding is not working after upgrading kernels? I'm not sure.

```
    sudo dkms status
bcmwl, 6.30.223.271+bdcom, 4.15.0-124-generic, x86_64: installed
bcmwl, 6.30.223.271+bdcom, 4.15.0-126-generic, x86_64: installed
hid-xpadneo, v0.9, 4.15.0-124-generic, x86_64: installed
nvidia, 450.80.02, 4.15.0-124-generic, x86_64: installed
nvidia, 450.80.02, 4.15.0-126-generic, x86_64: installed
openrazer-driver, 2.9.0, 4.15.0-124-generic, x86_64: installed
openrazer-driver, 2.9.0, 4.15.0-126-generic, x86_64: installed
virtualbox, 5.2.42, 4.15.0-124-generic, x86_64: installed
virtualbox, 5.2.42, 4.15.0-126-generic, x86_64: installed
zfs, 0.8.5, 4.15.0-124-generic, x86_64: installed
zfs, 0.8.5, 4.15.0-126-generic, x86_64: installed
```

I uninstalled it, re-installed it, and now it seems to be fine again.

I'm really hoping you continue supporting kernel 4.15 as it's an LTS kernel, which is why I'm using it, and I'd love to have new features from any newer versions of xpadneo.

This issue can possibly be closed now.

ipkpjersi, 2020-12-05:
Oh also one quick question while I'm here. Since I have an original Xbox One Bluetooth controller, is it safe to update the firmware of it to the latest - was support for the latest firmware added in 0.8.3 or only 0.9? Also, how can I check the firmware of my controller?

kakra, 2020-12-06:
The Xbox One should have a built-in app that updates the firmware of the controller. But I do not own one, so I cannot give further instructions.

For firmware related changes you can look at the news files. Keep in mind I don't update it unless tagging a new version so you might want to look at the git commit history, too. Regardless of the firmware, the controller should generally work, tho. However, some button mappings may be wrong. I'd recommend to always use the latest firmware.

kakra, 2020-12-08:
Is this still an open issue? Seems like only the module did not properly rebuilt...

ipkpjersi, 2020-12-08:
I think we can close this for now and re-open this if it happens again.

closed 2020-12-08 by ipkpjersi

ipkpjersi, 2020-12-19:
Good news for this, it definitely looks like it was a one time issue, my system updated to 4.15.0-128-generic and DKMS automatically rebuilt the module:
[image] :+1: