atar-axis/xpadneo #262: Xbox One Controller asks for pin during pairing

Mikaka27, 2020-12-13:
Hi,
This is not a problem with xpadneo (I think) as xpadneo module is not even loaded at this point.

I hope you allow this discussion, since here I'll be able to find people who use xbox one controller on linux.

I'm on manjaro KDE, and since like 2 weeks I cannot pair the controller at all (previously paired controller also didn't connect).
During connection my pc tells me to enter a (randomly generated) pin on the xbox one controller and obviously I cannot do that :)

**Connection attempt through gui (sorry it's not in english):**
[image]
[image]
[image]

**Connection attempt through bluetoothctl:**
```
[mikaka@stacjonarny ~]$ sudo bluetoothctl
[sudo] hasło użytkownika mikaka: 
Agent registered
[bluetooth]# scan on
Discovery started
[NEW] Device 5C:BA:37:7D:52:19 Xbox Wireless Controller
[bluetooth]# pair 5C:BA:37:7D:52:19
Attempting to pair with 5C:BA:37:7D:52:19
[CHG] Device 5C:BA:37:7D:52:19 Connected: yes
[CHG] Device 5C:BA:37:7D:52:19 Connected: no
[CHG] Device 5C:BA:37:7D:52:19 Connected: yes
Request PIN code
[agent] Enter PIN code: Request canceled
Failed to pair: org.bluez.Error.AuthenticationCanceled
[CHG] Device 5C:BA:37:7D:52:19 Connected: no
```

I hope I can find someone here with this problem.

**System information:**
```
[mikaka@stacjonarny ~]$ uname -a
Linux stacjonarny 5.8.18-1-MANJARO #1 SMP PREEMPT Sun Nov 1 14:10:04 UTC 2020 x86_64 GNU/Linux
```

```
[mikaka@stacjonarny ~]$ lsmod | grep xpadneo
[mikaka@stacjonarny ~]$ 
```

```
[mikaka@stacjonarny ~]$ cat /sys/module/bluetooth/parameters/disable_ertm
Y
```

```
[mikaka@stacjonarny ~]$ pamac list | grep bluez
bluez                              5.55-1                      extra                            2,6 MB
bluez-libs                         5.55-1                      extra                            336,4 kB
bluez-qt                           5.76.0-1                    extra                            2,0 MB
bluez-utils                        5.55-1                      extra                            7,2 MB
```

```
[mikaka@stacjonarny ~]$ pamac list | grep xpadneo
xpadneo-dkms                       0.9-2                       AUR        44,0 kB
```

```
Operating System: Manjaro Linux
KDE Plasma Version: 5.20.4
KDE Frameworks Version: 5.76.0
Qt Version: 5.15.2
Kernel Version: 5.8.18-1-MANJARO
OS Type: 64-bit
Processors: 4 × Intel® Core™ i5-6600K CPU @ 3.50GHz
Memory: 23.4 GiB of RAM
Graphics Processor: GeForce GTX 1660 Ti/PCIe/SSE2
```

**Possibly relevant information:**
Only thing bluetooth related I did in last weeks was to install this pulseaudio fork (which provides better bluetooth headset functionality), so this 2 packages:
https://aur.archlinux.org/packages/pulseaudio-hsphfpd/
https://aur.archlinux.org/packages/hsphfpd-git/

I'd appreciate any help if someone else has this problem :)

kakra, 2020-12-13:
I can only think of the correct pairing agent not being registered. Try in `bluetoothctl`:

* `agent off`
* `agent NoInputNoOutput`
* `agent DisplayYesNo`

and try again after each setting using `connect MAC` or `pair MAC` in `bluetoothctl`.

Mikaka27, 2020-12-14:
Hi,
Thanks for the help I was able to workaround the issue, although it's not perfect. I did like this:
```
sudo bluetoothctl
agent off
agent NoInputNoOutput
scan on
pair <mac>
trust <mac>
connect <mac>
```

Connection fails unfortunately. However I then do
```
power off
power on
connect <mac>
```

This succeeds. The gamepad is paired, however after every reboot I have to turn on xbox controller and then do:
```
sudo bluetoothctl
power off
power on
```

And it connects. This doesn't happen on live cd I guess something is f*cked in some configuration file somewhere.
This is the PC bluetooth hardware information command in bluetoothctl:
```
[Xbox Wireless Controller]# show FC:F8:AE:D5:52:B3
Controller FC:F8:AE:D5:52:B3 (public)
        Name: stacjonarny
        Alias: stacjonarny
        Class: 0x00000000
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4
        Pairable: yes
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
        Modalias: usb:v1D6Bp0246d0537
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
```

kakra, 2020-12-14:
Yeah, I've seen this behavior myself. I could fix it by pairing the controller again and again until the link key properly showed up in `/var/lib/bluetooth/HOST-MAC/DEV-MAC/info`. Before pairing again, you need to remove the device with `bluetoothctl remove MAC`.

There seems to be a race condition when using bluez with the KDE GUI client where it simply confirms pairing for a short time after powering the host radio on. The GUI client also sometimes mixes up the name of a connected device and its MAC, so there may actually be bugs. Usually, you should be able to get the controller connected by simply rebooting while it tries to connect.

Mikaka27, 2020-12-17:
Hi,
For me the keys are there (I think):
```
[mikaka@stacjonarny ~]$ sudo tree /var/lib/bluetooth   
/var/lib/bluetooth
└── FC:F8:AE:D5:52:B3
    ├── 5C:BA:37:75:08:44
    │   └── info
    ├── 5C:BA:37:7D:52:19
    │   └── info
    ├── C8:D7:B0:E4:DB:1B
    │   ├── attributes
    │   └── info
    ├── cache
    │   ├── 00:01:90:91:17:7F
    │   ├── 00:02:3C:77:72:A2
    │   ├── 00:02:3C:77:9C:83
    │   ├── 00:7C:2D:22:92:AF
    │   ├── 08:EF:3B:3A:00:54
    │   ├── 3C:BD:3E:6D:69:21
    │   ├── 50:A0:09:B0:F7:52
    │   ├── 5C:BA:37:75:08:44
    │   ├── 5C:BA:37:7D:52:19
    │   ├── 80:9F:9B:6C:E6:99
    │   ├── 98:54:1B:0D:44:62
    │   ├── C0:B5:D7:98:5F:1A
    │   ├── C4:57:6E:B6:BF:A7
    │   ├── C8:D7:B0:E4:DB:1B
    │   ├── EC:D0:9F:A0:87:B0
    │   ├── F3:AB:E7:AC:9D:AE
    │   ├── F4:FE:FB:82:89:AB
    │   └── FA:49:D7:79:3B:28
    ├── EC:D0:9F:A0:87:B0
    │   ├── attributes
    │   └── info
    └── settings

6 directories, 25 files
```

`5C:BA:37:75:08:44` and `5C:BA:37:7D:52:19` are 2 xbox one controllers I have. After repairing 2 times the behaviour is still the same, they don't connect unless I restart the bluetooth controller.
What strikes me odd now is that before and after power cycle (manual in bluetoothctl) the pc controller's bluetooth class is changing. See here:
```
[bluetooth]# list
Controller FC:F8:AE:D5:52:B3 stacjonarny [default]
[bluetooth]# show FC:F8:AE:D5:52:B3
Controller FC:F8:AE:D5:52:B3 (public)
        Name: stacjonarny
        Alias: stacjonarny
        Class: 0x007c0104
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4
        Pairable: yes
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
        Modalias: usb:v1D6Bp0246d0537
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
[bluetooth]# power off
Changing power off succeeded
[CHG] Controller FC:F8:AE:D5:52:B3 Powered: no
[CHG] Controller FC:F8:AE:D5:52:B3 Discovering: no
[CHG] Controller FC:F8:AE:D5:52:B3 Class: 0x00000000
[bluetooth]# power on
Changing power on succeeded
[CHG] Controller FC:F8:AE:D5:52:B3 Powered: yes
[bluetooth]# show FC:F8:AE:D5:52:B3
Controller FC:F8:AE:D5:52:B3 (public)
        Name: stacjonarny
        Alias: stacjonarny
        Class: 0x00000000
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4
        Pairable: yes
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
        Modalias: usb:v1D6Bp0246d0537
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
```

After powering on PC it says: `Class: 0x007c0104` but after controller restart it says: `Class: 0x00000000`. Not sure if that's relevant to the problem, but I'll keep digging.

Mikaka27, 2020-12-17:
Hi,
I seem to have fixed it. When I looked into 'journalctl' when connecting to gamepad I noticed this line:
`stacjonarny bluetoothd[850]: profiles/input/device.c:control_connect_cb() connect to 5C:BA:37:7D:52:19: Host is down (112)`

I don't know what this means exactly, however everything started to work (including pairing from gui) when I changed:
`#AutoEnable=false`
to
`AutoEnable=true`

In `/etc/bluetooth/main.conf`. I'm not sure why it helps though :(
Leaving it open if you want to comment.

kakra, 2020-12-17:
Bluetooth does usually not enable itself to accept connections until a desktop agent is ready. This option changes that.

Mikaka27, 2020-12-20:
Hi, sorry for late response. Do you mean I still have this "desktop agent" broken somehow that it's not ready?
Also do you know what this "agent" really is?

I mean:

- systemd service
- executable
- .so library
- ...

I'm not sure what to look for. Thanks for all the help.

kakra, 2020-12-20:
My understanding is that an agent is some running tool that registers with dbus to the Bluetooth service. By default, it's bluetoothctl itself unless you run a different tool.

kakra, 2021-01-31:
This may be a duplicate of #272.

kakra, 2021-01-31:
May also be related to #166 as the remaining pairing issues are probably with the authentication agent (once pairing in ERTM mode works from the kernel side).

kakra, 2021-05-03:
Is this still an issue with kernel 5.12 and ERTM enabled?

Mikaka27, 2021-05-03:
Yes if I revert the "fix" the issue still occurs. But it must be something with my setup, this doesn't happen on "live cd" linux for example.
This workaround is good enough for me, and it likely has nothing to do with xpadneo, so I guess we should close this ticket.

kakra, 2021-05-03:
I think the concurrent installation of two Bluetooth managers may cause it: They probably listen on the same dbus endpoints and may interfere with each other.

> if I revert the "fix" [`AutoEnable=true`] the issue still occurs

Usually, the desktop agent will enable the adapter. If this cannot happen properly due to interference between concurrent agents, you may need to just work with this setting. So I agree we should close it. It's even not really a Bluetooth issue in this case but something awkward with your setup which thankfully has a work-around.

lsantos7654, 2024-01-29:
I saw this post on reddit, It resolved this issue for me.

[penguinhasan](https://www.reddit.com/user/penguinhasan/)
·
[19 hr. ago](https://www.reddit.com/r/linux_gaming/comments/tdh3gq/comment/kjxg9bk/?utm_source=reddit&utm_medium=web2x&context=3)

After no solution from the internet, I worked days after days tinkering with bluetooth profiles and finally solved it. I am here to share my solution:

First of all, if you're using first party controller, use an external USB bluetooth which should connect without any tinkering. Then move to the solution bellow (this will not only work for xbox, but also third party generic bluetooth gamepads, chinese bluetooth gamepads, EasySMX gamepads with Nintendo Switch Controller mode and some bluetooth keyboards)

Edit /etc/bluetooth/input.conf (if not available, create one) and edit the options to match the following config:

# Configuration file for the input service

# This section contains options which are not specific to any
# particular interface
[General]

# Set idle timeout (in minutes) before the connection will
# be disconnect (defaults to 0 for no timeout)
#IdleTimeout=30

# Enable HID protocol handling in userspace input profile
# Defaults to false (HIDP handled in HIDP kernel module)
UserspaceHID=true

# Limit HID connections to bonded devices
# The HID Profile does not specify that devices must be bonded, however some
# platforms may want to make sure that input connections only come from bonded
# device connections. Several older mice have been known for not supporting
# pairing/encryption.
# Defaults to true for security.
ClassicBondedOnly=false

# LE upgrade security
# Enables upgrades of security automatically if required.
# Defaults to true to maximize device compatibility.
LEAutoSecurity=false

Now, your device will connect even if the pairing key is not exchanged or the bluetooth adapter fails to exchange the keys.

kakra, 2024-01-30:
@lsantos7654 Re-opening for including your suggestion into the documentation.

closed 2024-03-04 by kakra