atar-axis/xpadneo #267: Xbox Wireless Controller Model 1708 completely doesn’t work with pi4 and RetroPie

mjboggs, 2021-01-05:
Purchased pi 4 with 4GB RAM and 32GB SD.  Installed Retropie via the Windows Pi Imager.  Followed instructions to install xpadneo. Attempted for hours to get the controller to connect.  RetroPie has NEVER seen the controller as connected.  All these discussions about mapping, I wish I could get that far!  This just does not work with the Xbox Wireless Controller and RetroPie running on a pi 4.  Didn’t either on a pi 3 some years/months back.  Here is a screenshot of the bluetoothctl flow.  Right before pair, I pressed the connection button.  Pair was successful.  Trust was successful (interestingly, immediately, as if it didn’t do anything actually).  Upon trust command, it reported as connected, but was not as the lights on the controller never stopping blinking.  Eventually, appears to timeout and the light goes dark.

So, RetroPie does not work with XBox wireless controller even with xpadneo.  
[image]

kakra, 2021-01-05:
Any chance you could run a 5.4.84+ kernel, or 5.9.x, or 5.10.4+?

kakra, 2021-01-05:
Also, please check if ERTM is disabled. If it is enabled, you may see this result. Maybe try a different Bluetooth adapter (tho, the history of issues shows that this has the lesser impact on success than originally thought).

kakra, 2021-01-05:
There's one strange observation: Why are we seeing two different modalias lines in your screenshot? This almost looks like the controller is switching to a different operating mode internally. Maybe try pairing it to your mobile phone before pairing it to the Pi and see if that behavior changes.

mjboggs, 2021-01-06:
> Any chance you could run a 5.4.84+ kernel, or 5.9.x, or 5.10.4+?
> 
> Also, please try updating the firmware of the controller using either an Xbox or the Windows Xbox Accessories app. Firmware updates do not work with Bluetooth: Be sure to use either the original dongle, or the USB cable.

Thanks for the idea.  I don't know how to run it on those different kernels.  Can you give me an idea how to change the kernel version? As far as the firmware update, I already had tried that using the XBox Accessories in Windows 10 and a cable.  There were no updates.

mjboggs, 2021-01-06:
> Also, please check if ERTM is disabled. If it is enabled, you may see this result. Maybe try a different Bluetooth adapter (tho, the history of issues shows that this has a lesser impact on success than originally thought).

Thank you for the advice.  I had disabled ERTM before.  There isn't a way to try a different Bluetooth adapter to my knowledge as this controller has Bluetooth as does the pi 4.

mjboggs, 2021-01-06:
> There's one strange observation: Why are we seeing two different modalias lines in your screenshot? This almost looks like the controller is switching to a different operating mode internally. Maybe try pairing it to your mobile phone before pairing it to the Pi and see if that behavior changes.

That was odd.  I tried to pair it with Windows 10 again and it would not work.  Eventually, after I pulled the batteries out, it did re-pair with Windows 10.  I tried again with xpadneo and it is still not functioning.

kakra, 2021-01-06:
Here are a few more things to try as this seems similar to your observation: https://github.com/atar-axis/xpadneo/issues/259

The important bit may be the privacy extensions but you'll probably need more recent kernel for this to work.

kakra, 2026-10-25:
Please check whether #449 fixes your issue (also see the linked commit). If it is still a problem with modern kernels, please open a new issue report.

closed 2026-10-25 by kakra