atar-axis/xpadneo #268: [Help wanted and maybe a bug] xbox series x controller 0005:045E:0B13.0005 works with usb cable but wrongly mapped with bluetooth

TTTPOB, 2021-01-10:
## Version of xpadneo

2020.11.22

## Severity / Impact

* [ x ] I've read the docs and the bug reporting instructions
* [ ] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the bug

xbox series x controller 0005:045E:0B13.0005 works with usb cable but wrongly mapped with bluetooth

it reported it has 9 axis, and axis 9 will be affected by axis 3(lt) and axis 6(rt). back key was not responsible. share key mapped to key 7, left stick and right stick works normally. dpad seems normally too
I cannot see how other keys which I didnt mentioned work.

## Steps to Reproduce

## Expected behavior

## Screenshots/Gifs

## System information

```console
# uname -a
Linux laptop 5.10.5-arch1-1 #1 SMP PREEMPT Thu, 07 Jan 2021 09:50:43 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...

2986910699 1363

```

## Controller and Bluetooth information

Intel Corp. AX200 Bluetooth

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/5792659/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/5792660/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/5792662/xpadneo-lsusb.txt)

## Additional context

I cannot see the folder `/sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045Exxxx` if i connect the gamepad using usb

lights0123, 2021-01-10:
Looks like you have a different OID (first 3 bytes of MAC address) than mine when I added support. Mine started with `44:16:22`, yours starts with `98:7a:14`.

kakra, 2021-01-10:
Let's first find out if the kernel has the proper mapping: Does `evtest` show the proper mappings?

1. If yes: Then we need to fix toward working around wrong SDL2 mappings probably.
2. If no: Then we need to introduce a quirk to properly handle this model.

lights0123, 2021-01-10:
It's not going to because it doesn't have the 0x40 quirk.

kakra, 2021-01-10:
> I cannot see the folder `/sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045Exxxx` if i connect the gamepad using usb

That's expected: xpadneo does not support USB mode of the controller (which works very differently and uses a different protocol).

kakra, 2021-01-10:
> it reported it has 9 axis, and axis 9 will be affected by axis 3(lt) and axis 6(rt).

That last axis combines the triggers, it results from a feature request, that's okay. But we should see only 7 analog axes (LX,LY,RX,RY,LT,RT,Combined), not 9. If you are looking with `jstest`, you'd actually see 9 axes, axis 7+8 being the dpad. In that case it's okay, otherwise the HID parsing and fixes may need fixes.

So: Which program did you use to see the axes and mappings?

kakra, 2021-01-10:
> share key mapped to key 7, left stick and right stick works normally

@lights0123 This mostly looks like it needs the share button quirk? Do you want to send a patch?

TTTPOB, 2021-01-10:
> That last axis combines the triggers, it results from a feature request, that's okay. But we should see only 7 analog axes (LX,LY,RX,RY,LT,RT,Combined), not 9. If you are looking with `jstest`, [...]

yes, jstest based program, to be more specific, the one in kde settings

TTTPOB, 2021-01-10:
> Let's first find out if the kernel has the proper mapping: Does `evtest` show the proper mappings?
> 
> 1. If yes: Then we need to fix toward working around wrong SDL2 mappings probably.
> 2. If no: Then we need to introduce a quirk to properly handle this model.

no, `back` key is not responding

kakra, 2021-01-10:
@TTTPOB Try to reload the driver with these settings:
```
sudo rmmod xpadneo
sudo modprobe xpadneo quirks=98:7a:14:f2:2a:69:0x50
```

Does it fix the mappings?

kakra, 2021-01-10:
> yes, jstest based program, to be more specific, the one in kde settings

If you use that, be sure to NOT run the calibration. It won't fix anything and makes debugging more complicated. If you used it, you may need to remove the controller from `/var/lib/joystick/joystick.state` (if you distribution writes this file, otherwise simply disconnect and reconnect).

TTTPOB, 2021-01-10:
> > yes, jstest based program, to be more specific, the one in kde settings
> 
> If you use that, be sure to NOT run the calibration. It won't fix anything and makes debugging more complicated. If you used it, you may need to remove the controller from `/var/lib/joystick/joystick.state` (if your distribution writes this file, otherwise simply disconnect and reconnect).

thank you for the helpful reminder! I would never touch that.

lights0123, 2021-01-10:
@kakra I can send a patch, but I'm busy until Tuesday so it'll have to wait until then. If you just want to match on OID, though, it should just be a one-liner copy/paste.

kakra, 2021-01-10:
@lights0123 I'm seeing a problem here: `98:7a:14` also identifies the XBE2 controller, so we're having a conflict here and need to find another way of identifying that controller. Here's my XBE2 dmesg:
```
[39817.382285] xpadneo 0005:045E:0B05.0016: fixing up report size
[39817.382286] xpadneo 0005:045E:0B05.0016: fixing up Rx axis
[39817.382287] xpadneo 0005:045E:0B05.0016: fixing up Ry axis
[39817.382287] xpadneo 0005:045E:0B05.0016: fixing up Z axis
[39817.382288] xpadneo 0005:045E:0B05.0016: fixing up Rz axis
[39817.382288] xpadneo 0005:045E:0B05.0016: fixing up button mapping
[39817.382680] xpadneo 0005:045E:0B05.0016: battery detected
[39817.382685] xpadneo 0005:045E:0B05.0016: pretending XB1S Windows wireless mode (changed PID from 0x0B05 to 0x02E0)
[39817.382686] xpadneo 0005:045E:0B05.0016: working around wrong SDL2 mappings (changed version from 0x00000408 to 0x00000903)
[39817.382687] xpadneo 0005:045E:0B05.0016: disabling dead zones
[39817.382688] xpadneo 0005:045E:0B05.0016: enabling compliance with Linux Gamepad Specification
[39817.382742] input: Xbox Elite Wireless Controller as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2.7/2-1.2.7.2/2-1.2.7.2:1.0/bluetooth/hci0/hci0:75/0005:045E:0B05.0016/input/input28
[39817.382871] xpadneo 0005:045E:0B05.0016: input,hidraw15: BLUETOOTH HID v4.08 Gamepad [Xbox Elite Wireless Controller] on 00:1a:7d:da:71:15
[39817.382874] xpadneo 0005:045E:0B05.0016: controller quirks: 0x00000018
[39817.382875] xpadneo xpadneo_welcome_rumble start
[39818.375032] xpadneo xpadneo_welcome_rumble took 990ms
[39818.375037] xpadneo 0005:045E:0B05.0016: Xbox Elite Wireless Controller [98:7a:14:61:17:4d] connected
```

kakra, 2021-01-10:
@lights0123 The XBE2 controller has a bigger HID descriptor, we could use that:
```
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
...
3995898903 4781
```

I should probably store these data in the controller struct and maybe also log it to dmesg.

I'll prepare that but I don't own a XBXS controller to verify it would be working correctly. We would need to cross-check between both XBXS variants and the XBE2.

TTTPOB, 2021-01-10:
> @TTTPOB Try to reload the driver with these settings:
> 
> ```
> sudo rmmod xpadneo
> sudo modprobe xpadneo quirks=98:7a:14:f2:2a:69:0x50
> ```
> 
> Does it fix the mappings?

hmmm... does it has a differnt name under archlinux? 
`modprobe: FATAL: Module xpadneo not found in directory /lib/modules/5.10.5-arch1-1`
i am pretty sure i've installed correctly cause when i tried to install it again
[image]
what else can i help?

lights0123, 2021-01-10:
Yeah, it's hid-xpadneo in Arch

kakra, 2021-01-10:
> hmmm... does it has a differnt name under archlinux? 

@TTTPOB Oops, fixed above: `xpadneo -> hid-xpadneo`. Sorry.

TTTPOB, 2021-01-10:
thank you two!
worked this time, but still wrong mapping
back > not working
share > BTN_SELECT (value 90007)
start > BTN_START (value 90008)
Y > BTN_WEST (90004)
X > BTN_NORTH (90003)
A > noproblem
B > noproblem
lt and rt > i dont quite understand the value but seems not right
lb and rb > seems right(BTN_TL and BTN_TR)
the big x key > noproblem
sticks seems working right

oops i may made a mistake, I forgot that I've set the environmental variable mentioned on the github.io page
will this matter?
[image]

kakra, 2021-01-10:
> oops i may made a mistake, I forgot that I've set the environmental variable mentioned on the github.io page
> will this matter?

For SDL games: Yes, it does matter. For `evtest` and `jstest`: no. But you should remove it, should not be needed any longer.

kakra, 2021-01-10:
> thank you two!
> worked this time, but still wrong mapping
> back > not working
> share > BTN_SELECT (value 90007)
> start > BTN_START (value 90008)

Please post the dmesg lines that appeared right after reloading the driver.

> Y > BTN_WEST (90004)
> X > BTN_NORTH (90003)

That's correct, `BTN_WEST` is an alias for `BTN_Y` in the kernel (that's how Nintendo set up their original controllers). Games actually use `BTN_Y` to detect the Y button - so this is expected to work correctly.

> lt and rt > i dont quite understand the value but seems not right

It will eventually generate two values, one for the actual analog axis (`ABS_Z` or `ABS_RZ`), and one for the combined axis `ABS_MISC`. Code is the axis number, value is how far the trigger is pressed.

kakra, 2021-01-10:
@TTTPOB Please connect the controller, run `btmon`, then press SELECT, SHARE, BACK - do not touch the triggers or the sticks as it will clutter the log. Post the result and tell us which button generated which lines.

TTTPOB, 2021-01-11:
is there anyway to reduce the output? I can figure out which should be the line when I pressed the key

kakra, 2021-01-11:
@TTTPOB Apparently not. If you do not touch the sticks or triggers, it shouldn't generate too much noise. We are interested in the packets starting with `a1 01` so you could probably `btmon | grep -A4 "a1 01"` or something similar.

Usually, when I do that, I copy&paste the lines manually from a running `btmon`, scrolling to the position where values changed between press and release of a button.

TTTPOB, 2021-01-14:
[btmon.log](https://github.com/atar-axis/xpadneo/files/5814912/btmon.log)
sorry about the late reply, I was quite busy these days

kakra, 2021-01-14:
Ah okay, I forgot this is Bluetooth LE and that the packets look different.

Your log seems to miss the release events for each button but I think we can work with it: It looks like we can identify the bit for each button.

kakra, 2021-01-14:
Could you add the same log for buttons A, B, X, and Y? I need a reference for the bit offsets.

kakra, 2021-01-14:
From what I can see in the logs you provided, your buttons should work correctly with the fix I proposed in PR #269.

TTTPOB, 2021-01-15:
hmmm.. i think i have to test this a little later, my arch installation just broke

kakra, 2021-01-19:
@TTTPOB Are you able to test whether #269 fixes your issue?

kakra, 2021-01-22:
I think this will be fixed with the pull request. Closing. Feel free to re-open if the problem persists.

closed 2021-01-22 by kakra