atar-axis/xpadneo #270: The Pad is Still Flashing fast

drkshw, 2021-01-17:
## Version of xpadneo
0.9

## Severity / Impact

* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 

## Describe the bug

the pad does not connect correctly, I've followed the steps in the guide, bluetoothctl says that is paired and connected and actually the pad rumbles but the light is still flashing fast and is not working 
## Steps to Reproduce

on manjaro distro install the driver then follow the guide... (aorus b450 pro wifi 1.0, bt integrated) 
## Expected behavior

the light stops flashing and the pad is working
## Screenshots/Gifs

## System information

```console
# uname -a
Linux drk-Mainframe 5.10.2-2-MANJARO #1 SMP PREEMPT Tue Dec 22 08:14:42 UTC 2020 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
bash: xxd: command not found
4294967295 0

```

## Controller and Bluetooth information
Index list with 1 item
hci0:   Primary controller
        addr DC:1B:A1:xx:xx:xx version 10 manufacturer 2 class 0x3c0104
        supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr hs le advertising secure-conn debug-keys privacy configuration static-addr phy-configuration wide-band-speech 
        current settings: powered connectable discoverable bondable ssp br/edr le secure-conn 
        name drk-Mainframe
        short name 
hci0:   Configuration options
        supported options: public-address 
        missing options:

lights0123, 2021-01-17:
#239, #259

kakra, 2021-01-20:
This may be a duplicate as @lights0123 pointed out. But just FTR, did you disable ERTM?
```
cat /sys/module/bluetooth/parameters/disable_ertm
```

It should say `Y`, if not, write `Y` to it, then retry pairing the controller. You may need to remove it from the Bluetooth device database first before you retry.

drkshw, 2021-01-20:
The ERTM was and is disabled... (it say yes)
this is what I've done untill now following the links posted by @lights0123 :
i've sent in bluetoothctl first trust and then pair and the result is that the controller "rumbles" but the central led is still flashing fast
I've put the privacy on (doing power off, privacy on and power on) and when I sent the pair command the result was an error in pairing and then connect=no (no rumbles)

I've pasted this in main config file
 [General]
...
JustWorksRepairing = always

[Policy]
...
ReconnectIntervals=1,1,2,3,5,8,13,21,34,55
AutoEnable=true

and nothing...
I've put the advertising flag in mgmt but is not pairing 

I'm open to any suggest and if you explain the procedures I can post here any information is required for find out the problem...

thanks.

kakra, 2021-01-20:
At this point it seems very likely that the driver for your onboard Bluetooth chip is lacking some features. Are you able to try a USB Bluetooth dongle? A cheap one would do but it should be Bluetooth 4.0 capable at least.

What gamepad model are we talking about anyways? You didn't post the output of `dmesg` which shows the driver loading.

drkshw, 2021-01-20:
I have not a dongle at the moment.... but if it can help I've connected the controller to an android 10 phone successfully

yesterday i run an update so the current system now is:
Linux drk-mainframe 5.10.7-3-MANJARO #1 SMP PREEMPT Fri Jan 15 21:11:34 UTC 2021 x86_64 GNU/Linux

[dmesg.txt](https://github.com/atar-axis/xpadneo/files/5842582/dmesg.txt)

the controller is an xbox series X | S  (blue colored)  

I haven't post the output of dmesg becouse I didn't know how to do... I hope I've done this correctly 
[dmesg.txt](https://github.com/atar-axis/xpadneo/files/5842582/dmesg.txt)

if it is not correct please talk like I am a terminal

kakra, 2021-01-20:
Okay, opening drkshw shell prompt now: :-D
```
$ dmesg -w
# Remember the current line (e.g. press enter once or twice to insert some space)
# Now open a second terminal
# Connect your controller now and watch the first terminal (where dmesg is running)
# Post the lines that are logged during connection
```

Your controller is a BLE controller, so you probably need to enable Bluetooth privacy mode as outlined in the other reports. You can add that to the Bluez configuration file.

If every prerequisite has been setup correctly, you may need to remove your controller from bluez before starting the pairing process. You can use `bluetoothctl` to do this, running the `remove` command inside it:
```
$ bluetoothctl
# Turn your controller off
> remove MAC-OF-CONTROLLER
# Turn your controller on
# Start the pairing process as per instructions
```

drkshw, 2021-01-21:
I've turned on the privacy flag by doing power off, privacy on, power on in btmgmt

this is the dmesg result

[dmesg.txt](https://github.com/atar-axis/xpadneo/files/5849492/dmesg.txt)

this is what I've done on the second terminal

sudo bluetoothctl
Agent registered
[Bluetooth Keyboard]# scan on
Discovery started
[CHG] Controller DC:1B:A1:36:0E:42 Discovering: yes
[NEW] Device C0:8A:CD:EB:8F:6D drk_tv
[NEW] Device 98:7A:14:FB:0E:02 Xbox Wireless Controller
[Bluetooth Keyboard]# pair 98:7A:14:FB:0E:02
Attempting to pair with 98:7A:14:FB:0E:02
[CHG] Device 98:7A:14:FB:0E:02 Connected: yes
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device 98:7A:14:FB:0E:02 Connected: no
[CHG] Device 98:7A:14:FB:0E:02 RSSI: -46
[CHG] Device 98:7A:14:FB:0E:02 ManufacturerData Key: 0x0006
[CHG] Device 98:7A:14:FB:0E:02 ManufacturerData Value:
  03 00 80                                         ...             
[CHG] Device 98:7A:14:FB:0E:02 ManufacturerData Key: 0x0006
[CHG] Device 98:7A:14:FB:0E:02 ManufacturerData Value:
  00                                               .               
[CHG] Device C0:8A:CD:EB:8F:6D RSSI: -74
[CHG] Device C0:8A:CD:EB:8F:6D RSSI: -66
[CHG] Device 98:7A:14:FB:0E:02 ManufacturerData Key: 0x0006
[CHG] Device 98:7A:14:FB:0E:02 ManufacturerData Value:
  03 00 80                                         ...             
[CHG] Device 98:7A:14:FB:0E:02 RSSI: -54
[CHG] Device 98:7A:14:FB:0E:02 RSSI: -45
[DEL] Device 98:7A:14:FB:0E:02 Xbox Wireless Controller

drkshw, 2021-01-21:
I've done a btmon command while attempting to pair if it could help
[btmon.txt](https://github.com/atar-axis/xpadneo/files/5849695/btmon.txt)

kakra, 2021-01-21:
There's nothing logged to dmesg, xpadneo even doesn't become active. "Authentication failed" usually means ERTM is still turned on, or privacy is missing.

Please edit `/etc/bluetooth/main.conf` to include `Privacy = device` in the `[General]` section, then restart your system and try again. As previously, you may need to `remove 98:7A:14:FB:0E:02` in bluetoothctl before attempting a new pairing.

drkshw, 2021-01-21:
Privacy is set to device in the main.conf yet. and when I send 
cat /sys/module/bluetooth/parameters/disable_ertm
it shows 
Y
 this is the btmgmt info showing privacy enabled
[mgmt]# info
Index list with 1 item
hci0:   Primary controller
        addr DC:1B:A1:36:0E:42 version 10 manufacturer 2 class 0x3c0104
        supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr hs le advertising secure-conn debug-keys privacy configuration static-addr phy-configuration wide-band-speech 
        current settings: powered bondable ssp br/edr le secure-conn privacy 
        name drk-mainframe
        short name 
hci0:   Configuration options
        supported options: public-address 
        missing options: 

is any other place where this ERTM needs to be disabled I have no clue of what is going on @-@

kakra, 2021-01-21:
It's very likely your internal Bluetooth doesn't support the exact features required by this controller, and the controller is very picky about this anyways. Your best chance seems to be trying with a dedicated USB dongle. Apparently, I cannot give any recommendation here because your controller uses BLE, and the ones I own do use traditional Bluetooth, so any recommendation may be wrong because it may not work for BLE. Other people here who successfully managed to get the BLE controller running may share their setup.

drkshw, 2021-01-23:
I've got an Update... I I bought an Asus USB-BT400 Nano Bluetooth Stick, it supports bt 4.0 and the controller now is connected and works perfectly. No Issue at all even with xpadneo. So the problem is in the driver of my internal bt I guess...

axhav, 2021-01-23:
Hi, I'm seeing similar behaviour on my end with the AX200 chip on my ASUS B550-i. One thing I would like to add is that I see some traces from bluetoothd:
`bluetoothd[584]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlik>
bluetoothd[584]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
bluetoothd[584]: profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Request attribute has encountered an unlikely err>
bluetoothd[584]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encounter>
bluetoothd[584]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encounter>
`
Running bluetoothd version 5.55
Arch 5.10.7

kakra, 2021-01-24:
If problems remain, please check the following settings in your Bluez configuration:
```
[General]
ControllerMode = dual
Privacy = device
```

witem, 2021-02-15:
@kakra I have same problem on my laptop with AX200 after success pair can't connect to xbox controller. How I can help to resolve this issue?

Some info:

- ERTM disabled
- bluez: 5.55
- 5.10.15-1-MANJARO
- dmesg empty when trying to connect
- journactl log:
```
profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Request attribute has encountered an unlikely error
profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
```

- in `bluetoothctl` spams:
```
[CHG] Device 44:16:22:9E:AC:D6 Connected: no
[CHG] Device 44:16:22:9E:AC:D6 Connected: yes
[CHG] Device 44:16:22:9E:AC:D6 Connected: no
[CHG] Device 44:16:22:9E:AC:D6 Connected: yes
[CHG] Device 44:16:22:9E:AC:D6 Connected: no
[CHG] Device 44:16:22:9E:AC:D6 Connected: yes
[CHG] Device 44:16:22:9E:AC:D6 Connected: no
[CHG] Device 44:16:22:9E:AC:D6 Connected: yes
[CHG] Device 44:16:22:9E:AC:D6 Connected: no
```
- btmon log: [link](https://pastebin.com/5LRH8yFE)

kakra, 2021-02-15:
@witem Usually, with ERTM disabled, you should get past the reconnect loop. But sometimes it may not work. If you're using `bluetoothctl` to pair the controller, remove it again, and try with the Bluetooth GUI client of your desktop instead. Otherwise, you distribution probably needs to backport the L2CAP patch: https://lore.kernel.org/linux-bluetooth/20201208172912.4352-1-hadess@hadess.net/, or you wait for kernel 5.12 which should be ready in May 2021.

Other people had success transplanting a link key from an established Windows Bluetooth connection but I don't remember a link to a proper list of instructions on how to do that.

Also, it sometimes helps to pair the controller to a Windows box (using the original dongle) first. While at it, you should also check if the latest firmware is installed using the Xbox accessories app. If you have no access to Windows but own the original dongle, there's a pretty easy way to download an official eval copy of Windows 10 from Microsoft as ISO, install it in a qemu or VirtualBox VM, enable USB passthrough for the dongle and just pair and update it.

witem, 2021-02-15:
> @witem Usually, with ERTM disabled, you should get past the reconnect loop. But sometimes it may not work. If you're using `bluetoothctl` to pair the controller, remove it again, and try with the [...]

I try many times using `bluetoothctl` and also with KDE GUI, but no success. 
I try transplanting `[LinkKey]` and also try transplanting `[LongTermKey]` also without success.
Also, I reboot to Windows connect over usb, update firmware and connect over bluetooth(not original dongle) success. And reboot to Linux, but no success with pairing.

kakra, 2021-02-15:
I think there are still some features missing in the Linux Bluetooth stack. Will it pair to an Android phone for you? (which is closest to Linux as it uses the Linux kernel but fluoride instead of bluez for Bluetooth)

kakra, 2021-02-15:
Also, I started writing in the bluetooth mailing list: https://lore.kernel.org/linux-bluetooth/CAC2ZOYuG_BBG0uisXkoTtroxwgvv+JO-yMFC2+bRQZjsvwK1qw@mail.gmail.com/T/#u

We should probably get one combination of hardware working properly first, then port over the required fixes and quirks to other drivers.

witem, 2021-02-16:
> I think there are still some features missing in the Linux Bluetooth stack. Will it pair to an Android phone for you? (which is closest to Linux as it uses the Linux kernel but fluoride instead of bluez for Bluetooth)

Yes, with Android it's pair, connect and works good

kakra, 2021-02-17:
While that test doesn't test just one thing alone (bluez vs fluoride) because your Android uses different drivers, things are still pointing towards bluez as the problem. We already covered different drivers, and except some quirks (spontaneous connection loss, laggy input), they all behave more or less the same (except some onboard Intel chips). I'm pretty sure the kernel drivers are pretty solid at this point (if you're using kernel 5.10+). So we should probably shift focus towards bluez versions.

c-x-berger, 2021-05-07:
I'm still having the "unlikely error" on an AX200 chipset (integrated into my motherboard). This is on kernel 5.12, regardless of ERTM.

kakra, 2021-05-07:
@c-x-berger Could you try with a USB Bluetooth dongle so we could rule out other factors?

axhav, 2021-05-08:
@kakra I did a test with a TP-Link dongle (and kernel 5.12):
It seems to pair correctly, but the rumble functionallity seems to be very intense and "a lot" longer than compared when wired.
```
May 08 13:33:53 mars bluetoothd[3545]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
May 08 13:33:53 mars bluetoothd[3545]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
May 08 13:33:53 mars bluetoothd[3545]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
May 08 13:33:53 mars bluetoothd[3545]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
May 08 13:33:53 mars bluetoothd[3545]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
May 08 13:33:53 mars bluetoothd[3545]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
May 08 13:33:53 mars bluetoothd[3545]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
May 08 13:33:53 mars bluetoothd[3545]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
May 08 13:33:53 mars bluetoothd[3545]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
May 08 13:33:54 mars bluetoothd[3545]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
May 08 13:33:54 mars bluetoothd[3545]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
May 08 13:33:54 mars kernel: xpadneo 0005:045E:0B13.0045: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x02E0)
May 08 13:33:54 mars kernel: xpadneo 0005:045E:0B13.0045: working around wrong SDL2 mappings (changed version from 0x00000505 to 0x00000903)
May 08 13:33:54 mars kernel: xpadneo 0005:045E:0B13.0045: report descriptor size: 283 bytes
May 08 13:33:54 mars kernel: xpadneo 0005:045E:0B13.0045: fixing up Rx axis
May 08 13:33:54 mars kernel: xpadneo 0005:045E:0B13.0045: fixing up Ry axis
May 08 13:33:54 mars kernel: xpadneo 0005:045E:0B13.0045: fixing up Z axis
May 08 13:33:54 mars kernel: xpadneo 0005:045E:0B13.0045: fixing up Rz axis
May 08 13:33:54 mars kernel: xpadneo 0005:045E:0B13.0045: fixing up button mapping
May 08 13:33:54 mars kernel: xpadneo 0005:045E:0B13.0045: gamepad detected
May 08 13:33:54 mars kernel: xpadneo 0005:045E:0B13.0045: enabling compliance with Linux Gamepad Specification
May 08 13:33:54 mars kernel: input: Xbox Wireless Controller 987A as /devices/virtual/misc/uhid/0005:045E:0B13.0045/input/input119
May 08 13:33:54 mars kernel: xpadneo 0005:045E:0B13.0045: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller 987A] on 00:1a:7d:da:71:15
May 08 13:33:54 mars kernel: xpadneo 0005:045E:0B13.0045: controller quirks: 0x00000050
May 08 13:33:54 mars kernel: xpadneo xpadneo_welcome_rumble start
May 08 13:33:55 mars kernel: xpadneo xpadneo_welcome_rumble took 990ms
May 08 13:33:55 mars kernel: xpadneo 0005:045E:0B13.0045: Xbox Wireless Controller 987A [98:7a:14:f5:f1:60] connected
May 08 13:33:55 mars kernel: input: Microsoft X-Box 360 pad as /devices/virtual/input/input120
May 08 13:33:59 mars kernel: xpadneo 0005:045E:0B13.0045: consumer controls not detected
```
AX200:
```
May 08 13:57:46 mars kernel: Bluetooth: hci0: link tx timeout
May 08 13:57:48 mars kernel: xpadneo 0005:045E:0B13.0046: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x02E0)
May 08 13:57:48 mars kernel: xpadneo 0005:045E:0B13.0046: working around wrong SDL2 mappings (changed version from 0x00000505 to 0x00000903)
May 08 13:57:48 mars kernel: xpadneo 0005:045E:0B13.0046: report descriptor size: 283 bytes
May 08 13:57:48 mars kernel: xpadneo 0005:045E:0B13.0046: fixing up Rx axis
May 08 13:57:48 mars kernel: xpadneo 0005:045E:0B13.0046: fixing up Ry axis
May 08 13:57:48 mars kernel: xpadneo 0005:045E:0B13.0046: fixing up Z axis
May 08 13:57:48 mars kernel: xpadneo 0005:045E:0B13.0046: fixing up Rz axis
May 08 13:57:48 mars kernel: xpadneo 0005:045E:0B13.0046: fixing up button mapping
May 08 13:57:48 mars kernel: xpadneo 0005:045E:0B13.0046: gamepad detected
May 08 13:57:48 mars kernel: xpadneo 0005:045E:0B13.0046: enabling compliance with Linux Gamepad Specification
May 08 13:57:48 mars kernel: input: Xbox Wireless Controller 987A as /devices/virtual/misc/uhid/0005:045E:0B13.0046/input/input121
May 08 13:57:48 mars kernel: xpadneo 0005:045E:0B13.0046: input,hidraw7: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller 987A] on 08:5b:d6:58:1f:f0
May 08 13:57:48 mars kernel: xpadneo 0005:045E:0B13.0046: controller quirks: 0x00000050
May 08 13:57:48 mars kernel: xpadneo xpadneo_welcome_rumble start
May 08 13:57:49 mars kernel: xpadneo xpadneo_welcome_rumble took 990ms
May 08 13:57:49 mars kernel: xpadneo 0005:045E:0B13.0046: Xbox Wireless Controller 987A [98:7a:14:f5:f1:60] connected
May 08 13:57:50 mars kernel: input: Microsoft X-Box 360 pad as /devices/virtual/input/input122

##### Pairing flashing stops and "normal" flashing start  ################

May 08 13:59:23 mars kernel: xpadneo 0005:045E:0B13.0047: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x02E0)
May 08 13:59:23 mars kernel: xpadneo 0005:045E:0B13.0047: working around wrong SDL2 mappings (changed version from 0x00000505 to 0x00000903)
May 08 13:59:23 mars kernel: xpadneo 0005:045E:0B13.0047: report descriptor size: 283 bytes
May 08 13:59:23 mars kernel: xpadneo 0005:045E:0B13.0047: fixing up Rx axis
[... cut ...]
May 08 14:00:05 mars bluetoothd[3545]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
May 08 14:00:05 mars bluetoothd[3545]: profiles/input/hog-lib.c:output_written_cb() Write output report failed: Request attribute has encountered an unlikely error
May 08 14:00:06 mars kernel: xpadneo xpadneo_welcome_rumble took 990ms
May 08 14:00:06 mars kernel: xpadneo 0005:045E:0B13.0048: Xbox Wireless Controller 987A [98:7a:14:f5:f1:60] connected

```

kakra, 2021-05-08:
In wired mode, you're not using xpadneo as the driver. Which driver are you using? It should not be too different for rumble. Which software did you use to compare it?

There's a difference with some drivers which allow sending too high rumble strengths to the motors (0-255) while only 0-100 is allowed, and compensate for that by dividing rumble power by 4 (which ends up with 0-63), so xpadneo would actually be more correct here. But timing should be the same. However, the other driver may not use ff-memless which could make a difference. But without knowing about the other driver you're using for USB, this is pure speculation.

axhav, 2021-05-08:
Xpad driver. Actually just ran another test with the controller wired, There was no rumble at all. I was doing a quick test with the game Fury via Steam and Proton.

I just did a full retest again: So I might actually have missinterpreted the rumble, its the impulse triggers (wasn't aware that that was a thing, so thats my bad) that are acting wild. Maybe the Xpad driver doesn't support that feature and thats why there is no feedback when running wired. 

Retested with https://gamepad-tester.com/ both wireless and wired, rumble is equal as far as I can tell.

kakra, 2021-05-08:
Yes, xpad doesn't support the impulse triggers. Neither does the Linux kernel. However, xpadneo uses a trick to make them work by using the higher amplitude of both main rumble motors, and then applying it to the impulse triggers based on how far those are pressed currently. This works quite well for many games, and I actually think it works very well for Cyberpunk 77 while driving the car.

ryannathans, 2021-05-17:
I have ax200 and xbox series x controller on kernel 5.12. I get the rumble as described but the controller keeps blinking. Works perfectly fine in Windows 10 on same PC. Gigabyte x570 Aorus Master motherboard. Is there anything you need to assist debugging?

kakra, 2021-05-17:
From kernel 5.12, ERTM should be enabled (thus `disable_ertm=N`).

But besides that, the Ax200 chipset seems to have several problems in Linux, not only with the Xbox gamepads. I think the Ax200 driver is still under heavy development.

For the CSR dongles: There seems to exists a lot of clone/fake chipsets that don't work that well. But you can try a few things: Upgrade bluez to the latest version, enable ERTM in kernel 5.12, disable before 5.12, try to get a CSR dongle from a well known vendor (mine is from TP-LINK, it works fine, model BT-400), pair the controller to your mobile phone, then try to pair it with Linux again, use a Windows box to upgrade the controller firmware to its latest version (Xbox Accessories app).

ryannathans, 2021-05-25:
Okay I'm retarded, I was not pressing the pair button on the controller. For some reason it's not necessary in Windows but is required here otherwise all hell breaks loose!

Works perfectly so far, tested in steam games natively like super meat boy and even with proton like fallout 76, button mappings are correct.

For reference my kernel is 5.12.6-xanmod1 and I am on Pop!_OS with AX200 chipset.

Maybe the website should say AX200 works fine now? Presently it suggests it doesn't work at https://atar-axis.github.io/xpadneo/

kakra, 2021-05-25:
Yeah, there's a problem with the HCI event ordering: https://github.com/bluez/bluez/issues/127

One can work around this by putting the gamepad into pairing mode (hold connect), then manually connect from the Bluetooth GUI applet.

ryannathans, 2021-05-25:
@kakra Seems this bug doesn't exist for me, maybe it has been fixed? I can turn my controller off and on repeatedly after initial pairing without needing to put it in pairing mode. The controller connects rapidly and vibrates every time. For the initial pairing I needed to press pair and then I paired it in bluetoothctl as the instructions specified but for subsequent reconnects I haven't even needed to click anything.

afiestas, 2021-06-08:
I needed to update the firmware (by using the xbox app thingy from Microsoft). After that everything seemed to work great.

Kernel: 5.12.9-arch1-1
Xpadneo:  f17a4e2f080deb4
Bluez 5.58

afiestas, 2021-06-09:
After using the controller (1914)  for a while, I can confirm that everything seems to work as expected

- Pairing from Plasma and CLI works great
- Long press in the X button (the one at the center) makes the controller reconnect
- After a while of not using it, it is turning itself off

If anybody needs me to test anything else just tell me.

kakra, 2021-06-27:
This is most likely a duplicate of #297 and isn't related to AX200 at all.

closed 2021-06-27 by kakra

mhalano, 2021-11-21:
What was the model of Bluetooth card? AX200? I couldn't find this information, I just found the start of the OUI and it just inform is Intel.

acuteaura, 2024-01-11:
I can provide a data point:

Did not work on my AX200 (`Bus 003 Device 004: ID 8087:0029 Intel Corp. AX200 Bluetooth`), reconnect loops so hard it made my touchpad stuttery (either due to Plasma or actual interrupts). Same result when the controller is restarted normally or directly into pairing mode. 

NixOS unstable with kernel 6.6.10, which as of now has xpadneo 0.9.5 and bluez 5.70.

Plugged in a UGREEN Realtek radio (`Bus 001 Device 004: ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio`) and disabled the AX200 via KDE control panel; controller works fine.

ryannathans, 2024-01-11:
Odd, Pop!_OS 22.04 with kernel 6.6.6 works fine with xpadneo and `Bus 003 Device 005: ID 8087:0029 Intel Corp. AX200 Bluetooth` and has done for years. When did your controller last get a firmware update?

acuteaura, 2024-01-11:
It's a Starfield Edition controller, so I'd be very surprised if the firmware is older than the last post in this thread, but I'll check tomorrow!