atar-axis/xpadneo #279: Connect failing with XBox X|S controller on RetroPie with RPi4

gramster, 2021-03-26:
## Version of xpadneo

I cloned the repo about 30 minutes ago, so I guess 0.9

## Severity / Impact

* [ ] I've read the docs and the bug reporting instructions
* [x] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the bug

I am using an Xbox X|S controller, the new pulse red color one (https://news.xbox.com/en-us/2021/01/11/introducing-new-xbox-wireless-controller-pulse-red/).

When going through the connection process, the last step failed:

```
[bluetooth]# pair 44:16:22:C4:AC:C1
Attempting to pair with 44:16:22:C4:AC:C1
[CHG] Device 44:16:22:C4:AC:C1 Connected: yes
Request authorization
[agent] Accept pairing (yes/no): yes
[CHG] Device 44:16:22:C4:AC:C1 Connected: no
[CHG] Device 44:16:22:C4:AC:C1 ManufacturerData Key: 0x0006
[CHG] Device 44:16:22:C4:AC:C1 ManufacturerData Value:
  00                                               .

[bluetooth]# trust 44:16:22:C4:AC:C1
[CHG] Device 44:16:22:C4:AC:C1 Trusted: yes
Changing 44:16:22:C4:AC:C1 trust succeeded

[bluetooth]# connect 44:16:22:C4:AC:C1
Attempting to connect to 44:16:22:C4:AC:C1
Failed to connect: org.bluez.Error.Failed
```

## Steps to Reproduce

As described above. I cloned the repo, ran ./install.sh:

```
* copying module to '/usr/src/hid-xpadneo-v0.9-44-g58f234a'
* replacing version string if necessary
* adding module to DKMS

Creating symlink /var/lib/dkms/hid-xpadneo/v0.9-44-g58f234a/source ->
                 /usr/src/hid-xpadneo-v0.9-44-g58f234a

DKMS: add completed.
* installing module (using DKMS)

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.10.17-v7l+ -C /lib/modules/5.10.17-v7l+/build M=/var/lib/dkms/hid-xpadneo/v0.9-44-g58f234a/build/src modules....
cleaning build area...

DKMS: build completed.

hid-xpadneo.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.17-v7l+/kernel/drivers/hid/

Running the post_install script:
Disabling ERTM permanently (requires reboot)...
HINT: If you want to prevent this in the future, run 'ln -snf /dev/null /etc/modprobe.d/99-xpadneo-bluetooth.conf'.
Disabling ERTM instantly...
Installing modalias database...
Installing udev rules...
Reloading udev...

depmod...

Warning: Unable to find an initial ram disk that I know how to handle.
Will not try to make an initrd.

DKMS: install completed.
```

I rebooted, and went through the pairing process.

The last few lines after the connect failed in the kernel log are:

```
[   18.393803] Bluetooth: hci0: command 0x1003 tx timeout
[   20.064436] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   20.473794] Bluetooth: hci0: command 0x1001 tx timeout
[   22.553814] Bluetooth: hci0: command 0x1009 tx timeout
```

## Expected behavior

Connect step succeeds.

## Screenshots/Gifs

## System information

```console
# uname -a
Linux retropie 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0
```

(Note: after seeing the above, I tried running install.sh again, even though it did seem to be successful the first time, but I get a `already installed!` output).

## Controller and Bluetooth information

I get no output, presumably because it is not connecting.

```
[    1.481335] hid: raw HID events driver (C) Jiri Kosina
[    1.481523] usbcore: registered new interface driver usbhid
[    1.481538] usbhid: USB HID core driver
[    2.499493] input:   Mini Keyboard as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:1997:2433.0001/input/input0
[    2.564289] hid-generic 0003:1997:2433.0001: input,hidraw0: USB HID v1.01 Keyboard [  Mini Keyboard] on usb-0000:01:00.0-1.3/input0
[    2.571496] input:   Mini Keyboard Mouse as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:1997:2433.0002/input/input1
[    2.572142] input:   Mini Keyboard System Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:1997:2433.0002/input/input2
[    2.633516] input:   Mini Keyboard Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:1997:2433.0002/input/input3
[    2.633852] hid-generic 0003:1997:2433.0002: input,hidraw1: USB HID v1.01 Mouse [  Mini Keyboard] on usb-0000:01:00.0-1.3/input1
[    3.636007] input: Logitech Inc. Logitech USB Headset H340 Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.1/1-1.4.1:1.3/0003:046D:0A38.0003/input/input4
[    3.703514] input: Logitech Inc. Logitech USB Headset H340 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.1/1-1.4.1:1.3/0003:046D:0A38.0003/input/input5
[    3.704319] hid-generic 0003:046D:0A38.0003: input,hiddev96,hidraw2: USB HID v1.11 Device [Logitech Inc. Logitech USB Headset H340] on usb-0000:01:00.0-1.4.1/input3
[    4.656683] input: Microsoft Microsoft Pro Intellimouse Mouse as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.2/1-1.4.2.1/1-1.4.2.1:1.0/0003:045E:082A.0004/input/input6
[    4.659063] hid-generic 0003:045E:082A.0004: input,hiddev97,hidraw3: USB HID v1.11 Mouse [Microsoft Microsoft Pro Intellimouse] on usb-0000:01:00.0-1.4.2.1/input0
[    4.671876] input: Microsoft Microsoft Pro Intellimouse Keyboard as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.2/1-1.4.2.1/1-1.4.2.1:1.1/0003:045E:082A.0005/input/input7
[    4.734631] input: Microsoft Microsoft Pro Intellimouse Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.2/1-1.4.2.1/1-1.4.2.1:1.1/0003:045E:082A.0005/input/input8
[    4.735009] input: Microsoft Microsoft Pro Intellimouse as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.2/1-1.4.2.1/1-1.4.2.1:1.1/0003:045E:082A.0005/input/input9
[    4.735378] input: Microsoft Microsoft Pro Intellimouse as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.2/1-1.4.2.1/1-1.4.2.1:1.1/0003:045E:082A.0005/input/input10
[    4.735989] hid-generic 0003:045E:082A.0005: input,hiddev98,hidraw4: USB HID v1.11 Keyboard [Microsoft Microsoft Pro Intellimouse] on usb-0000:01:00.0-1.4.2.1/input1
[    5.980516] input: Yubico YubiKey OTP+FIDO+CCID as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.2/1-1.4.2.4/1-1.4.2.4.3/1-1.4.2.4.3:1.0/0003:1050:0407.0008/input/input12
[    6.044438] hid-generic 0003:1050:0407.0008: input,hidraw5: USB HID v1.10 Keyboard [Yubico YubiKey OTP+FIDO+CCID] on usb-0000:01:00.0-1.4.2.4.3/input0
[    6.051801] hid-generic 0003:1050:0407.0009: hiddev99,hidraw6: USB HID v1.10 Device [Yubico YubiKey OTP+FIDO+CCID] on usb-0000:01:00.0-1.4.2.4.3/input1
[    6.999995] input: Apple Inc. Apple Keyboard as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.3/1-1.4.3.1/1-1.4.3.1:1.0/0003:05AC:024F.0006/input/input13
[    7.064509] apple 0003:05AC:024F.0006: input,hidraw7: USB HID v1.11 Keyboard [Apple Inc. Apple Keyboard] on usb-0000:01:00.0-1.4.3.1/input0
[    7.066595] input: Apple Inc. Apple Keyboard as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.3/1-1.4.3.1/1-1.4.3.1:1.1/0003:05AC:024F.0007/input/input14
[    7.143898] apple 0003:05AC:024F.0007: input,hidraw8: USB HID v1.11 Device [Apple Inc. Apple Keyboard] on usb-0000:01:00.0-1.4.3.1/input1
```

No idea which one this is; there is no explicit dongle; I am using the BT on the RPi4 board.

## Additional context

kakra, 2021-03-26:
This is a problem with the bluetooth kernel driver and/or the bluez daemon, the driver cannot fix it apparently. Please try updating to a newer kernel and latest bluez version.

kakra, 2021-03-26:
BTW: Kernel 5.12+ will have a patch for Bluetooth L2CAP which enables using ERTM. It may be needed to properly pair this controller. If you're able to patch and compile your own kernel, there's a patch in the misc folder. It applies to any kernel version I know of back to 4.15. Also, there are commits in my patch-queue which remove the ERTM work-arounds from xpadneo which you may need then: https://github.com/kakra/xpadneo/commits/queue/for-0.10

Please note: My fork of the repo doesn't have any tags, so instead of cloning it to it's own directory, add it as a secondary remote using `git remote add ...` then `git remote update` and checkout the branch. Otherwise it may not properly build.

gramster, 2021-03-27:
Thanks Kakra; I will probably wait for the kernel update and hopefully your changes will be upstreamed by then; I have a wired controller I can use so its not urgent.