atar-axis/xpadneo #280: Divinity : Original Sin native Linux does not recognize the XBox gamepad

Odelpasso, 2021-03-27:
## Version of xpadneo
git version 58f234a

## Describe the bug
My XBox Series  gamepad works great with xpadneo module with Windows games + Wine.
But with D:OS native linux game, it does not recognize the gamepad + bluetooth.

## Steps to Reproduce
Pair and connect the gamepad, launch the game. Go to Options -> Game and you can see that the Controler option is grey.

## Expected behavior
I hope something can be found^^

## Screenshots/Gifs
"enable controler" = "activer la manette" in French
[image]

## System information

```console
# uname -a
Linux debian-laptop 5.11.10 #1 SMP Sat Mar 27 18:52:01 CET 2021 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363
```

## Controller and Bluetooth information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/6216203/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/6216204/xpadneo-dmesg.txt)

Bus 003 Device 003: ID 1358:c123 Realtek Bluetooth Radio

## Additional context
[ds4drv](https://github.com/chrippa/ds4drv) is an userspace driver which allow to use the DS4 pad as XBox 360 controller.
There is an other workaround, to use "xboxdrv" driver to emulate the X360 gamepad.
Maybe something can be done in this module to emulate X360 gamepad ?

kakra, 2021-03-27:
Could you run `evtest` with the ds4drv: It lists the name of the devices. I suspect that this game only looks for the name of a device to identify it.

Odelpasso, 2021-03-27:
```console
# evtest
/dev/input/event12:	Sony Computer Entertainment Wireless Controller
```

kakra, 2021-03-27:
Do you run the game through Steam? If yes, please try disabling or enabling Steam Input for Xbox controllers (it can be switched in Big Screen mode, or per game individually). This may have an impact on how controllers are detected by games. Since it doesn't use an Xbox name for your Sony controller, my next best guess is that you have Playstation controller support enabled in Steam Input but you don't have for Xbox controllers.

Odelpasso, 2021-03-27:
No, I don’t use Steam. I have the game from GoG. Steam is not installed on my system.

ds4drv use the PS4 pad but emulate a X360 gamepad. For others native/Wine games, if I use ds4drv it’s like i use a real XBox gamepad.

kakra, 2021-03-27:
Okay, I've looked at the ds4 driver: It doesn't do anything specific to mimic a X360 pad, it's just a HID standard driver. So I suspect the culprit being somewhere at the upper layers.

Do you have anything in the `$HOME/.local/share/gamecontrollerdb.local.txt` file?

Could you try running the game with `SDL_JOYSTICK_HIDAPI=0` (environment setting)?

kakra, 2021-03-27:
BTW: There seems to be some process stealing the device from us and exposing it via uinput:
```
[ 1756.090628] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.000A/input/input33
```

Comes anything to mind you might be running that implements user space drivers for gamepads?

kakra, 2021-03-27:
> Okay, I've looked at the ds4 driver: It doesn't do anything specific to mimic a X360 pad,

Ah wait, I found this:
```python
create_mapping(
    "xpad", "Microsoft X-Box 360 pad",
    # Bus type,      vendor, product, version
    ecodes.BUS_USB,  1118,   654,     272,
    # Axes
    {
        "ABS_X":  "left_analog_x",
        "ABS_Y":  "left_analog_y",
        "ABS_RX": "right_analog_x",
        "ABS_RY": "right_analog_y",
        "ABS_Z":  "l2_analog",
        "ABS_RZ": "r2_analog"
    },
```

I'll look into it, possibly using the proper PID is what would work around your problem, or the name. Which emulation profile are you using? xpad or xpad_wireless?

kakra, 2021-03-27:
Could you try this hack:
```patch
diff --git a/hid-xpadneo/src/hid-xpadneo.c b/hid-xpadneo/src/hid-xpadneo.c
index 8876cea..306bb8e 100644
--- a/hid-xpadneo/src/hid-xpadneo.c
+++ b/hid-xpadneo/src/hid-xpadneo.c
@@ -922,6 +922,9 @@ static int xpadneo_input_configured(struct hid_device *hdev, struct hid_input *h
                break;
        }

+       xdata->gamepad->id.product = 0x028E;
+       xdata->gamepad->id.version = 0x00000110;
+
        if (hdev->product != xdata->gamepad->id.product)
                hid_info(hdev,
                         "pretending XB1S Windows wireless mode "
```

Odelpasso, 2021-03-27:
You're right, add these three lines fix the issue but the mapping is completely broken.
I compare with the Win version + Wine. Like i said, this hack is not needed but the mapping problem is the same.

I tried xpad_neo without the hack + `SDL_JOYSTICK_HIDAPI=0`, without effect.

kakra, 2021-03-27:
In my patch queue, I've added a module parameter `xpad_emulation=1` which may fix this for you. To properly use that, use `git remote add kakra https://github.com/kakra/xpadneo && git remote update` and checkout my queue branch. Do not clone that to a separate repository because it will break versioning. You may add a worktree, tho, after adding the remote: `git worktree add ../xpadneo-kakra kakra/queue/for-0.10` should do the trick. But that's a rebase-branch, you cannot properly pull from it for future updates (use `pull --rebase` instead).

Odelpasso, 2021-03-28:
Sorry but how to use `xpad_emulation=1` module parameter ?
It's possible to use it only for Divinity : OS and not for others games ?

Odelpasso, 2021-03-28:
very good job, the game recognize correctly the gamepad and the mapping is not broken :-)

kakra, 2021-03-28:
I moved this to a pull request so you can better follow the updates or easily extract a patch for local package installation:

* https://github.com/atar-axis/xpadneo/pull/283
* https://github.com/atar-axis/xpadneo/pull/283.patch

kakra, 2022-05-30:
@Kzimir Can you retry with the latest version without applying the Xbox360 emulation hack? We changed some quirks and fixups which may affect how such games handle and detect the controller.

kakra, 2022-09-17:
Closing as probably fixed in latest master.

closed 2022-09-17 by kakra