atar-axis/xpadneo #287: [DEL] RPI-4 Can't pair/connect my controller.

whizps, 2021-04-28:
## Version of xpadneo
Latest one from when im writing this, nothin will have changed from now on.

## Severity / Impact
5/5

## Describe the bug
I have a new RPI4 and a brand new bluetooth controller "Xbox One" and when i try to pair or rather after i get these error messages and my entire bluetoothctl freezes and i have to reboot the PI for it to work again. And it simply wont let me pass that stage.

## Steps to Reproduce
Bluetoothctl -> scan on -> find the device -> pair -> and it all freezes every time.

## Expected behavior
I Expected it to pair and then then just work!

## Screenshots/Gifs
[image]

## System information

```console
Linux HomeBerry 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

```

```console
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory

```

## Controller and Bluetooth information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/6395122/xpadneo-btmon.txt)

```console
[    1.418636] hid: raw HID events driver (C) Jiri Kosina
[    1.418826] usbcore: registered new interface driver usbhid
[    1.418840] usbhid: USB HID core driver

```

```console
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Dont have a dongle, its a bluetooth controller but ^

```

xx bluetooth controller.

## Additional context
As stated its a bluetooth controller so i cant really provide you with a dongle since bluetooth is built in to the device.

kakra, 2021-04-29:
I may need to get an RPi myself to test this but I think it would mostly miss the L2CAP kernel patch with enabled ERTM to somewhat reliably pair the device.

kakra, 2021-04-29:
Could you please try turning scanning off again before running the actual pair command?

whizps, 2021-04-29:
> 
> 
> Could you please try turning scanning off again before running the actual pair command?

Hi Kakra!

I already tried that, it gives me the same output, and worth noting is that after that error appears and i restart the RPI4 it just starts pending between connected on/off as a few users before me.

But yeah that's after it freezes and i have to restart it.

kakra, 2021-04-29:
You mean the whole RPi freezes? Keyboard attached by USB?

whizps, 2021-04-29:
> 
> 
> You mean the whole RPi freezes? Keyboard attached by USB?

Sorry! No, the bluetooth module freezes, i can exit it but when i go back to bluetoothctl its just stuck there until i restart the module/service or restart the Pi, but then when i try the same procedure again, the same thing happends

kakra, 2021-04-29:
Then there's probably not much we can do from here now and it's a third party problem: No matter if the device pairs / connects / loops / whatever, the kernel Bluetooth module should not stop doing anything. This should probably reported upstream to either https://github.com/bluez/bluetooth-next or https://github.com/bluez/bluez. If you report the problem, you probably need to post your full `uname -a` and a `btmon` log of the complete pairing process up to the point when the module stops to work.

But even if that particular problem is fixed, the controller is still difficult to pair, especially with some Bluetooth drivers. This is nothing that xpadneo could fix because the Bluetooth driver stack is completely invisible to xpadneo. We can only try to work around some general issues by setting Bluetooth module parameters (which is already questionable in itself as we could modify behavior of other devices that way).

> EDIT: And no, no keyboard attatched.

I wanted to ensure that the freeze isn't just seemingly a system freeze but maybe just a problem with other Bluetooth-connected input devices you may be using which become confused by this action.

I already posted an issue there about pairing/connection issues: https://github.com/bluez/bluez/issues/127

There's also another report which is why I suggested turning off scanning before running the pair command: https://github.com/bluez/bluez/issues/123

Sadly, both issues there do not receive much attention, neither there nor on the official mailing list. This may be an issue with too few people involved in development, and priorities are just towards more prominent problems. But at least some report is displayed somewhere, and some time in the future, someone may look at it, even if they don't comment on it.

If possible, try to get a newer kernel for the RPi first, and maybe also update bluez and linux-firmware (or if that has a special RPi-related firmware package, update that, too). Usually, Bluez and the kernel module for Bluetooth have a constant inflow of patches and fixes.

whizps, 2021-04-29:
@kakra 

Do you think it would help or even is doable if i get a usb bluetooth adapter? Or would the base problem still be the same?

kakra, 2021-04-29:
You may have a better chance with a USB Bluetooth dongle (they cost like 8-12€ here in Europe, so it's not too bad if it doesn't work). I suggest using a BT 4.0 or higher dongle. This one works for me: https://www.amazon.de/gp/product/B07NQ5YGDW/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

If you're going that solution, I strongly recommend turning onboard Bluetooth RF off, so it doesn't interfere.

whizps, 2021-04-29:
@kakra 

I got one actually, i know it's BT4.0 atleast, so, if i smack that one in the RPI, do i have to change anything or will that one automaticlly take over? or can i turn the on board BT off? "if you know"

kakra, 2021-04-29:
You can use `btmgmt` to power off single radios but I'm not sure how to select the right dongle first.

The devices paired won't automatically migrate: The Bluetooth library has a pairing database per Bluetooth radio, so you need to pair it again. You can probably switch which radio is used in `bluetoothctl` but I never tried that.

whizps, 2021-04-29:
@kakra 

So i did ALOT of troubleshooting yet i find myself back here! last straw of hope before i give up, i have actually been able to get rid of that nasty error message, but instead a new one is presented to me.

After i install your driver and try to pair it now it instead gives me
```console
Failed to pair: org.bluez.Error.AuthenticationFailed

```
And, i can't seem to find anyone who kinda has the same error, but is this a better errormessage than the previous one or a worse one 🤔

kakra, 2021-04-29:
Try `agent default` before pairing, or one of the other agents. Usually, GUI apps take that part. If it doesn't work, try pairing the controller to another Linux computer, then copy the device info file (from /var/lib/bluetooth) over from the working computer to your other computer (whole bluetooth services are stopped, then reboot).

whizps, 2021-04-29:
Don't hae another linux computer unfortenatly. But it did something else, but still the same in the end, this time it spat out connected first and then failed
```console
[bluetooth]# default-agent
No agent is registered
[bluetooth]# pair 44:16:22:72:4C:80
Attempting to pair with 44:16:22:72:4C:80
[CHG] Device 44:16:22:72:4C:80 Connected: yes
Failed to pair: org.bluez.Error.Failed
[CHG] Device 44:16:22:72:4C:80 Connected: no
```

kakra, 2021-04-30:
Okay, so you're now probably at the reconnect loop which could be fixed with the L2CAP patch in kernel 5.12.

kakra, 2021-04-30:
Please run `bluetoothctl info 44:16:22:72:4C:80`. Does it say "Trusted: yes" and "Paired: yes"?

If yes: Could you recreate a new `btmon` log from this new state, i.e. try to connect the controller while btmon is logging?

If no: Try `agent DisplayYesNo` or `agent NoInputNoOutput` - but I'm just guessing here. Background: I was consequently more successful pairing the controller with the KDE Plasma Bluetooth GUI instead of using `bluetoothctl` alone - and the GUI has a registered agent to ask for pairing permissions.

ByteRaper, 2021-05-01:
Basically https://github.com/atar-axis/xpadneo/issues/259#issuecomment-826150567 in #259 
Got myself a 8BitDo SN30 Pro+ dont wanted to wait for a kernel patch.

kraliq233, 2021-05-14:
S

AlexandreLaborde, 2021-05-21:
I managed to connect my Xbox Series S|X controller to my Raspberry Pi 4.
I had to pair with the Pi, pair to a new device (my Windows PC in this case) and reuse the credentials for the connection between controller and my PC on my Pi.

This is how I did it:
1. Attempt to pair the controller to the Pi using `bluetoothclt`
a.	This will cause the old “Connected: no, Connected: yes” loop
b.	You need this step to create the folder with connection keys associated with the controller.
c.	Write down the address of your controller. 44:16:22:6D:77:3F In my case.
2.	Pair the controller with my Windows PC.
3.	Download PsExec from Microsoft [ PsExec - Windows Sysinternals | Microsoft Docs](https://docs.microsoft.com/en-us/sysinternals/downloads/psexec)
4.	Open a cmd as Admin and run `psexec.exe -s -i regedit`
a.	This opens the Registry Editor in a mode with more permissions.
b.	Navigate to `HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Services\BTHPORT\Parameters\Keys`
5.	Inside the Keys directory you should have a directory whose name is the address of your computer’s Bluetooth card.
6.	Inside said directory you should find one directory per paired Bluetooth device. Find the one that matches step 1.c
[image]
7.	Right-click it and Export ( I did it as txt )
8.	Open the file with the exported data it should contain this information (the aspect changes a bit depending on how you save the file)
```
[HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\BTHPORT\Parameters\Keys\302432c9aa3a\4416226d773f]
"LTK"=hex:c2,01,70,07,1c,00,16,c2,01,70,07,1c,00,17,c2,01
"KeyLength"=dword:00000010
"ERand"=hex(b):70,07,1c,00,17,c2,01,70
"EDIV"=dword:0000001b
"IRK"=hex:76,37,ed,cc,eb,ea,da,43,44,16,22,6d,77,3f,37,76
"Address"=hex(b):3f,77,6d,22,16,44,00,00
"AddressType"=dword:00000000
"CSRKInbound"=hex:76,37,3d,ac,cb,aa,9c,ef,ad,31,5e,ac,a3,8d,37,76
"InboundSignCounter"=hex(b):ff,ff,ff,ff,ff,ff,ff,ff
"CSRK"=hex:08,32,ae,0a,58,4f,1a,97,42,d2,4a,f6,e1,b2,39,26
"OutboundSignCounter"=dword:00000000
"MasterIRKStatus"=dword:00000001
"AuthReq"=dword:0000002d
```
9. Disable Bluetooth on the PC ( just to be safe) and go back to the Pi.
10. Open a terminal and type `sudo -i`
11. Navigate to `/var/lib/bluetooth/<ADDRESS_OF_PI_BLUETOOTH_CARD>`
a. After this you should see one directory per Bluetooth device plus cache identity and settings directories. Something like this
`30:59:B7:77:B3:80  44:16:22:6D:77:3F  7C:1E:52:0C:80:AB  cache  identity  settings`
12. Navigate to the one with your controller’s address. 
Here you have 2 files. attributes and info. Open the latter `nano info`
13. This is the part where we overwrite the data with the settings from the other connection.

| Windows Key  | Pi Group | Pi Key  | Notes |
| ------------- | ------------- | ------------- | ------------- |
| IRK  | IdentityResolvingKey  | Key  | uppercase no spaces no commas  |
| CSRK  | LocalSignatureKey  | Key  | uppercase no spaces no commas  |
| LTK  | LongTermKey  | Key  | uppercase no spaces no commas  |
| KeyLength  | LongTermKey  | EncSize  | Convert from hex to decimal  |
| EDIV  | LongTermKey  | EDiv  | Convert from hex to decimal  |
| ERand  | LongTermKey  | Rand  | Flip the key byte by byte before converting to decimal. Key A1B2C3D4 flips to D4C3B2A1.  |

14. Save and reboot the Pi.
15. Your Xbox Series S|X controller should now connect to you Raspberry Pi 4. 
( Controller rumbles when connected and Xbox logo stays on )

On Steam Link some of the key mappings are wrong #291

kakra, 2021-05-21:
So if transplanting the key works, this is probably another instance of https://github.com/bluez/bluez/issues/127

dolikemedo, 2021-11-17:
Hello!
@AlexandreLaborde 
Im trying your solution, but I dont have CSRK from my windows connection.
My KeyLenght and ERand are both full 0. 
What do you mean by 'Pi Key' , is that the variable's name you have to give value?
like LTK IRK CSRK should be like Key=... Key=... Key =..
Im sorry for being so stupid about this, can you provide and example, how does your edited file look like on the Pi? 
Thank you!!

kakra, 2021-11-17:
Are there any more error logs in the journal or syslog?

Rak1ta, 2021-11-19:
Hello. There was the same problem, but noticed it after updating the raspberry. I wrote my little observation on the official raspberry forum. In short, my bluetooth gamepad connects without third-party (xpadneo) packages, but with the bluez version, which is considered obsolete. If someone could explain this to me, I would be grateful. If it were not for the successful connection of the gamepad sometime ago, I would also read the forum complaining about bluez.
[>click here<](url=https://forums.raspberrypi.com/viewtopic.php?p=1939922&hilit=gamepad+connection#p1939922)

Guiorgy, 2021-11-20:
@Rak1ta link is incorrect: `Not Found`

Rak1ta, 2021-11-20:
@Guiorgy
https://forums.raspberrypi.com/viewtopic.php?t=324144

closed 2025-01-19 by github-actions[bot]