atar-axis/xpadneo #289: Is xpadneo licensed with the optional "or later" GPL license clause?

MetaDark, 2021-05-07:
I can't find any mention of the GPL3 or later license clause. This is usually indicated in license headers in source files or in the README, but I could only find licensing information in LICENSE. The LICENSE file by itself doesn't specify this since it's an optional clause.

See the difference between the GPL-3.0-or-later & GPL-3.0-only license header:
- https://spdx.org/licenses/GPL-3.0-or-later.html#licenseHeader
- https://spdx.org/licenses/GPL-3.0-only.html#licenseHeader

kakra, 2021-05-08:
I don't think this optional clause was added by @atar-axis back that time.

Also, I wonder what this means in context of maybe porting this over to the kernel at some later time.

kakra, 2021-06-23:
@kira-bruneau What would it need to change the license to GPL3+? Do I need to check with every author, or can I just change it?

atar-axis, 2021-06-23:
* The LICENSE was auto-generated back when the project started.
* The Project was originally licensed under GPL3, as indicated by GitHub, not GPL3+. Which is, as you said, optional. 
* Is there any specific use case for that? There is no GPL4 around, right?

* If this is still (partly) *my* decision (from a legally point of view): I would be fine if @kakra should decide to switch to GPL3+.

kira-bruneau, 2021-06-27:
> Is there any specific use case for that? There is no GPL-3.0-or-later around, right?

There's no specific use case yet, but most package managers treat `GPL-3.0-only` & `GPL-3.0-or-later` as separate licenses. I was mostly just curious if I was using the right license in the nixpkgs package that I maintain: https://github.com/NixOS/nixpkgs/blob/master/pkgs/os-specific/linux/xpadneo/default.nix#L35.

> What would it need to change the license to GPL3+? Do I need to check with every author, or can I just change it

> If this is still (partly) my decision (from a legally point of view): I would be fine if @kakra should decide to switch to GPL3+.

I don't know too much about the legal details, I don't think there would be any problems since you'd just be changing an optional clause. A lot of project specify this by providing a copyright notice in each source file, but it should be enough to mention the copyright header in `README.md`.

```
This program is free software: you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation, either version 3 of the License, or (at your option) any later version.
```

See https://github.com/jtojnar/nixpkgs-hammering/blob/master/explanations/unclear-gpl.md.

seancmonahan, 2021-11-09:
> I don't think this optional clause was added by @atar-axis back that time.
> 
> Also, I wonder what this means in context of maybe porting this over to the kernel at some later time.

If I'm reading the [Linux kernel licensing rules](https://www.kernel.org/doc/html/latest/process/license-rules.html) correctly, in order for `xpadneo` to be incorporated into the Linux kernel, it would need to be relicensed _down_ to GPL-2.0 or GPL-2.0-or-later.

Neither GPL-3.0+ nor GPL-3.0-only are listed as being compatible with the Linux kernel, which is licensed under GPL-2.0-only.

Kernel-compatible licenses are listed as:

* GPL-1.0-or-later
* GPL-2.0-only
* GPL-2.0-or-later
* LGPL-2.0-only
* LGPL-2.0-or-later
* LGPL-2.1-only
* LGPL-2.1-or-later

(Dual-licensed code with a more permissive license is also allowed as long as one of the licenses is from the above list.)

kakra, 2021-11-09:
Then maybe it would be good to re-license down to a kernel-compatible license, so we can keep that option open for the future. I've never done something like that. Do we need to ask all past contributors for permission? Or would it be sufficient to send an informal note to all contributors? Do we need to take any of those steps at all?

kakra, 2024-11-21:
Planned to switch to GPL-2.0-or-later with v0.10 release.