atar-axis/xpadneo #299: Not displaying battery levels using XBox Series X Controller + Microsoft dongle + xow

hamsdac, 2021-06-16:
## Version of xpadneo

0.9.r63.gf17a4e2-1
Installed via AUR (Arch Linux Package **xpadneo-dkms-git**)
## Severity / Impact

* [x] I've read the docs and the bug reporting instructions
* [ ] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [x] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the bug

I cannot see battery level, as the controller is not connected via bluetooth (according to linux) and therefore does not show up under bluetooth devices at all.

## Steps to Reproduce

- XBox Series X Wireless Controller (1914) + official USB dongle
- USB dongle connected to PC (only working on USB 2.0 port)
- linux does not see USB dongle as bluetooth device (`bluetoothctl scan on` -> No default controller available)
- using `xow-git` (AUR Package) and sending `sudo systemctl kill -s SIGUSR1 xow` enters pairing mode
- controller now works but no battery level displayed (even though `systemctl status xow` states battery level)

## Expected behavior

- battery level should should be displayed
- controller should be visible in bluetooth menu (that may not be your problem though as far as I understand)

## Screenshots/Gifs

## System information

```console
# uname -a
Linux archpc 5.12.10-arch1-1 #1 SMP PREEMPT Thu, 10 Jun 2021 16:34:50 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0
```

## Controller and Bluetooth information

Output of `sudo btmon | tee xpadneo-btmon.txt` is empty (file is empty).

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/6666124/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/6666125/xpadneo-lsusb.txt)

## Additional context

kakra, 2021-06-17:
The file is empty because xpadneo does not (yet) support the dongle, please look at the xow (or xone) project: https://github.com/medusalix/xow

closed 2021-06-17 by kakra

kakra, 2021-06-17:
BTW: xow is not able to report battery levels through upowerd, so your desktop environment won't show the battery levels. This is a known and (most probably) unsolvable technical issue. The dongle included with the controller isn't a Bluetooth dongle (it's a WiFi dongle with proprietary firmware), so it won't show up as a Bluetooth device, xpadneo cannot work with it.

hamsdac, 2021-06-17:
Thank you for the clarification.
Great work on xpadneo btw :)