atar-axis/xpadneo #302: Home and Share Buttons doesn't work at Series X|S Controller

mende1, 2021-07-06:
## Version of xpadneo

## Controller Model

* [ ] Xbox One S controller
* [ ] Xbox Elite 2 controller
* [x] Xbox Series X|S controller
* [ ] Other:

* [X] Bluetooth connection
* [ ] Xbox Dongle connection (not yet supported)
* [ ] USB cable (not yet supported)

## Installed Software

* [X] Steam Input (enabled by default via Steam Desktop client)
* [ ] Steam Link (usually via Raspberry Pi or other micro computers)
* [ ] xow (alternative driver using the Xbox dongle)
* [ ] Anti-Micro (may affect button mappings)
* [ ] netstick (shares input devices via network similar to Steam Link)
* [ ] xboxdrv (user-space gamepad driver)

## Severity / Impact

* [X] I've read the docs and the bug reporting instructions
* [X] I've applied the latest firmware update to the controller
* [X] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the Bug

### The Middle Button of Controller (The button to turn on) (the home button by some people) doesn't work, neither Share Button.
I use arch, btw, and KDE interface, when I finished my setup I connected my controller via Bluetooth, and by some way it works, including home button. But without some controller features that xpadneo can give me. So I installed xpadneo, and the Home Button stop working.
If I go to jstest-gtk or even systemsetting joystick from KDE and test my controller, it gives me that my controller only have 10 buttons, all these except home button.

Note that I've installed Steam and RetroArch before install xpadneo, and also xboxdrv package (Later uninstalled, but still doesn't working). 

## Steps to Reproduce
Install xpadneo by Github version, following every step described on instructions. Reboot my machine. Connected my Xbox Series X|S Controller via Bluetooth, and bug appeared. Home Button stop working. Share button also, but it already doesn't working before install xpadneo.

## Expected Behavior
Home and Share Button working.

## Screenshots / GIFs / Videos
[image]
(Only 10 buttons, without home and share)

[image]
(Event time with error when I press Home and Share Buttons, respectively)

## System Information

```console
# uname -a
Linux arch 5.12.14-arch1-1 #1 SMP PREEMPT Thu, 01 Jul 2021 07:26:06 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
```

## Controller and Bluetooth Information

Run `dmesg | egrep -i 'hid|input|xpadneo' | tee xpadneo-dmesg.txt`.
[Output](https://gist.github.com/mende1/ffaa999059bd3a26e10c6b31f91bbadb)

Run `lsusb` and pick the device number of your dongle.
Bus 001 Device 004: ID 8087:0a2a Intel Corp. Bluetooth wireless interface

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/6771785/xpadneo-lsusb.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/6771786/xpadneo-dmesg.txt)
[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/6771787/xpadneo-btmon.txt)
[log.txt](https://github.com/atar-axis/xpadneo/files/6771784/log.txt)

## Additional Context

kakra, 2021-07-06:
The Share button has no Linux equivalent for a gaming input button: We cannot send it to the joystick device except by faking it for a different one - which might have consequences in the future as it might conflict with other buttons added to the mapping.

The Xbox button could be sent but this confuses some games because the gamepad would have more than 10 gaming input buttons then, even MS doesn't put it into their API. Instead, it is sent as a separate device event.

xpadneo does the same: Both buttons are moved to a secondary device. There are no immediate plans on fixing this but in the future you may be able to use the yet to be implemented feature for mapping profiles to map the button to the gamepad device.

mende1, 2021-07-06:
@kakra, so, it's ok only appear 10 button on my controller? It suppose to be that... It's ok, I understand your point.
But you know some way to map the Xbox button event to I use on RetroArch or even Steam? Maybe use a different drive than udev, or some configuration, anything?

kakra, 2021-07-06:
@mende1 Yes, there is this branch currently: https://github.com/atar-axis/xpadneo/pull/283

But I didn't merge it yet because the solution seems not quite right but I'm working on it. You can use that branch for RetroArch, it may act funky with Steam.

closed 2021-07-12 by mende1

pascal260303, 2024-06-29:
I would like to have the xbox button (home button) to behave like a regular button, because I use an Xbox controller for Xbox cloud gaming and would need the button to exit a game. jstest already shows 11 buttons and button 8 which would be the xbox button is not used or at least never gets pressed when I try all buttons. So would it be possible to have the xbox button as button 8?

kakra, 2024-07-01:
@pascal260303 Use the in-kernel driver instead then. xpadneo doesn't support the funky ways of browsers implementing input support. They are manually re-doing the mapping fixups we already do in the driver. Since this is mostly the only thing we are doing, removing that would make xpadneo essentially the in-kernel driver.