atar-axis/xpadneo #304: CSR8510 A10  - BLE dongle stability issues

habakuk1979, 2021-07-18:
hello!

My name is peter and I am from Braunau /Austria. This is probably a bit off topic but: 
Since about 2 years I am working ambitiouly on connecting a GoPro to an Odroid via HTTP, UDP and **BLE** in order to get a powerful node-red controlled dashcam/balckbox for my cars (fleet control).
(please see: https://github.com/cepoon/GoProRemote, https://github.com/KonradIT/goprowifihack, https://github.com/sepp89117/GoPro-Multiple-Smart-Remote-ESP8266)

However I stumbled over your thread: https://github.com/atar-axis/xpadneo/issues/96

and figured that you are deep into debugging of BLE connections. 

Our hardware configuration (unfortunately old):

- Odroid C2 with BluetoothModule2 (CSR8510 A10)
- Ubuntu 18.04 (Kernel 3.16.85+)
- WifiModule 5A (concurrent mode (STA+Client) on hostapd 2.9)
- Bluez 5.53
- btusb 0.7
(- GoPro Hero5 balck)

However, we do suffer from an comparable annoying problem since more than a year(!). We do have an unstable BLE Connection between Dongle and Gopro. When just paired, the first connection stays up for about 30mins, but the reconnect is a problem: gets connected but after 2 - 15 sec: disconnect. Mainly because of:

#Issue 1: (btmon) Why is it deconnection after MTU request?
-----------------------------------------------------------
@ Discovering: 0x00 (6)
> HCI Event: Command Status (0x0f) plen 4                 #111 [hci0] 55.051270
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 19                 #112 [hci0] 55.626431
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 71
        Role: Master (0x00)
        Peer address type: Random (0x01)
        Peer address: C9:0F:D9:47:48:B4 (Static)
        Connection interval: 70.00 msec (0x0038)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Master clock accuracy: 0x00
@ Device Connected: C9:0F:D9:47:48:B4 (2) flags 0x0000
< ACL Data TX: Handle 71 flags 0x00 dlen 7                #113 [hci0] 55.627789
      ATT: Exchange MTU Request (0x02) len 2
        Client RX MTU: 517
> HCI Event: Disconnect Complete (0x05) plen 4            #114 [hci0] 56.035562
        Status: Success (0x00)
        Handle: 71
        Reason: Connection Failed to be Established (0x3e)
@ Device Disconnected: C9:0F:D9:47:48:B4 (2) reason 0

#issue 2: btmon: Error: Disconnect after attribute not found!?
-----------------------------------------------------------
        Data length: 23
        Flags: 0x02
          LE General Discoverable Mode
        16-bit Service UUIDs (partial): 1 entry
          GoPro, Inc. (0xfea6)
        Company: GoPro, Inc. (754)
          Data: 0101133f00b44847d90fc90f
        RSSI: -67 dBm (0xbd)
> HCI Event: LE Meta Event (0x3e) plen 35                  #77 [hci0] 15.103486
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Random (0x01)
        Address: C9:0F:D9:47:48:B4 (Static)
        Data length: 23
        Flags: 0x02
          LE General Discoverable Mode
        16-bit Service UUIDs (partial): 1 entry
          GoPro, Inc. (0xfea6)
        Company: GoPro, Inc. (754)
[... 467 lines cut ...]
> ACL Data RX: Handle 71 flags 0x02 dlen 26               #135 [hci0] 23.319981
      ATT: Read By Group Type Response (0x11) len 21
        Attribute data length: 20
        Attribute group list: 1 entry
        Handle range: 0x0042-0x0047
        UUID: Vendor specific (b5f90090-aa8d-11e3-9046-0002a5d5c51b)
< ACL Data TX: Handle 71 flags 0x00 dlen 11               #136 [hci0] 23.320349
      ATT: Read By Group Type Request (0x10) len 6
        Handle range: 0x0048-0xffff
        Attribute group type: Primary Service (0x2800)
> HCI Event: Number of Completed Packets (0x13) plen 5    #137 [hci0] 23.389838
        Num handles: 1
        Handle: 71
        Count: 1
> ACL Data RX: Handle 71 flags 0x02 dlen 9                #138 [hci0] 23.458793
      ATT: Error Response (0x01) len 4
        Read By Group Type Request (0x10)
        Handle: 0x0048
        **Error: Attribute Not Found (0x0a)**
> HCI Event: Disconnect Complete (0x05) plen 4            #139 [hci0] 24.160066
        Status: Success (0x00)
        Handle: 71
        Reason: Connection Timeout (0x08)
@ Device Disconnected: C9:0F:D9:47:48:B4 (2) reason 1
------------------------------------------------

kakra, 2021-07-19:
Servus!

This is probably addressed to @atar-axis - I'm not sure if he can still give some insight here. He's the original author of the driver and worked through the early stages of getting the controller work. But maintainership has been transferred to me since, and I didn't really dug deep into Bluetooth, or BLE for that matter.

But I collected a bunch of known problems with work-around or fixes here:
* #198 

You may want to take a look there as a starting point.

Also, there are a few reports at https://github.com/bluez/bluez/issues?q=is%3Aissue+is%3Aopen+ble about BLE, maybe even look through the complete set of issues there. There have been a lot of fixes lately to bluez, I don't think your problem is really related to the CSR A10 dongle, or its kernel driver, at least not if you're using the latest kernels (mainline or LTS 5.10). Also, I'd be guessing that you tried different dongles if you struggled for years - so that probably rules out kernel or driver issues. Thus, the people from the bluez project may most likely help you best - although it seems they are lacking some man power.

My latest experience for connection problems is instability of previous working connections, even unable to connect again when paired successfully previously. What worked for me was stopping the Bluetooth services, then purging `/var/lib/bluetooth/$DONGLE_MAC/cache` clean, rebooting, then removing the affected device and pairing it again. There may be some artifacts left in the attribute caches due to previous bugs in bluez. I think bluez has an option to disable attribute caching, maybe that helps.

All I can say is that currently, all my controllers work fine (after doing the above cleanup, with kernel 5.10.50 with L2CAP patch and bluez 5.60 on Gentoo): Xbox Series One S, Xbox Elite 2 (both classic Bluetooth), and Xbox Series X|S (BLE).

There's on issue, tho: The XB1S controller feels a bit laggy, the XBE2 controller loses connection sometimes (which may be related to intermediate scanning activity of the bluetooth service, and games do not detect that, the controller eventually times out after 30-60s), and the XBXS controller sometimes drops the connection then reconnects a few seconds later (and games immediately detect the connection loss, which may partly be due to my BLE latency settings I've put into the bluez config).

Possibly related:
* https://github.com/bluez/bluez/issues/156
* https://github.com/bluez/bluez/issues/127
* https://github.com/bluez/bluez/issues/123

habakuk1979, 2021-07-21:
@kakra 

thanks  for your input - unfortunately nothing new on that side - I hope I can get in touch with atar-axis.

habakuk1979, 2021-07-21:
@atar-axis 

Servus and hello!

I would really appreciate your input on my issue here (see above!)

Please help.

kakar said, you were possibly the right gy tio help me out on this.

atar-axis, 2021-07-22:
Hey there,
It took me a while to find the pairing issue/bug back when xpadneo started because I only know a little about the Bluetooth Specifications.

What I did back then was compiling the Bluetooth stack by myself and replacing the original module. This helped me especially to find where the problem is (I added some `printk(...)` commands). I then found out that the original module was not fully implementing a special but optional feature.

Anyway, I would suggest to talk to the Bluetooth guys at the kernel mailing list.

kakra, 2022-09-17:
Closing old Bluetooth issues, please report to the bluez project if the problem persists.

closed 2022-09-17 by kakra