atar-axis/xpadneo #307: Antimicrox not getting expected button count.

RolledR, 2021-08-07:
## Version of xpadneo

v0.9-83-g45fef91
## Controller Model

* [ ] Xbox One S controller
* [ ] Xbox Elite 2 controller
* [x] Xbox Series X|S controller
* [ ] Other:

### Connection mode

* [x] Bluetooth connection
* [ ] Xbox Dongle connection (not yet supported)
* [x] USB cable (not yet supported)

## Installed Software

* [ ] Steam Input (enabled by default via Steam Desktop client)
* [ ] Steam Link (usually via Raspberry Pi or other micro computers)
* [ ] xow (alternative driver using the Xbox dongle)
* [x] Anti-Micro (may affect button mappings)
* [ ] netstick (shares input devices via network similar to Steam Link)
* [ ] xboxdrv (user-space gamepad driver)

## Severity / Impact

* [x] I've read the docs and the bug reporting instructions
* [x] I've applied the latest firmware update to the controller
* [ ] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [x] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the Bug

Antimicrox has the wrong number of buttons + the guide button isn't working. I had it working a month ago in Manjaro, then I switched to Arch.

## Steps to Reproduce

Install this as per instructions
Build + install antimicrox.
Connect the controller.
Open antimicrox, see 21 buttons & the guide button doesn't work.

## Expected Behavior

Antimicrox has 11 buttons total, including the guide button and excluding the share button, as it did before I switched to Arch a few weeks ago.

## Screenshots / GIFs / Videos

Wireless:
[image]
Wired looks the same except it says "XInput Controller" Instead of "Xbox One Wireless Controller".

## System Information

```console
# uname -a
Linux LUXURIA 5.13.8-arch1-1 #1 SMP PREEMPT Wed, 04 Aug 2021 16:57:44 +0000 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...

```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/6948481/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/6948483/xpadneo-dmesg.txt)

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/6948486/xpadneo-lsusb.txt)

## Additional Context

I was using Manjaro, and had xpadneo + antimicrox installed. Everything worked except the share button (including guide). I switched to Arch, and now I cannot make it work for the life of me. I've looked at other issues, and I can see that share + guide being on a separate device is expected behavior. I remember distinctly that, when I had it working, antimicrox said "Xbox One Wireless Controller" or something to that effect, not XInput Controller. The controller / antimicrox behavior was also consistent both wired and wireless.

Honestly not sure why it worked before, since as far as I can tell now, it shouldn't due to the whole separate devices thing.
I also ran into this problem before, when I set it up on Manjaro, but like an idiot, I didn't write down how to fix it.

Antimicrox seems to use the Generic X-Box pad buttons, since it has the same 21 button bindings regardless of whether it's connected by USB-C or Bluetooth. The only difference is that the guide button only works via USB-C. I
 still have all the files from my Manjaro install if anything from there is needed. I've tried imitating my old setup exactly to no avail.
Honestly not sure if I should post this here or on the antimicrox page.

Thanks.

kakra, 2021-08-07:
I've just launched antimicro to check it out: It seems the software is expecting a specific layout of the button map hard-coded into the source code instead of using what evdev provides as information. It also completely ignores the additional sub-devices of the controller. I can understand why it ignores keyboards and mouse (after all it wants to remap **from** controller **to** keyboard/mouse, not remap keyboards or mice on their own) but maybe it should not ignore consumer control devices which is the type of sub-device which has the share and guide button.

But yes, it's currently by design that the guide and share button are not part of the gamepad sub-device: That is supposed to have exactly only 10 buttons.

But this feature will go under revision when we add or prepare support for Elite 2 profiles in one of the next release cycles. To make this change compatible with previous driver versions, some internals need to be changed which will also avoid some quirks with SDL2 (as integrated in Proton). Hopefully, it may fix antimicro, too. I'll put it on my agenda to test.

The USB-mode works with a completely different driver, it's not using xpadneo then.

RolledR, 2021-08-07:
Ok, that makes sense. I'll close it, and just live without the guide button for the time being. Thanks for the help!

closed 2021-08-07 by RolledR