atar-axis/xpadneo #317: xbox elite series 2 stopped working on steam my steel controller works as expected

bjb1959, 2021-11-09:
## Version of xpadneo

v0.9-88-g2900363!

## Controller Model

Xbox Elite Series 2

### Connection mode

Bluetooth connection

## Installed Software

Steam Input (enabled by default via Steam Desktop client)

## Severity / Impact

It does not work at all
It used to work 

## Describe the Bug
It was working but was a bit glitchy so I tried installing Xow to see if it would work better with the xbox dongle. That is when it broke. even after uninstalling xow and rplacing the xbox dongle with my regular dongle. uninstalling and reinstalling xpadneo, uninstalling and reinstalling steam it still does not work. I am able to connect to the controller on bluetooth and steam sees it but it no longer functions. nothing happens when the buttons are pressed or the sticks are moved.

## Steps to Reproduce
see above description

## Expected Behavior
Want it to work as it did before installing xow

## Screenshots / GIFs / Videos

## System Information
Linux bartb1959-OMEN-by-HP-Desktop-PC-880-p0xx 5.10.0-051000-generic #202012132330 SMP Sun Dec 13 23:33:36 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

```

<!-- Paste the output below the line prepended with # -->
```console
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory

```

## Controller and Bluetooth Information

[  751.936215] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  751.936219] Bluetooth: HIDP socket layer initialized
[  753.054153] hid-generic 0005:045E:0B05.0009: unknown main item tag 0x0
[  753.054318] input: Xbox Elite Wireless Controller as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:9/0005:045E:0B05.0009/input/input36
[  753.054462] input: Xbox Elite Wireless Controller Consumer Control as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:9/0005:045E:0B05.0009/input/input37
[  753.054540] input: Xbox Elite Wireless Controller Keyboard as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:9/0005:045E:0B05.0009/input/input39
[  753.054629] hid-generic 0005:045E:0B05.0009: input,hidraw8: BLUETOOTH HID v9.03 Gamepad [Xbox Elite Wireless Controller] on 2c:6f:c9:1d:25:de
[  753.278376] loaded hid-xpadneo v0.9-88-g2900363
[  753.278397] xpadneo 0005:045E:0B05.0009: pretending XB1S Windows wireless mode (changed PID from 0x0B05 to 0x02E0)
[  753.278399] xpadneo 0005:045E:0B05.0009: report descriptor size: 1225 bytes
[  753.278400] xpadneo 0005:045E:0B05.0009: fixing up report descriptor size
[  753.278401] xpadneo 0005:045E:0B05.0009: fixing up Rx axis
[  753.278401] xpadneo 0005:045E:0B05.0009: fixing up Ry axis
[  753.278402] xpadneo 0005:045E:0B05.0009: fixing up Z axis
[  753.278402] xpadneo 0005:045E:0B05.0009: fixing up Rz axis
[  753.278403] xpadneo 0005:045E:0B05.0009: fixing up button mapping
[  753.278974] xpadneo 0005:045E:0B05.0009: battery detected
[  753.278987] xpadneo 0005:045E:0B05.0009: gamepad detected
[  753.278988] xpadneo 0005:045E:0B05.0009: enabling compliance with Linux Gamepad Specification
[  753.279028] input: Xbox Elite Wireless Controller as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:9/0005:045E:0B05.0009/input/input40
[  753.279107] xpadneo 0005:045E:0B05.0009: consumer control detected
[  753.279129] input: Xbox Elite Wireless Controller Consumer Control as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:9/0005:045E:0B05.0009/input/input41
[  753.279169] xpadneo 0005:045E:0B05.0009: mapping profiles detected
[  753.279171] xpadneo 0005:045E:0B05.0009: keyboard detected
[  753.279193] input: Xbox Elite Wireless Controller Keyboard as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:9/0005:045E:0B05.0009/input/input43
[... 279 lines cut ...]
      iInterface              4 Bluetooth Radio
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0031  1x 49 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0031  1x 49 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered

## Additional Context

kakra, 2021-11-09:
I see you cross-posted here: https://github.com/medusalix/xow/issues/192

xow and xpadneo are completely separate and should not interfere. xow uses the Xbox dongle, xpadneo is made for Bluetooth dongles. xow presents devices on a virtual input bus, xpadneo presents devices on the Bluetooth HID bus, so even drivers or configuration won't interfere. As far as both software components are concerned, they see your controller as completely separate devices.

First things first:

You've probably put the controller back into Bluetooth connect mode as we see it on bus 0x0005:
```dmesg
[ 1036.448789] xpadneo 0005:045E:0B05.000A: input,hidraw8: BLUETOOTH HID v9.03 Gamepad [Xbox Elite Wireless Controller] on 2c:6f:c9:1d:25:de
[ 1036.448791] xpadneo 0005:045E:0B05.000A: controller quirks: 0x00000018
[ 1036.448792] xpadneo xpadneo_welcome_rumble start
[ 1037.431493] xpadneo xpadneo_welcome_rumble took 980ms
```

Did the controller actually rumble on connect?

If it did, the connection was established and the driver can communicate. Run `evtest` next to see if the device is shown (it probably shows three device modes), select the gamepad device, then create some input data to see if it shows up on the screen.

If it did not rumble, remove the controller from your Bluetooth applet, then pair it again.

bjb1959, 2021-11-09:
Yes, it did rumble but I have an update and will close the ticket. I kept messing around and uninstalled xpadneo again and installed the xboxdrv included with Ubuntu 21.10 and rebooted but the controller wouldn't connect using that or the default xpad kernel driver so I uninstalled xboxdrv using synaptic and reinstalled xpadneo and rebooted one last time and now it works. Don't ask me why but there you go.

kakra, 2021-11-09:
There was probably some conflicting udev rule which was cleared by install xboxdrv and uninstalling it again. Or there was some left over from some old xpadneo version that was cleaned up.

bjb1959, 2021-11-09:
I was a bit premature. it isn't working. it works with the usb plugged in but not wirelessly so it is still an issue.

kakra, 2021-11-09:
Please remove the controller from your Bluetooth pairings, then stop the bluetooth service and clear the contents of `/var/lib/bluetooth/`. Reboot. Pair the controller again.

BTW: My XBE2 model isn't working reliably with Bluetooth currently for me, too. So there's probably some issue in bluez.

bjb1959, 2021-11-09:
I will give that a shot

bjb1959, 2021-11-09:
That got the wireless working but now I am back to the original issues that I was hoping using xow and the microsoft dongle would fix. the stick gets stuck when running and won't stop. the character keeps running and the only way to stop it is to turn the controller off and then back on which is a non-starter and more than just an inconvenience. USB mode works perfectly but not wireless mode. Not sure if it is specific to the xbox elite series 2 controllers or not but using wireless is not an option, even with your driver.

kakra, 2021-11-09:
This is the problem I'm seeing, too: The controller stops sending input, sometimes just for brief periods, sometimes forever until you restart it. It may help turning rumble off. We should probably close this as the pairing issue itself was resolved. You already opened another report for the intermittent input loss problems.

closed 2021-12-17 by no-response[bot]