atar-axis/xpadneo #319: Xbox Elite Series 2 get's stuck moving the character and won't stop unless you turn the controller off, then back on.

bjb1959, 2021-11-09:
## Version of xpadneo

v0.9-88-g2900363!

## Controller Model

Xbox Elite Series 2

### Connection mode

Bluetooth connection

## Installed Software

Steam Input (enabled by default via Steam Desktop client)

## Severity / Impact

It does not work at all
It used to work 

## Describe the Bug
Issue with using the left stick and moving/running a character. after a minute or so of playing the character gets stuck running/moving in the direction you last pointed the stick in. So if you are running forward the character keeps running forward, left it runs in a big left circle, right, right circle etc. The only way to stop it is to turn the controller off and start it again but then the same issues happens again until eventually I switch to USB mode.

## Steps to Reproduce
see above description

## Expected Behavior
Want it to work as it did before installing xow

## Screenshots / GIFs / Videos

## System Information
Linux bartb1959-OMEN-by-HP-Desktop-PC-880-p0xx 5.10.0-051000-generic #202012132330 SMP Sun Dec 13 23:33:36 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

```

<!-- Paste the output below the line prepended with # -->
```console
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory

```

## Controller and Bluetooth Information

[  751.936215] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[  751.936219] Bluetooth: HIDP socket layer initialized
[  753.054153] hid-generic 0005:045E:0B05.0009: unknown main item tag 0x0
[  753.054318] input: Xbox Elite Wireless Controller as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:9/0005:045E:0B05.0009/input/input36
[  753.054462] input: Xbox Elite Wireless Controller Consumer Control as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:9/0005:045E:0B05.0009/input/input37
[  753.054540] input: Xbox Elite Wireless Controller Keyboard as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:9/0005:045E:0B05.0009/input/input39
[  753.054629] hid-generic 0005:045E:0B05.0009: input,hidraw8: BLUETOOTH HID v9.03 Gamepad [Xbox Elite Wireless Controller] on 2c:6f:c9:1d:25:de
[  753.278376] loaded hid-xpadneo v0.9-88-g2900363
[  753.278397] xpadneo 0005:045E:0B05.0009: pretending XB1S Windows wireless mode (changed PID from 0x0B05 to 0x02E0)
[  753.278399] xpadneo 0005:045E:0B05.0009: report descriptor size: 1225 bytes
[  753.278400] xpadneo 0005:045E:0B05.0009: fixing up report descriptor size
[  753.278401] xpadneo 0005:045E:0B05.0009: fixing up Rx axis
[  753.278401] xpadneo 0005:045E:0B05.0009: fixing up Ry axis
[  753.278402] xpadneo 0005:045E:0B05.0009: fixing up Z axis
[  753.278402] xpadneo 0005:045E:0B05.0009: fixing up Rz axis
[  753.278403] xpadneo 0005:045E:0B05.0009: fixing up button mapping
[  753.278974] xpadneo 0005:045E:0B05.0009: battery detected
[  753.278987] xpadneo 0005:045E:0B05.0009: gamepad detected
[  753.278988] xpadneo 0005:045E:0B05.0009: enabling compliance with Linux Gamepad Specification
[  753.279028] input: Xbox Elite Wireless Controller as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:9/0005:045E:0B05.0009/input/input40
[  753.279107] xpadneo 0005:045E:0B05.0009: consumer control detected
[  753.279129] input: Xbox Elite Wireless Controller Consumer Control as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:9/0005:045E:0B05.0009/input/input41
[  753.279169] xpadneo 0005:045E:0B05.0009: mapping profiles detected
[  753.279171] xpadneo 0005:045E:0B05.0009: keyboard detected
[  753.279193] input: Xbox Elite Wireless Controller Keyboard as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-10/1-10:1.0/bluetooth/hci0/hci0:9/0005:045E:0B05.0009/input/input43
[... 279 lines cut ...]
      iInterface              4 Bluetooth Radio
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0031  1x 49 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0031  1x 49 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered

## Additional Context

kakra, 2021-11-09:
This problem is tracked here: https://github.com/bluez/bluez/issues/123

It can be caused by radio interference from nearby Bluetooth devices running concurrent scanning requests, and it can also be caused by sending rumble commands to the control. This problem cannot be fixed by xpadneo, it's somewhere in the Bluetooth stack of Linux and doesn't occur on other stacks (like Windows, or Android).

Using a different dongle may work around that problem, e.g. a CSR branded (however, there are a lot of CSR clones out there that may behave oddly). Reverting to an older kernel or older bluez packages may also work around that. The root cause hasn't been reliably identified yet.

bjb1959, 2021-11-09:
Any suggested brand of CSR dongle you recommend? I am ok with usb mode but it would be nice if I could get wireless to work consistantly.

kakra, 2021-11-09:
I never tested the Realtek dongles... But I've been most successful with TP-LINK UB400 dongles which are actually CSR chipset inside. But the nature of the error suggests that this isn't really a one hardware vs another issue - the problem is more likely somewhere in the software stack - and MS may be involved here in how they implemented the controller firmware as I could lately reproduce some Bluetooth quirks with the controllers even on Windows using CSR 4.0 dongles.

kakra, 2021-11-09:
BTW: There's probably a firmware update coming in the next months which upgrades all Xbox Bluetooth controllers to the Bluetooth LE protocol - that's the one used by Xbox Series X/S controllers, and it seems to work much more reliable with latest kernels and bluez versions. So you may just wait until the firmware update is officially available.

If you own a Xbox (or a friend does) and your account is enrolled in the Xbox Insiders program, it may be possible to update the firmware already to an alpha version. xpadneo can support these but the button mappings may be wrong (this is currently being worked on). If you have the chance to upgrade the firmware, be prepared to tinker around, you could help figuring out the needed patches for xpadneo to support the mappings correctly. We have an open pull request for upgrade Xbox One S controllers.

bjb1959, 2021-11-09:
thanks for the heads up. I'll just wait until the firmware rolls out for Ubuntu.

kakra, 2021-11-09:
You won't get the firmware in Linux, you'd need to install it using an Xbox or a Windows machine. It will be applied to the controller, then you can pair it in Linux again.

bjb1959, 2021-11-10:
I have windows 10 and 11 installed on VMware so I can do it through one of those. Thanks.

closed 2022-04-30 by kakra