atar-axis/xpadneo #321: Wrong button mapping for XBox Series X controller (045e:0b13)

KuleRucket, 2021-11-14:
## Version of xpadneo
0.9.1

## Controller Model

* [ ] Xbox One S controller
* [ ] Xbox Elite 2 controller
* [X] Xbox Series X|S controller
* [ ] Other:

Device ID 045e:0b12 via USB and 045e:0b13 via bluetooth

### Connection mode

* [X] Bluetooth connection
* [ ] Xbox Dongle connection (not yet supported)
* [ ] USB cable (not yet supported)

## Installed Software

* [X] Steam Input (enabled by default via Steam Desktop client)
* [ ] Steam Link (usually via Raspberry Pi or other micro computers)
* [ ] xow (alternative driver using the Xbox dongle)
* [ ] Anti-Micro (may affect button mappings)
* [ ] netstick (shares input devices via network similar to Steam Link)
* [ ] xboxdrv (user-space gamepad driver)

## Severity / Impact

* [ ] I've read the docs and the bug reporting instructions
* [X] I've applied the latest firmware update to the controller
* [ ] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [X] I can code and I want to help

## Describe the Bug

When connected via USB, the button mapping is as follows (using jstest):
Axis 0: Left Stick L/R
Axis 1: Left Stick U/D
Axis 2: LT
Axis 3: Right Stick L/R
Axis 4: Right Stick U/D
Axis 5: RT
Axis 6: D-Pad L/R
Axis 7: D-Pad U/D
Button 0: A
Button 1: B
Button 2: X
Button 3: Y
Button 4: LB
Button 5: RB
Button 6: Select
Button 7: Start
Button 8: XBox Button
Button 9: L.Click
Button 10: R.Click

When using xpadneo via bluetooth, everything is the same except for the final three buttons which map as:
Button 8: L.Click
Button 9: R.Click
Button 10: Nothing
This means that the button mapping for the joystick clicks are inconsistent between USB and BT and the X Box button is not detected via neoxpad. Sometimes I use USB and sometimes Bluetooth so I would like them to be consistent.

## Steps to Reproduce

Connect pad via USB, run jstest and record mapping. Disconnect cable and connect via BT, repeat.

## Expected Behavior

I expect the mapping to be the same regardless of the connection method

## Screenshots / GIFs / Videos

Not useful.

## System Information

Arch

```console
# uname -a
Linux monster 5.14.16-arch1-1 #1 SMP PREEMPT Tue, 02 Nov 2021 22:22:59 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0      
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/7533513/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/7533514/xpadneo-dmesg.txt)

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/7533515/xpadneo-lsusb.txt)

## Additional Context

kakra, 2021-11-14:
The Xbox button is not strictly a gamepad button, and historically, games (especially jsdev games) expect the 10 first buttons to only have gaming buttons, and most Xbox compatible gamepads did exactly this layout. xpadneo was developed to fix this situation very early at the driver level before SDL picked up, simply by excluding it from the jsdev map. The problem is that jsdev orders the buttons by key codes, and the Xbox buttons comes at position 9 while it should be in position 11. The only way to fix this, is by excluding it from the mapping. Instead, the button is exposed on a secondary device like the Xbox Elite 2 controllers do it.

You really don't need to look at jstest, it has no importance for games today. No current games uses this dying API. You should look at evtest instead.

This problem may be addressed in a later update, after v0.10 or v0.11.

Closing as duplicate of #286

closed 2021-11-14 by kakra

KuleRucket, 2021-11-14:
Thanks for the info. I also get the same inconsistency between button 8/9/10 using sdl2-jstest