atar-axis/xpadneo #322: Can't build with DKMS 3

lights0123, 2021-11-16:
## Version of xpadneo
5aee029c09002de5a8cf616bc98c2c07953827d9

## Describe the Bug
When running `sudo ./install.sh`, it outputs:
```
Deprecated feature: REMAKE_INITRD
Deprecated feature: REMAKE_INITRD
Deprecated feature: REMAKE_INITRD
* creating dkms.conf
* installing module (using DKMS)
Error! Could not find module source directory.
Directory: /usr/src/hid-xpadneo-v0.9-89-g5aee029 does not exist.
```

This is related to https://github.com/dell/dkms/issues/177. DKMS 3 now recently ships with Arch Linux.

It's possible (and likely) that the linked issue will be resolved by DKMS maintainers, but if not, the install script will require some tweaking.

kakra, 2021-11-17:
I'm not sure why it shows `Deprecated feature: REMAKE_INITRD` because that was removed from xpadneo some time ago. Does it show errors from unrelated packages? Or do you have old versions of xpadneo lying around in the DKMS archive?

Also, I'm pretty sure we call installation with both version number and name. Could you try running `sudo ./install.sh --verbose` to run bash in debug mode? It should show each command executed.

lights0123, 2021-11-17:
`Deprecated feature: REMAKE_INITRD` is likely because `dkms status` prints that out when other DKMS modules use it, so it's not xpadneo's fault.

> Also, I'm pretty sure we call installation with both version number and name. Could you try running `sudo ./install.sh --verbose` to run bash in debug mode? It should show each command executed.

(my username replaced with `$USER`)

```
* verbose mode enabled
+ source lib/installer.sh
++ '[' 0 -ne 0 ']'
+++ git rev-parse --show-toplevel
++ GIT_ROOT=/home/$USER/.src/xpadneo
+++ __version
+++ git describe --tags --dirty
++ VERSION=v0.9-89-g5aee029
+++ type -p dkms
++ DKMS_BIN=/usr/bin/dkms
++ : /usr/bin/dkms
++ INSTALLED=($(get_dkms_versions_installed))
+++ get_dkms_versions_installed
+++ /usr/bin/dkms status
+++ tr -s ':, ' ' '
+++ awk -- '$1 == "hid-xpadneo" { print $2 }'
+++ sort -nu
Deprecated feature: REMAKE_INITRD
Deprecated feature: REMAKE_INITRD
Deprecated feature: REMAKE_INITRD
+ [[ -z '' ]]
+ set -e
+ echo '* creating dkms.conf'
* creating dkms.conf
+ sed 's/"@DO_NOT_CHANGE@"/"v0.9-89-g5aee029"/g'
+ echo '* installing module (using DKMS)'
* installing module (using DKMS)
+ dkms install --verbose hid-xpadneo --force
Error! Arguments <module> and <module-version> are not specified.
Usage: install <module>/<module-version> or
       install -m <module>/<module-version> or
       install -m <module> -v <module-version>
+ cat_dkms_make_log
+ local last_error=1
+ '[' -n --verbose ']'
+ cat /var/lib/dkms/hid-xpadneo/v0.9-89-g5aee029/build/make.log
cat: /var/lib/dkms/hid-xpadneo/v0.9-89-g5aee029/build/make.log: No such file or directory
+ true
+ exit 1
```

Modifying `install.sh` to add the version explicitly:
```
* verbose mode enabled
+ source lib/installer.sh
++ '[' 0 -ne 0 ']'
+++ git rev-parse --show-toplevel
++ GIT_ROOT=/home/$USER/.src/xpadneo
+++ __version
+++ git describe --tags --dirty
++ VERSION=v0.9-89-g5aee029-dirty
+++ type -p dkms
++ DKMS_BIN=/usr/bin/dkms
++ : /usr/bin/dkms
++ INSTALLED=($(get_dkms_versions_installed))
+++ get_dkms_versions_installed
+++ /usr/bin/dkms status
+++ tr -s ':, ' ' '
+++ awk -- '$1 == "hid-xpadneo" { print $2 }'
+++ sort -nu
Deprecated feature: REMAKE_INITRD
Deprecated feature: REMAKE_INITRD
Deprecated feature: REMAKE_INITRD
+ [[ -z '' ]]
+ set -e
+ echo '* creating dkms.conf'
* creating dkms.conf
+ sed 's/"@DO_NOT_CHANGE@"/"v0.9-89-g5aee029-dirty"/g'
+ echo '* installing module (using DKMS)'
* installing module (using DKMS)
+ dkms install --verbose hid-xpadneo-v0.9-89-g5aee029-dirty --force
Error! Arguments <module> and <module-version> are not specified.
Usage: install <module>/<module-version> or
       install -m <module>/<module-version> or
       install -m <module> -v <module-version>
+ cat_dkms_make_log
+ local last_error=1
+ '[' -n --verbose ']'
+ cat /var/lib/dkms/hid-xpadneo/v0.9-89-g5aee029-dirty/build/make.log
cat: /var/lib/dkms/hid-xpadneo/v0.9-89-g5aee029-dirty/build/make.log: No such file or directory
+ true
+ exit 1
```

kakra, 2021-11-17:
This is just dumb... It looks like DKMS is maintained by people who don't know the whole product. Previously, without explicitly mentioning the version, DKMS would auto-install the source to its archive. But if we add the version, it no longer does it but instead complains that it doesn't exist, and I have to manually copy it.

kakra, 2021-11-17:
@lights0123 Please retry with PR #324 and let me know if it works, our CI pipeline doesn't have DKMS 3 yet.

lights0123, 2021-11-17:
PR does work. I would still wait for a response on that linked issue though, it's entirely possible someone made that change completely ignoring what it's supposed to do.

kakra, 2021-11-17:
From dell/dkms#173 it looks like this mode of operation may actually be deprecated and replaced by a (not yet existing) function `ldfolder`. I think we just keep it that way and merge the PR, and I'll add a note to consider it for rewrite later.

closed 2021-11-17 by kakra