atar-axis/xpadneo #326: Xbox Series x Button mapping wrong over bluetooth

Kivarnis, 2021-11-17:
## Version of xpadneo

## Controller Model

* [ ] Xbox One S controller
* [ ] Xbox Elite 2 controller
* [ * ] Xbox Series X|S controller
* [ ] Other:

### Connection mode

* [ * ] Bluetooth connection
* [ ] Xbox Dongle connection (not yet supported)
* [ ] USB cable (not yet supported)

## Installed Software

* [ * ] Steam Input (enabled by default via Steam Desktop client)
* [ ] Steam Link (usually via Raspberry Pi or other micro computers)
* [ * ] xow (alternative driver using the Xbox dongle)
* [ ] Anti-Micro (may affect button mappings)
* [ ] netstick (shares input devices via network similar to Steam Link)
* [ ] xboxdrv (user-space gamepad driver)

## Severity / Impact

* [ ] I've read the docs and the bug reporting instructions
* [ * ] I've applied the latest firmware update to the controller
* [ ] It does not work at all
* [ ] It used to work in a previous version
* [ * ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the Bug

The mappings are incorrect, L3/R3 do not function as intended, Share and the new series x middle button are flipped. This works 100% correctly in xow using the dongle, however, as probably aware, this connect/disconnect loops over the dongle and is extremely frustrating, trying bluetooth instead via xpadneo has given these results, which is also unusable.

## Steps to Reproduce

Pair to bluetooth, experience issue.

## Expected Behavior

L3/R3 and left and middle buttons to be mapped and working correctly.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a

# Linux Trinity 5.15.2-zen1-1-zen #1 ZEN SMP PREEMPT Fri, 12 Nov 2021 19:22:07 +0000 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

# fish: No matches for wildcard “/sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor”. See `help expand`.
xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
             ^

```

## Controller and Bluetooth Information

Didn't fully understand this, still new to Linux.

## Additional Context

Kivarnis, 2021-11-17:
Sorry for formatting I also know very little about that too, lol.

kakra, 2021-11-18:
> The mappings are incorrect, L3/R3 do not function as intended,

By that you mean the thumb stick buttons?

> Share and the new series x middle button are flipped

What's the "new series x middle button"? You mean the Xbox button? Those should not be flipped at all but that's probably caused outside of xpadneo.

> This works 100% correctly in xow using the dongle, however, as probably aware, this connect/disconnect loops over the dongle and is extremely frustrating, trying bluetooth instead via xpadneo has given these results, which is also unusable.

It's possible that SDL still has read access to the hidraw device, this happens on some distributions and I'm not sure why. Could you try the following to sort things out at the easiest level first:

1. Stop Steam, connect the controller
2. Look at `dmesg` to find the hidraw number for the controller
3. Run `chmod a-rwx /dev/hidrawNUMBER`
4. Start Steam, check your inputs
5. Still wrong? Disable Steam Input support for Xbox controllers, try again

Thanks.

Kivarnis, 2021-11-18:
I do yes.

https://media.gamestop.com/i/gamestop/11108954/Microsoft-Xbox-Series-X-Wireless-Controller-Carbon-Black?$pdp$$&fmt=webp  the lower middle button, it's swapped function with the upper left button and that one does absolutely nothing. This all works in xow with the wireless dongle.

I have will try this, I will note however I never had the support enabled to begin with. As a Dual Shock 4 user from Windows, disabling that was habit for use with DS4Windows which was a vastly better service. It was the first thing I did upon booting Steam, before ever using the controller with this install and i do not run my current game through Steam, either.

kakra, 2021-11-18:
The button function swap you describe still look like the current double mapping fix that SDL applied to our hidraw device (although it doesn't need to do that, our hidraw device sends correct mappings already). Thus, disabling access to the hidraw device will force SDL to fall back to normal evdev operation, and it applies no mapping fixups then.

You could run `gamepad-tool`: If your controller shows up as an ID starting with `03` it would indicate that it runs as an emulated USB controller via hidraw - which causes problems here. If you disable hidraw user permissions, the controller will show up with an ID starting with `05` (which means Bluetooth), so then it would read the controller correctly.

Kivarnis, 2021-11-18:
I'm sorry but this is a huge amount of text and visually parsing I simply found nothing that said anything about hidraw anything. Plenty of entries regarding the controller, but not that, as such no ideas what to put. However I do believe you are correct.

kakra, 2021-11-18:
https://github.com/atar-axis/xpadneo/issues/326#issuecomment-972642699 list item 3

Kivarnis, 2021-11-18:
I will repeat what I told you, there is thousands of lines from this output. I would need a week of doing nothing else. So I do not have the hidraw'numberhere' if I understood that correctly, I do not have the number to finish that command. Please advise.

Kivarnis, 2021-11-18:
I copied literally everything to a text editor, and searched "hidraw" and found the ID, testing now.

Kivarnis, 2021-11-18:
This worked, now the question is, this was all done automatically, and without steam ever having support for xbox enabled at any time. Prehaps this interaction should be looked into? As it stands, this seems to be the out of box experience for anyone, and it wasn't like this did nothing. Before installing xpadneo my controller would move the camera indefinitely, installing this, fixed that, but had incorrect mapping of some buttons. So it partially worked, apparently. It is unreasonable to assume most whom need a gamepad will not find themselves in precisely the same situation.

kakra, 2021-11-18:
Well, usually, if you just connected the gamepad, the important information will be in the end of the `dmesg` output (last page usually). So the thousands of lines before do not matter. But sorry, I didn't understand that you were actually referring to the dmesg output when you wrote "huge amount of text", the context wasn't obvious to me.

`dmesg` output is chronologically, so you'd look from button up to find something about the Xbox controller, then look out for hidraw in a nearby context, no need to search the whole file.

You can use `dmesg | less` to get the output in a page, then press `G` (upper case) to get to the end of the file, press `?hidraw` to search backward and find something nearby lines about Xbox controller detection.

> This worked, now the question is, this was all done automatically

We are already shipping with configuration that would revoke permissions from the hidraw device. But for unknown reasons it doesn't work on Arch Linux (and probably others).

Also, this is not a one-time fix: If you re-connect the controller, the hidraw device will be recreated with wrong permissions for you, and the number may even change.

So this was only meant as steps identifying the problem: Your's is actually the hidraw permissions problems we've been seeing lately. There are some udev files which change uaccess session access to devices on Arch Linux which I even don't have on my system, and which are also not triggered by udev. We currently do not know what's going on, we only know how to identify the problem, and how to solve it temporarily.

We've found a work-around in the Discord channel which can acts as a temporary fix that survives reboots and reconnects.

Kivarnis, 2021-11-18:
Well happy to test, as is xow is unsuable for wireless dongle and the series x, so this is my only sustainable option as it sits. But please do understand while I am technically proficient, I am not as savy as some Linux users to this sort of thing.

kakra, 2021-11-18:
We have a very nice Discord community which is happy to help with technical and low-level stuff. I'd like to invite you: https://discord.gg/nCqfKa84KA - It has quite a few people across multiple projects (xpadneo, xow, xone) which usually have a good understanding of the current bugs and issues, how to work around them, or even properly identifying what the real underlying problem is.

This issue tracker is more a technical channel, and support usually blows discussions up until I cannot quite well follow them for development.

For now, I'm closing this as a duplicate of #303 so we can keep issue reports tight and clear for development. Your issue is currently already under active investigation, following/subscribing #303 will keep you informed about updates.

closed 2021-11-18 by kakra