atar-axis/xpadneo #329: ramble dont work with many games throu proton on steam

sysdba1977, 2021-12-02:
## Version of xpadneo

from master
## Controller Model

* [ ] Xbox One S controller
* [ ] Xbox Elite 2 controller
* [ *] Xbox Series X|S controller
* [ ] Other:

### Connection mode

* [ *] Bluetooth connection
* [ ] Xbox Dongle connection (not yet supported)
* [ ] USB cable (not yet supported)

## Installed Software

* [ *] Steam Input (enabled by default via Steam Desktop client)
* [ ] Steam Link (usually via Raspberry Pi or other micro computers)
* [ ] xow (alternative driver using the Xbox dongle)
* [ ] Anti-Micro (may affect button mappings)
* [ ] netstick (shares input devices via network similar to Steam Link)
* [ ] xboxdrv (user-space gamepad driver)

## Severity / Impact

* [* ] I've read the docs and the bug reporting instructions
* [ *] I've applied the latest firmware update to the controller
* [ *] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [* ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the Bug

in manjaro after clean install i have blacklit xpad and install xpadneo from your master git. have installed steam. ramble only wor in forza 4 when i disable steam input in settings. added sdl option from your wiki have no result. 

## Steps to Reproduce

aplly sdl option and disable steam input.

## Expected Behavior

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a

Linux amd-reos 5.14.18-1-MANJARO #1 SMP PREEMPT Fri Nov 12 20:24:59 UTC 2021 x86_64 GNU/Linux

<!-- Paste the output below the line prepended with # -->
```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363

## Controller and Bluetooth Information
<!-- Also follow these steps to create addition information
     about your Bluetooth dongle and connection: -->

<!-- First, disconnect the controller. -->

<!-- Run `sudo btmon | tee xpadneo-btmon.txt` and connect the controller. -->

<!-- Run `dmesg | egrep -i 'hid|input|xpadneo' | tee xpadneo-dmesg.txt`. -->

<!-- Run `lsusb` and pick the device number of your dongle. -->

<!-- Run `lsusb -v -s## | tee xpadneo-lsusb.txt` where `##` is the device
     number picked in the previous step -->

<!-- Attach the resulting files, do not bundle the files into a single
     archive. If some files are too big, gzip them individually. Drag
     and drop the files below. -->

## Additional Co
[gamer.zip](https://github.com/atar-axis/xpadneo/files/7641257/gamer.zip)
ntext
<!-- Add any other context about the problem here. -->
rable not work in steam doom witcher 3 forza 5. i will test in other games and give you info

kakra, 2021-12-02:
Do you have RGB software installed or some software for reprogrammable keyboards? Also, if it works without Steam Input, then just let it disabled. Is there anything you'd need from Steam Input?

btw: xpad and xpadneo do not conflict, you don't need to blacklist/uninstall it, xpad is for wired devices only. But xpadneo may conflict with hid-microsoft / hid-generic, tho we have work-around in place. Nothing to blacklist, if xpadneo shows up in dmesg after connecting the controller.

sysdba1977, 2021-12-02:
It works with xpad driver when i connect my gamepad to usb and only forza 4 works. Witch bluetooth connection no ramble in all steam games i have tested. With dissbled steam input it also not work. Sdl option dont help. Sdl jstest gives ramble on bluetooth.

sysdba1977, 2021-12-02:
I dont need steam input. Had rgb keyboard and mouse

sysdba1977, 2021-12-02:
i found how get it work. problem was in steam linux. i disable steam input for all installed steam games and had to disable steam remote play completly. now rumble work wit witcher3, doom, forza horizon 4. ins forza 5 still dont work but i think its game problem. issue can be close now. i very happy. thank you for your driver)

sysdba1977, 2021-12-02:
all woks fine

closed 2021-12-02 by sysdba1977

kakra, 2021-12-02:
Just FTR, OpenRGB installs udev rules for hidraw which conflict with Steam Input and xpadneo - but a fix for it should be in the next version.