atar-axis/xpadneo #330: [Minor improvement to README] Add instructions for openSUSE

ryanrms, 2021-12-03:
Hi, I tried the instructions for Fedora and they got me most of the way there, but kernel-devel-`uname -r` returned no package found since the output of the uname command was "5.15.5-1-default", I am pretty sure it installed kernel-devel which is just a generic package that points at or is updated to the current kernel.

I think you would be fine with this line:

`sudo zypper install dkms make bluez bluez-tools kernel-devel kernel-source`

but kernel-source (which includes the headers) might only get you halfway there, just to be sure I installed the pattern "devel_kernel" which is installed this way:

`sudo zypper in -t pattern devel_kernel`

All on one line, this would be:

`sudo sh -c 'zypper install -t pattern "devel_kernel" && zypper install dkms make bluez bluez-tools kernel-devel'`

I'll have to test in a VM later today and update the issue.  But it's working great here!

OS: openSUSE Tumbleweed
Kernel: 5.15.5
disable_ERTM: not required

kakra, 2021-12-03:
Yep, we should probably remove that `uname` thing since usually there's an unversioned meta-package available which pulls in the kernel-specific package. Feel free to submit a PR if you found a working solution.

ryanrms, 2021-12-03:
My bad, I was unclear. I think the install instructions work for Fedora, but I am using openSUSE and used the Fedora instructions as a starting point.  The above commands use `zypper`, which is the openSUSE (Tumbleweed, Leap) package manager and don't have anything to do with Fedora except using .rpm files. 

I'm testing it in a vm to see if the first command is enough to build the module, or if you need the full "devel_kernel" pattern.

closed 2021-12-03 by kakra