atar-axis/xpadneo #333: Mouse mode doesn't move the cursor

JonisK, 2022-01-09:
## Version of xpadneo
git master at commit cf392a7c93e05d76b6d3befc5b70e19e894f6823

## Controller Model
* [ ] Xbox One S controller
* [ ] Xbox Elite 2 controller
* [x] Xbox Series S controller
* [ ] Xbox Series X controller
* [ ] Other:

### Connection mode

* [x] Bluetooth connection
* [ ] Xbox Dongle connection (not yet supported)
* [ ] USB cable (not yet supported)

## Installed Software

* [ ] Steam Input (enabled by default via Steam Desktop client)
* [ ] Steam Link (usually via Raspberry Pi or other micro computers)
* [ ] xow (alternative driver using the Xbox dongle)
* [ ] Anti-Micro (may affect button mappings)
* [ ] netstick (shares input devices via network similar to Steam Link)
* [ ] xboxdrv (user-space gamepad driver)
* [ ] AntiMicroX

## Severity / Impact

* [x] I've read the docs and the bug reporting instructions
* [ ] I've applied the latest firmware update to the controller
* [ ] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [x] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the Bug
When activating mouse mode by simultaneously pressing the Xbox and Select button, I cannot move my cursor or click buttons. The dmesg log indicates that mouse mode has been activated though:
``` console
[ 2389.685259] xpadneo 0005:045E:0B13.0007: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x02E0)
[ 2389.685268] xpadneo 0005:045E:0B13.0007: working around wrong SDL2 mappings (changed version from 0x00000501 to 0x00000903)
[ 2389.685272] xpadneo 0005:045E:0B13.0007: report descriptor size: 283 bytes
[ 2389.685275] xpadneo 0005:045E:0B13.0007: fixing up Rx axis
[ 2389.685277] xpadneo 0005:045E:0B13.0007: fixing up Ry axis
[ 2389.685279] xpadneo 0005:045E:0B13.0007: fixing up Z axis
[ 2389.685281] xpadneo 0005:045E:0B13.0007: fixing up Rz axis
[ 2389.685282] xpadneo 0005:045E:0B13.0007: fixing up button mapping
[ 2389.685618] xpadneo 0005:045E:0B13.0007: gamepad detected
[ 2389.685620] xpadneo 0005:045E:0B13.0007: enabling compliance with Linux Gamepad Specification
[ 2389.685684] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0007/input/input23
[ 2389.685957] xpadneo 0005:045E:0B13.0007: input,hidraw3: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 14:f6:d8:41:ba:30
[ 2389.686029] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:0B13.0007/input/input24
[ 2389.686089] xpadneo 0005:045E:0B13.0007: consumer control added
[ 2389.686095] xpadneo 0005:045E:0B13.0007: controller quirks: 0x00000050
[ 2389.686097] xpadneo xpadneo_welcome_rumble start
[ 2390.668646] xpadneo xpadneo_welcome_rumble took 980ms
[ 2390.668654] xpadneo 0005:045E:0B13.0007: Xbox Wireless Controller [44:16:22:70:c4:fc] connected
[ 2405.209875] xpadneo 0005:045E:0B13.0007: mouse mode enabled
```

Mouse mode works though when I re-install from @kakra's branch in pull request #282

As a gamepad, the controller works fine with xpadneo. Thanks a lot btw for this cool piece of software!

## Steps to Reproduce
1. Connect controller
2. Press Xbox+Select
3. Move left stick

## Expected Behavior
While in mouse mode, I expect the left stick to move the mouse cursor and A or LT to emulate a mouse click.

## Screenshots / GIFs / Videos

## System Information
* Ubuntu 21.10 x86 64bit
* Intel AX200 WiFi+BT card

```console
# uname -a
Linux ubuntu-desktop 5.13.0-23-generic #23-Ubuntu SMP Fri Nov 26 11:41:15 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:0B13.0074/report_descriptor: Permission denied
4294967295 0
```

## Controller and Bluetooth Information
[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/7834795/xpadneo-btmon.txt)

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/7834793/xpadneo-lsusb.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/7834794/xpadneo-dmesg.txt)

## Additional Context

kakra, 2022-01-09:
Only the mode switching has been implemented in master yet. The mouse movements itself still live in a branch:
https://github.com/kakra/xpadneo/commits/queue/reimplement-mouse-mode

I'm not yet happy with sensitivity and still need to figure out if we can bring up a on-screen keyboard somehow.

JonisK, 2022-01-10:
Hi, 
Thanks for the quick response! 
I wasn't aware that only the mouse mode but no actual mouse functionality have been merged into main yet. This explains why I need to checkout `kakra:queue/reimplement-mouse-mode` to get cursor movement.

Personally, I will then stick with the branch from the pull request since it is usable enough already to switch between games without getting up from the couch :) 

Could you rephrase the text in the README.md? Currently it advertises the mouse mode as if it was fully functional already:
> Works as a mouse if you're are in couch-mode (press Guide+Select)

Cheers

kakra, 2022-01-10:
> Could you rephrase the text in the README.md?

Good point, that shouldn't have been there yet. Probably happened during cherry-picking.

JonisK, 2022-01-10:
Ok, that solves my issue then I guess. Thanks for the clarification!

kakra, 2022-01-10:
Don't close it on your behalf, otherwise I'm missing that point on my agenda.

JonisK, 2022-01-10:
kk

Bitwolfies, 2022-05-09:
Should split it into planned features and implemented features, since its still not in mainline (didnt work for me), should also have guides for some of this stuff, no idea how to enable the "high precision mode" or the auto trigger rumble.

kakra, 2022-05-09:
"high precision mode" disables dead zones in the Linux input layer so Proton games do not apply a second dead zone but the only dead zone (this is actually something that happens in SDL2 and wine). I've fixed the docs, also temporarily removing the line about mouse mode. Mouse mode is queued in my feature queue and scheduled for merge when or shortly after tagging v0.10.

Automatic trigger rumble rumbles the triggers based on pressure when one of the main motors rumble. This is used because the kernel does not yet support a third and fourth rumble motor (a kernel patch for that is queued).

Bitwolfies, 2022-05-09:
Does the auto rumble interfere with games that actually natively use the rumble triggers? There's not a lot of them but they do exist.

kakra, 2022-05-09:
No, it cannot interfere... If SDL supports it for you, it does so through the hidraw interface. The kernel does not yet provide the proper interfaces. And if it does, I have to adjust xpadneo to actually make use of it. I'm probably going to look into some heuristics to detect whether a game sends trigger rumble or doesn't and automatically adjust how the driver works, so older games can still benefit from our "simulation".