atar-axis/xpadneo #334: xbox one s 9th axis?

redkurn, 2022-01-13:
## Version of xpadneo
0.9.1 git repo and aur

## Controller Model
Xbox One S controller

### Connection mode
Bluetooth connection

## Installed Software
Steam Input (enabled by default via Steam Desktop client)

## Severity / Impact
Questionable, some buttons are mapped incorrectly as well X shouldn't be north?

I've read the docs and the bug reporting instructions
I've applied the latest firmware update to the controller

I probably didn't figure it all out but it's too early to give up

## Describe the Bug
When calibrating with Gamepad in plasma I am asked to calibrate a 9th axis, LT and RT seem to activate this axis, xpadneo not installed it asks for 8, but some buttons do not work.

With xpadneo puNES doesn't recognize the select button, SNES9x does, ePSXe recognizes them all.
So those are likely app related issues.

X and Y mapping appears to be swapped and the 9th axis that doesn't exist (does it?) is activating RT and LT

LT does negative for 9th axis
RT does positive for 9th axis

connection also drops occasionally.

## Steps to Reproduce
install 0.9.1 and go to calibrate

## Expected Behavior
8 axis

## Screenshots / GIFs / Videos
https://imgur.com/a/DKqtWKn

## System Information

```console
# uname -a

Linux Overlord 5.16.0-arch1-1 #1 SMP PREEMPT Mon, 10 Jan 2022 20:11:47 +0000 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

[x@Overlord ~]$ xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81  u........$...%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05  ...%.u..............
000000c8: 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07  ..#...%...u.....%.u.
000000dc: 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75  ........!........%.u
000000f0: 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25  .......%.u......p..%
00000104: 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75  du......Pf..U...&..u
00000118: 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00  .........&..u.....e.
0000012c: 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 05 06 09 20  U..|..&..u......... 
00000140: 85 04 15 00 26 ff 00 75 08 95 01 81 02 c0                    ....&..u......
3560615638 1558
```

## Controller and Bluetooth Information

Bus 007 Device 003: ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/7859503/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/7859504/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/7859505/xpadneo-lsusb.txt)

## Additional Context
I do have a HTC vive box connected to my pc, the Vive wireless system as well.
Not sure if it is conflicting, but can unplug the vive box or disable if someone teaches me how on arch.

I've been trying to work this out for a few days after taking a week off out of frustration, finally got it to connect when the kernel was updated to 5.15.13(5.16.0 according to arch) and I am actually running EndeavourOS, but that probably doesn't really matter.

kakra, 2022-01-13:
> Steam Input (enabled by default via Steam Desktop client)

This may cause some wrong button mappings, usually if you also have RGB control software installed. This is due to Steam Input internally using SDL to query the controller, and SDL uses wrong assumptions about the HID driver. It will also affect software not running through Steam while Steam is running.

> Questionable, some buttons are mapped incorrectly as well X shouldn't be north?

The X button is aliased "north" internally in the Linux kernel but actually, it is not north on Xbox controllers. Games use the symbolic names instead of the cardinal names, so mapping ends up correctly.

> When calibrating with Gamepad in plasma I am asked to calibrate a 9th axis, LT and RT seem to activate this axis, xpadneo not installed it asks for 8, but some buttons do not work.

Xbox controllers do not need calibration, the evdev driver already returns the correct value range and dead zones to the jsdev driver. Also, that old jsdev interface is usually not used by any modern game which makes calibration useless anyways.

> With xpadneo puNES doesn't recognize the select button, SNES9x does, ePSXe recognizes them all, but RT is recognized as 456 when lightly pressed and 453 when fully pressed. So those are likely app related issues.

The RT issue may be linked to the 9th axis.

> X and Y mapping appears to be swapped

Nintendo has X and Y swapped. I'd expect emulators to have a switch for swapping that as the user prefers. If it doesn't, there's a quirk option in xpadneo to pretend Nintendo mapping (swaps X and Y, and actually makes X the north button).

> and the 9th axis that doesn't exist (does it?) is activating RT and LT [...]
> LT does negative for 9th axis RT does positive for 9th axis

The 9th axis was requested by simulator users who wanted to use the triggers for rudder but needed that as a combined axis. Software supporting gamepads should usually ignore that as they expect only 8 well defined axes.

> connection also drops occasionally.

This may be a problem with your Bluetooth dongle. We actually can do nothing about it from xpadneo, xpadneo doesn't talk to Bluetooth, xpadneo talks to a HID device presented to us by Bluetooth.

> I do have a HTC vive box connected to my pc, the Vive wireless system as well. Not sure if it is conflicting, but can unplug the vive box or disable if someone teaches me how on arch.

From the perspective of xpadneo, this makes no difference. But I'm not able to guess the same for other software that you may be using.

> I've been trying to work this out for a few days after taking a week off out of frustration, finally got it to connect when the kernel was updated to 5.15.13(5.16.0 according to arch) and I am actually running EndeavourOS, but that probably doesn't really matter.

The connection problems are usually in bluez itself (due to neighboring devices, or software like kdeconnect which interferes with Bluetooth in a bad way), sometimes within Bluetooth drivers in the kernel. While we still try to provide support for Bluetooth issues, there's nothing that xpadneo as a driver itself could do about it because it doesn't talk to Bluetooth.

Usually, CSR with original chipsets dongles work best (Cambridge Silicon Radio, used by vendors like TP-LINK).

redkurn, 2022-01-13:
Steam wasn’t open, I guess it has something to with having it installed.

I do have openRGB installed and running.

What steps should I take to stop steam or other from messing with the
mapping?

kakra, 2022-01-13:
> I do have openRGB installed and running.

Please uninstall openRGB to see if that resolves your issue... OpenRGB interferes with access permissions to raw HID devices and tries to make the controller a RGB device controlled by openRGB. If that fixes your issue, a work-around would be to override the ACLs provided by openRGB to the controller HID device until the issue is resolved in openRGB.

This problem usually only affects software using SDL2 for input.

redkurn, 2022-01-14:
Found a different solution, installed retroarch and switched to that and playing wired since my controller won't connect again.

I do have the BCM20702A0 (I believe) which I used for hackintosh, saw it on the list and I could try it instead.
BT4.0 vs 5.0, but it will be fine.

closed 2024-02-13 by kakra