atar-axis/xpadneo #335: Permanent rumble on Steam

paulleboulenger76, 2022-02-19:
## Version of xpadneo
v0.9-93-gcf392a7!

## Controller Model

* [ ] Xbox One S controller
* [ ] Xbox Elite 2 controller
* [X] Xbox Series X|S controller
* [ ] Other:

### Connection mode

* [X] Bluetooth connection
* [ ] Xbox Dongle connection (not yet supported)
* [ ] USB cable (not yet supported)

## Installed Software

* [X] Steam Input (enabled by default via Steam Desktop client)
* [ ] Steam Link (usually via Raspberry Pi or other micro computers)
* [ ] xow (alternative driver using the Xbox dongle)
* [ ] Anti-Micro (may affect button mappings)
* [ ] netstick (shares input devices via network similar to Steam Link)
* [ ] xboxdrv (user-space gamepad driver)

## Severity / Impact

* [X] I've read the docs and the bug reporting instructions
* [X] I've applied the latest firmware update to the controller
* [ ] It does not work at all
* [X] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [X] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the Bug

Whenever I start a game in Steam my controller starts rumbling continuously. It only stops if I exit Steam.
The controller works perfectly when using Lutris.
I tried disabling Steam Input but it completely messes with button mappings.

## Steps to Reproduce

## Expected Behavior

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux pop-os 5.15.15-76051515-generic #202201160435~1642693824~21.10~97db1bb SMP Thu Jan 20 17:35:05 U x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363

```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/8102839/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/8102840/xpadneo-dmesg.txt)

## Additional Context

kakra, 2022-02-19:
In the near past, this issue was due to the user having access to hidraw devices in `/dev`. After the controller connected, check dmesg for the hidraw device and try to revoke all permissions from it: `chmod a+rw /dev/hidraw#`. Then try again.

If it fixes it, chances are that an RGB software improperly widens access permissions to each and every hidraw device.

The underlying cause for this is that SDL doesn't read the hid descriptor properly but instead looks at the Xbox controller input packet size to decide the format. Normally, SDL should just not use hidraw devices, it also doesn't properly support some of the firmware fixups needed to not crash the controller on rumble (at least not in some earlier versions).

For Lutris, it probably just works because your system provided or Lutris provided SDL doesn't support hidraw. For Lutris wine, I'm not even sure if it uses SDL at all. Proton wine uses SDL. Steam Input also uses SDL and creates a new virtual device while hiding the original from games.

You may also have an obsolete SDL controller mapping lying around - either in your environment or in the Steam installation folder.

paulleboulenger76, 2022-02-20:
> In the near past, this issue was due to the user having access to hidraw devices in /dev. After the controller connected, check dmesg for the hidraw device and try to revoke all permissions from it: chmod a+rw /dev/hidraw#. Then try again.

Thanks, that fixes the permanent rumble.
Unfortunately my controller doesn't rumble anymore in some games (that used to work before).

> If it fixes it, chances are that an RGB software improperly widens access permissions to each and every hidraw device.

I have openRGB installed, will try uninstalling.

I'll mark the issue as closed.

closed 2022-02-20 by paulleboulenger76

kakra, 2022-02-20:
> Unfortunately my controller doesn't rumble anymore in some games (that used to work before).

Which games? I didn't observe such an issue yet but usually I also disable Steam Input because it caused some input lag in the past.