atar-axis/xpadneo #338: Support for device id 0B22

Fellhahn, 2022-03-11:
**Is your feature request related to a problem? Please describe.**

Xbox Elite Series 2 Controller on latest firmware is not recognised/bound by xpadneo driver

**Describe the solution you'd like**

Controller/device is detected and bound by the xpadneo driver

**Describe alternatives you've considered**

Works over USB still using the Xpad driver

**Additional context**

I have a series 2 elite controller, writing on the back claims model 1797. This was updated to the latest firmware yesterday, by connecting to a windows laptop over USB and updating via the Xbox Accessories app.

Controller Firmware version is now at 5.13.3143.0
Kernel 5.16.11-arch1-2

Relevant output from dmesg indicates this device id is now 0B22:

```
[ 2386.499015] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B22.000E/input/input54
[ 2386.499242] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B22.000E/input/input55
[ 2386.499392] hid-generic 0005:045E:0B22.000E: input,hidraw6: BLUETOOTH HID v5.13 Gamepad [Xbox Wireless Controller] on e8:48:b8:c8:20:00
```

The device is bound by the hid-generic driver and is largely unusable, axis and button mappings are all screwed. As stated earlier though still working fine over USB.

I hadn't used xpadneo prior to today, I'd always connected via usb cable. Recently I purchased a bluetooth adapter (TP-Link UB500), I updated the firmware after reading about various problems with BTLE, which the recent firmware is meant to address I think.

I spent some time mucking with /etc/modprobe.d/xpadneo.conf and udev rules thinking if I could get them to match the id then the device would bind.

I then resorted to trying to manually unbind/bind the driver. Successfully unbinds from hid-generic, but when attempting to bind to xpadneo:

```
[root@server ~]# echo -n "0005:045E:0B22.000E" > /sys/bus/hid/drivers/xpadneo/bind
-bash: echo: write error: No such device
[root@server ~]# 
```

Even though the device is definitely present and shows under /sys/bus/hid/devices/

Reading indicates this error is generic and not necessarily descriptive of the exact issue.

Poking around in ./hid-xpadneo/src/hid-xpadneo.c seems to indicate this device id is not currently supported, which might be why it refuses to bind? Possibly device id changed with latest firmware, sorry I can't confirm what it was prior to update.

kakra, 2022-03-11:
It looks like new firmware emerges as PID `0x0B2*` for each model. Unfortunately, I didn't receive the update yet for XBE2 nor XB1S, only my XBXS controller was updated. It looks like MS still throttles firmware updates when using the Windows Xbox app. How did you update your controllers?

FWIW, a proper fix to add your controller may be to use [this commit](https://github.com/atar-axis/xpadneo/commit/4fd620cd6cb80fb0c1490dc1c864108679d91ab1) as a template. I suspect that your firmware also needs the `XPADNEO_QUIRK_SHARE_BUTTON` despite it having no Share button.

If you'd like to submit a PR, I'd happily consider it for merging.

> Reading indicates this error is generic and not necessarily descriptive of the exact issue.

Well, bind and unbind need the kernel device name which may be different, see here:
* https://github.com/atar-axis/xpadneo/commit/4fd620cd6cb80fb0c1490dc1c864108679d91ab1#diff-87e15635204bea5d9ed0a3d9ca179f8845d9177e157e83a857f108a902d4c7aeR2

Tho, for subsystem hid, it should be exactly what you provided. So probably, yes, the PID is probably missing in the device database in the source code, and adding it should get you one step further.

If possible, please also document the USB PID because xpadneo is supposed to support USB mode in a future version.

julian-barske, 2022-03-15:
I also have a controller with a 0B22 device id.
Thanks, @kakra, I can confirm that it is working using the mentioned commit as a template with 0B22 instead.

Rumble  feels weird and sometimes as if it were "saving up" (after a longish period of rumbling, it just kept going for a few more seconds). This sounds just like #337 

Since you asked for the USB PID:
# lsusb
Bus 001 Device 007: ID 045e:0b00 Microsoft Corp. Controller

closed 2022-04-30 by kakra