atar-axis/xpadneo #339: Xbox Series X Controller: Connected, No Rumble, Not detected

Austin-Marks, 2022-03-13:
After following the the connection process, I get what appears to be a successful connection (light stays solid on controller). However, the controller does not rumble. Furthermore, Steam doesn't detect any controller, and Game Controller does not show any gamepads. Running Fedora 35 KDE spin. 

dmesg -wdH Provides this message upon starting/connecting the controller:
[<    5.015416>] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.00CD/input/input227
[<    0.000388>] hid-generic 0005:045E:0B13.00CD: input,hidraw3: BLUETOOTH HID v5.13 Gamepad [Xbox Wireless Controller] on f4:26:79:69:35:6a
[<    0.002195>] Loading of unsigned module is rejected

Troubleshooting lists possible driver problem if there is no rumble, yet I do not know what that entails.

Austin-Marks, 2022-03-13:
Clarification: Bluetooth connection

kakra, 2022-03-13:
It looks like module signing gets in the way. You either need to sign xpadneo (I don't know how to do that, if you used your distribution package, it should probably do that), or you need to turn signature verification off (this depends on your distribution).

Austin-Marks, 2022-03-15:
As I have little experience with linux, I've been doing a lot of reading about this. I've tried running uninstall.sh, but hid-xpadneo stays around (typing modprobe and checking the autocomplete shows hid-xpadneo as a possibility [but modprobe says hid-xpadneo's key was rejected by service]). In sudo ./install.sh I get "Module hid-xpadneo/v0.9-96-g4fd620c already installed on kernel 5.16.13-200.fc35.x86_64 (x86_64)." 

When I installed Fedora I chose not to create a super user... Maybe that is preventing dkms from adding the xpadneo driver to the kernel correctly? That's all I've got. Best wishes with the xpadneo project, I'd try to contribute, but I'm definitely years of experience away from being able to contribute.

closed 2022-03-24 by Austin-Marks