atar-axis/xpadneo #341: xpadneo: item fetching failed at offset 18/22 | parse failed | failed with error -22

Hisashin7331, 2022-03-24:
## Version of xpadneo
0.9.1

## Controller Model
* [ ] Xbox One S controller
* [ ] Xbox Elite 2 controller
* [x] Xbox Series X|S controller
* [ ] Other:

### Connection mode
* [x] Bluetooth connection
* [ ] Xbox Dongle connection (not yet supported)
* [ ] USB cable (not yet supported)

## Installed Software
* [x] Steam Input (enabled by default via Steam Desktop client)
* [ ] Steam Link (usually via Raspberry Pi or other micro computers)
* [ ] xow (alternative driver using the Xbox dongle)
* [ ] Anti-Micro (may affect button mappings)
* [ ] netstick (shares input devices via network similar to Steam Link)
* [ ] xboxdrv (user-space gamepad driver)

## Severity / Impact
* [x] I've read the docs and the bug reporting instructions
* [ ] I've applied the latest firmware update to the controller
* [x] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the Bug
As in docs controller is connected but did not rumble. So I ran dmesg and it showed my a xpadneo and hid-generic errors

## Steps to Reproduce
1. Install prerequisites
2. Install xpadneo from github
3. connect to the controller
4. there you are

## Expected Behavior
Controller connected properly

## System Information
```console
# uname -a
Linux charlotte 5.15.28-1-MANJARO #1 SMP PREEMPT Fri Mar 11 14:12:57 UTC 2022 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0
```

## Controller and Bluetooth Information
[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/8346286/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/8346287/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/8346288/xpadneo-lsusb.txt)

kakra, 2022-03-25:
This error even occurs in hid-generic:
```
[14273.930768] hid-generic 0005:045E:0B13.0009: item fetching failed at offset 18/22
[14273.930780] hid-generic: probe of 0005:045E:0B13.0009 failed with error -22
```

I suspect the controller isn't paired correctly in Bluetooth, or a Bluetooth module/feature is missing. A descriptor size of 22 bytes seems rather short.

Hisashin7331, 2022-03-25:
Okay so today I was able to connect controller to my friend's windows pc and update its firmware, now I'm able to connect properly without any config.

closed 2022-03-25 by Hisashin7331

kakra, 2022-03-25:
Do you know which firmware version it had before the update?

Hisashin7331, 2022-03-27:
Probably the first this controller ever had, I bought it on release and used it under Windows all the time.

gladish, 2022-06-24:
Same issue for me. Controller is running 5.13.3143.0. I removed the device using bluetoothctl and then re-added it. Works ok now.

* scan on
* pair <mac>
* trust <mac>
* scan off

[wingman]: uname -a
Linux wingman 5.8.0-1032-raspi #35-Ubuntu SMP PREEMPT Wed Jul 14 10:51:21 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

[wingman]: /lib/bluetooth/bluetoothd --version
5.63