atar-axis/xpadneo #344: Xbox Elite 2 shows up as normal Xbox one controller

MNS26, 2022-03-31:
## Version of xpadneo

0.9.1-1 (from AUR)

## Controller Model

* [ ] Xbox One S controller
* [x] Xbox Elite 2 controller
* [ ] Xbox Series X|S controller
* [ ] Other:

### Connection mode

* [x] Bluetooth connection
* [ ] Xbox Dongle connection (not yet supported)
* [ ] USB cable (not yet supported)

## Installed Software

* [x] Steam Input (enabled by default via Steam Desktop client)
* [ ] Steam Link (usually via Raspberry Pi or other micro computers)
* [ ] xow (alternative driver using the Xbox dongle)
* [ ] Anti-Micro (may affect button mappings)
* [ ] netstick (shares input devices via network similar to Steam Link)
* [ ] xboxdrv (user-space gamepad driver)

## Severity / Impact

* [x] I've read the docs and the bug reporting instructions
* [ ] I've applied the latest firmware update to the controller
* [ ] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [x] I probably didn't figure it all out but it's too early to give up
* [x] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [x] I can code and I want to help

## Describe the Bug

My Xbox Elite 2 controller show up as a "normal" Xbox controller when using xpadneo
but when not using it the controller shows as a Elite 2 but has 32 buttons in my system page (see screenshots)

## Steps to Reproduce

not having xpadneo installed (or loaded)

## Expected Behavior

when xpadneo is loaded for the controller to show up as a Elite 2 instead of a Xbox one controller

## Screenshots / GIFs / Videos

     (without kernel module)
[image]
[image]

(with xpadneo installed and loaded)
[image]
[image]

## System Information

```console
# uname -a
Linux ROG 5.16.11-xanmod0-rog-1 #1 SMP Thu, 10 Mar 2022 16:02:46 +0000 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81  u........$...%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 01 09 01 a1 00 09 40 09  ...%.u............@.
000000c8: 41 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 01 a1  A..'......u.........
000000dc: 00 09 43 09 44 15 00 27 ff ff 00 00 95 02 75 10 81 02 c0 05  ..C.D..'......u.....
000000f0: 01 09 42 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06  ..B..&....u.....%.u.
00000104: 95 01 81 03 05 01 09 45 15 00 26 ff 03 95 01 75 0a 81 02 15  .......E..&....u....
00000118: 00 25 00 75 06 95 01 81 03 05 01 09 37 15 01 25 08 35 00 46  .%.u........7..%.5.F
0000012c: 3b 01 66 14 00 75 04 95 01 81 42 75 04 95 01 15 00 25 00 35  ;.f..u....Bu.....%.5
00000140: 00 45 00 65 00 81 03 05 09 19 10 29 1e 15 00 25 01 75 01 95  .E.e.......)...%.u..
00000154: 0f 81 02 15 00 25 00 75 01 95 01 81 03 05 0c 0a 82 00 15 00  .....%.u............
00000168: 25 01 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 05 0c  %...u.....%.u.......
0000017c: 09 01 a1 01 0a 81 00 15 00 26 ff 00 95 01 75 04 81 02 15 00  .........&....u.....
00000190: 25 00 95 01 75 04 81 03 0a 84 00 15 00 26 ff 00 95 01 75 04  %...u........&....u.
000001a4: 81 02 15 00 25 00 95 01 75 04 81 03 0a 85 00 15 00 26 ff 00  ....%...u........&..
000001b8: 95 01 75 08 81 02 0a 99 00 15 00 26 ff 00 95 01 75 04 81 02  ..u........&....u...
000001cc: 15 00 25 00 95 01 75 04 81 03 0a 9e 00 15 00 26 ff 00 95 01  ..%...u........&....
000001e0: 75 08 81 02 0a a1 00 15 00 26 ff 00 95 01 75 08 81 02 0a a2  u........&....u.....
000001f4: 00 15 00 26 ff 00 95 01 75 08 81 02 0a a3 00 15 00 26 ff 00  ...&....u........&..
00000208: 95 01 75 08 81 02 0a a4 00 15 00 26 ff 00 95 01 75 08 81 02  ..u........&....u...
0000021c: 0a b9 00 15 00 26 ff 00 95 01 75 08 81 02 0a ba 00 15 00 26  .....&....u........&
00000230: ff 00 95 01 75 08 81 02 0a bb 00 15 00 26 ff 00 95 01 75 08  ....u........&....u.
00000244: 81 02 0a be 00 15 00 26 ff 00 95 01 75 08 81 02 0a c0 00 15  .......&....u.......
00000258: 00 26 ff 00 95 01 75 08 81 02 0a c1 00 15 00 26 ff 00 95 01  .&....u........&....
0000026c: 75 08 81 02 0a c2 00 15 00 26 ff 00 95 01 75 08 81 02 0a c3  u........&....u.....
00000280: 00 15 00 26 ff 00 95 01 75 08 81 02 0a c4 00 15 00 26 ff 00  ...&....u........&..
00000294: 95 01 75 08 81 02 0a c5 00 15 00 26 ff 00 95 01 75 08 81 02  ..u........&....u...
000002a8: 0a c6 00 15 00 26 ff 00 95 01 75 08 81 02 0a c7 00 15 00 26  .....&....u........&
000002bc: ff 00 95 01 75 08 81 02 0a c8 00 15 00 26 ff 00 95 01 75 08  ....u........&....u.
000002d0: 81 02 c0 05 0c 09 01 85 02 a1 01 05 0c 0a 23 02 15 00 25 01  ..............#...%.
000002e4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
000002f8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
0000030c: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
00000320: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
00000334: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
00000348: 00 75 08 95 01 91 02 c0 05 06 09 20 85 04 15 00 26 ff 00 75  .u......... ....&..u
0000035c: 08 95 01 81 02 06 00 ff 09 01 a1 02 85 06 09 01 15 00 25 64  ..................%d
00000370: 75 08 95 01 b1 02 09 02 15 00 25 64 75 08 95 01 b1 02 09 03  u.........%du.......
00000384: 15 00 26 ff 00 75 08 95 01 b1 02 09 04 26 ff 00 75 08 95 3c  ..&..u.......&..u..<
00000398: b2 02 01 c0 06 00 ff 09 02 a1 02 85 07 09 05 15 00 25 64 75  .................%du
000003ac: 08 95 01 b1 02 09 06 15 00 25 64 75 08 95 01 b1 02 09 07 15  .........%du........
000003c0: 00 25 64 75 08 95 01 b1 02 c0 06 00 ff 09 03 a1 02 85 08 09  .%du................
000003d4: 08 15 00 25 64 75 08 95 01 b1 02 09 09 15 00 25 64 75 08 95  ...%du.........%du..
000003e8: 01 b1 02 09 0a 15 00 26 ff 00 75 08 95 01 b1 02 c0 06 00 ff  .......&..u.........
000003fc: 09 04 a1 01 85 09 09 0b 15 00 25 64 75 08 95 01 b1 02 09 0c  ..........%du.......
00000410: 15 00 25 64 75 08 95 01 b1 02 09 0d 15 00 25 64 75 08 95 01  ..%du.........%du...
00000424: b1 02 09 0e 15 00 26 ff 00 75 08 95 01 b1 02 09 0f 26 ff 00  ......&..u.......&..
00000438: 75 08 95 3c b2 02 01 c0 06 00 ff 09 05 a1 01 85 0a 09 10 15  u..<................
0000044c: 00 27 ff ff ff 7f 75 20 95 01 81 02 09 11 15 00 27 ff ff ff  .'....u ........'...
00000460: 7f 75 20 95 01 81 02 09 12 15 00 26 ff 00 75 08 95 02 81 02  .u ........&..u.....
00000474: 09 13 15 00 26 ff 00 75 08 95 01 81 02 c0 06 00 ff 09 06 a1  ....&..u............
00000488: 02 85 0b 09 14 15 00 25 64 75 08 95 01 b1 02 c0 c0 05 01 09  .......%du..........
0000049c: 06 a1 01 85 05 05 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81  .........)...%.u....
000004b0: 02 95 01 75 08 81 03 95 06 75 08 15 00 25 65 05 07 19 00 29  ...u.....u...%e....)
000004c4: 65 81 00 c0                                                  e...
302767832 5688

```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/8390058/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/8390062/xpadneo-dmesg.txt)

`lsusb`
N/A (bluetooth connection)

`lsusb -v -s## | tee xpadneo-lsusb.txt`
N/A (Bluetooth connection)

## Additional Context

kakra, 2022-03-31:
This is mostly a cosmetic issue. However, we do not really support the paddle buttons yet, thus they won't show up. But they work as default mapped to A,B,X,Y. Paddle support should come during v0.9, cosmetic issues will be fixed in v0.10 or later.

Actually, xpadneo pretends to be a USB-connected Xbox One controller (10 buttons, 6 axes) to be compatible with as many games as possible. So you can think of it as a feature. We are still passing the correct model name but software may work differently and use their own names based on device IDs.

So I'll look into the Steam Input cosmetic issue after v0.10.

MNS26, 2022-03-31:
navigating just with the paddles mapped to the Dpad
https://streamable.com/2byaqw

MNS26, 2022-03-31:
@kakra what also interesting is that the L joystick X axis (going L&R) acts as the simulator yaw axis when not using xpadneo
and the Dpad has the same shenanigans

kakra, 2022-03-31:
xpadneo has one additional axis behind the Xbox standard axes which combines the triggers into a rudder axis. It was asked as a feature some time ago. Do you mean that?

MNS26, 2022-03-31:
yes thats the one 

when i dont have xpadneo the rudder axis becomes the left joystick

kakra, 2022-04-01:
Rumble commands need to be throttled below 100 Hz, otherwise the controller firmware crashes

MNS26, 2022-04-01:
and how can it be throttled?

kakra, 2022-04-01:
xpadneo does it in the driver, SDL seems to do it, too, according to the source code, but from received reports I believe it doesn't work properly. The driver you're using has to implement it, it's not a setting you could adjust.

MNS26, 2022-04-01:
ah ok, in that case i will keep it disabled for now and wait until xpadneo shows the controller as the Elite 2 (have the paddles maped to gear-shifter in most racing games)

MNS26, 2022-04-08:
@kakra i see you've made some commits, is there any way i can help test/ provide extra information?

MNS26, 2022-04-08:
ive also added the usb connection data to the 1st message

kakra, 2022-04-08:
From within the existing git checkout, run `git remote add kakra https://github.com/kakra/xpadneo.git` if you didn't do that before. Then run `git remote update --prune` to update all remote trees. You now see a queue branch for v0.10 to checkout.

MNS26, 2022-04-09:
i have cloned the repo and added the remote branch and ran the install.sh file 

what caught my eye in the journal was the line `changed PID from 0x0B05 to 0x02E0`
[image]

kakra, 2022-04-10:
The line you mentioned is really a way to work around an issue with SDL to not mess up our mapping fixups (because it doesn't look at the descriptor but uses hard-coded assumptions from controllerdb). Switching to `0x02E0` selects an SDL2 controllerdb entry which actually matches our mappings (and the of the original Xbox controller).

Back when we introduced this hack, it was the only way to have consistent mappings in old non-SDL games, SDL1 games, and SDL2 games across various versions. Most of this is gone my now thanks to improvements in SDL to actually load newer system SDL versions dynamically into games even when the games are statically linked with an old SDL version (unless it's just too old).

The plan is to remove the need for this during the next development cycle, see #286.

MNS26, 2022-04-12:
oh an issue i found in the is branch was that the uninstall.sh didnt actually uninstalled the kernel modules (0 installs found) even though it was still there

kakra, 2022-04-30:
Could you retry now that we present the paddles to user-space?

MNS26, 2022-05-24:
@kakra i totally forgot about xpadneo oops i will test it right now

MNS26, 2022-05-24:
for some reason it keeps reconnecting the controller too with a random delay in between disconnects

kakra, 2022-05-25:
> ok so i cant seem to install the github one, it throws

I'm pretty sure this is due to #346 - please try cloning the repo as root, too.

> doesnt seem to work... it still shows up as a `Xbox one controller`

I'm not sure what the problem is here. Where does it show up? In SDL apps (and thus Steam Input)? That's because SDL does not actually use device names from the kernel but uses one from its database. That doesn't mean that your model wouldn't work with all its functions. It's usually a cosmetic issue. It should be shown correctly in `dmesg`.

Usually, our driver should not have an entry in the SDL mapping database at all because it was made to actually match the generic controller mappings and show up as "Generic XInput controller". But people keep submitting controllerdb entries to the SDL project even when there's nothing wrong with the mapping, artificially limiting the number of buttons to the device of the submitter or whatever a driver offered at that time. That's like tilting at windmills, I've decided to go with just one SDL mapping that matches our HID reports.

To get an idea what the kernel sees, run `evtest` instead of looking at some high-level apps with sometimes very creative assumptions. If evtest is fine and shows the correct name and all buttons, then xpadneo is fine. It's up to the higher levels to make something useful from it.

> oh and i had to manually delete the `ko.zst` file because the uninstaller doesnt seem to work

That's probably a packaging issue or when you mix and match installs via AUR and manually checked out from Github. The xpadneo project itself doesn't care at all about properly installing or uninstalling the files, it leaves that exercise to dkms because that's what is made for that task.

MNS26, 2022-05-25:
the branch im currently on is `master` correct?

> I'm pretty sure this is due to https://github.com/atar-axis/xpadneo/issues/346 - please try cloning the repo as root, too.

that was indeed the issue with it not installing

> That's probably a packaging issue or when you mix and match installs via AUR and manually checked out from Github.

After i installed the github one (making sure the AUR one was fully removed) the `uninstall.sh` reported ` found 0 registered instance(s) on your system` while after a reboot the driver was still there.

dmesg output

```
[ 6655.777300] xpadneo 0005:045E:0B05.0013: pretending XB1S Windows wireless mode (changed PID from 0x0B05 to 0x02E0)
[ 6655.777315] xpadneo 0005:045E:0B05.0013: report descriptor size: 1038 bytes
[ 6655.777321] xpadneo 0005:045E:0B05.0013: fixing up report descriptor size
[ 6655.777325] xpadneo 0005:045E:0B05.0013: fixing up Rx axis
[ 6655.777328] xpadneo 0005:045E:0B05.0013: fixing up Ry axis
[ 6655.777331] xpadneo 0005:045E:0B05.0013: fixing up Z axis
[ 6655.777334] xpadneo 0005:045E:0B05.0013: fixing up Rz axis
[ 6655.777337] xpadneo 0005:045E:0B05.0013: fixing up button mapping
[ 6655.778960] xpadneo 0005:045E:0B05.0013: battery detected
[ 6655.779015] xpadneo 0005:045E:0B05.0013: gamepad detected
[ 6655.779018] xpadneo 0005:045E:0B05.0013: enabling compliance with Linux Gamepad Specification
[ 6655.779704] input: Xbox Elite Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:512/0005:045E:0B05.0013/input/input111
[ 6655.780605] xpadneo 0005:045E:0B05.0013: consumer control detected
[ 6655.780738] input: Xbox Elite Wireless Controller Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:512/0005:045E:0B05.0013/input/input112
[ 6655.780865] xpadneo 0005:045E:0B05.0013: mapping profiles detected
[ 6655.780877] xpadneo 0005:045E:0B05.0013: keyboard detected
[ 6655.780944] input: Xbox Elite Wireless Controller Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:512/0005:045E:0B05.0013/input/input114
[ 6655.781233] xpadneo 0005:045E:0B05.0013: input,hidraw6: BLUETOOTH HID v9.03 Gamepad [Xbox Elite Wireless Controller] on 20:79:18:bf:94:96
[ 6655.781245] xpadneo 0005:045E:0B05.0013: controller quirks: 0x00000018
[ 6655.781248] xpadneo xpadneo_welcome_rumble start
[ 6656.768168] xpadneo xpadneo_welcome_rumble took 987ms
[ 6656.768175] xpadneo 0005:045E:0B05.0013: Xbox Elite Wireless Controller [f4:6a:d7:37:61:b9] connected
[ 6656.824913] xpadneo 0005:045E:0B05.0013: battery registered
[ 6657.245156] input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input115

``` 

evtest output
```
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Lid Switch
/dev/input/event1:      Sleep Button
/dev/input/event10:     HDA NVidia HDMI/DP,pcm=3
/dev/input/event11:     MOSART Semi. GXT 161 Gaming Mouse
/dev/input/event12:     MOSART Semi. GXT 161 Gaming Mouse
/dev/input/event13:     MOSART Semi. GXT 161 Gaming Mouse Consumer Control
/dev/input/event14:     MOSART Semi. GXT 161 Gaming Mouse System Control
/dev/input/event15:     MOSART Semi. GXT 161 Gaming Mouse
/dev/input/event16:     Asus WMI hotkeys
/dev/input/event17:     HDA NVidia HDMI/DP,pcm=9
/dev/input/event18:     HDA Intel PCH Headset Mic
/dev/input/event19:     HDA Intel PCH Headphone
/dev/input/event2:      Power Button
/dev/input/event20:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event21:     Asus Keyboard
/dev/input/event22:     Asus Keyboard
/dev/input/event23:     HDA NVidia HDMI/DP,pcm=7
/dev/input/event24:     ELAN1200:00 04F3:307A Mouse
/dev/input/event25:     HDA NVidia HDMI/DP,pcm=8
/dev/input/event256:    HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event257:    HD 450BT (AVRCP)
/dev/input/event258:    Xbox Elite Wireless Controller
/dev/input/event259:    Xbox Elite Wireless Controller Consumer Control
/dev/input/event26:     ELAN1200:00 04F3:307A Touchpad
/dev/input/event260:    Xbox Elite Wireless Controller Keyboard
/dev/input/event261:    Microsoft X-Box 360 pad 0             <---- this is the one only seen by steam
[... 76 lines cut ...]
[ 6937.570842] xpadneo 0005:045E:0B05.0014: report descriptor size: 1038 bytes
[ 6937.570848] xpadneo 0005:045E:0B05.0014: fixing up report descriptor size
[ 6937.570852] xpadneo 0005:045E:0B05.0014: fixing up Rx axis
[ 6937.570855] xpadneo 0005:045E:0B05.0014: fixing up Ry axis
[ 6937.570858] xpadneo 0005:045E:0B05.0014: fixing up Z axis
[ 6937.570863] xpadneo 0005:045E:0B05.0014: fixing up Rz axis
[ 6937.570866] xpadneo 0005:045E:0B05.0014: fixing up button mapping
[ 6937.571829] xpadneo 0005:045E:0B05.0014: battery detected
[ 6937.571863] xpadneo 0005:045E:0B05.0014: gamepad detected
[ 6937.571866] xpadneo 0005:045E:0B05.0014: enabling compliance with Linux Gamepad Specification
[ 6937.571954] input: Xbox Elite Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:512/0005:045E:0B05.0014/input/input129
[ 6937.572249] xpadneo 0005:045E:0B05.0014: consumer control detected
[ 6937.572338] input: Xbox Elite Wireless Controller Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:512/0005:045E:0B05.0014/input/input130
[ 6937.572485] xpadneo 0005:045E:0B05.0014: mapping profiles detected
[ 6937.572495] xpadneo 0005:045E:0B05.0014: keyboard detected
[... cut ...]
[ 6937.572812] xpadneo 0005:045E:0B05.0014: controller quirks: 0x00000018
[ 6937.572817] xpadneo xpadneo_welcome_rumble start
[ 6938.559802] xpadneo xpadneo_welcome_rumble took 987ms
[ 6938.559808] xpadneo 0005:045E:0B05.0014: Xbox Elite Wireless Controller [f4:6a:d7:37:61:b9] connected
[ 6939.465146] xpadneo 0005:045E:0B05.0014: battery registered
[ 6939.828093] input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input133
[ 6970.998721] audit: type=1334 audit(1653477229.200:307): prog-id=50 op=LOAD
```

MNS26, 2022-05-25:
Disable Steam Input is always disabled, ive never touched the permissions (and havent looked into it and or what files)

dkms status shows (before manually removing it)
```
❯ sudo dkms status
[sudo] password for noah: 
hid-xpadneo-v0.9-100/g2b16d6f: installed
nvidia/515.43.04, 5.17.9-arch1-1, x86_64: installed
```
after i removed it from:
```
/var/lib/dkms/
/usr/lib/<kernel>/updates/dkms/  <--- ko.zst was here
/usr/lib/<kernel>/modules.alias
/usr/lib/<kernel>/modules.dep
/usr/lib/<kernel>/modules.order
```
dkms status
```
❯ sudo dkms status
nvidia/515.43.04, 5.17.9-arch1-1, x86_64: installed
░▒▓  │   ~ ▓▒░   
```

kakra, 2022-05-25:
> hid-xpadneo-v0.9-100/g2b16d6f: installed
> nvidia/515.43.04, 5.17.9-arch1-1, x86_64: installed

This is silly... Why does `dkms status` change its output syntax with every version? We had `module: version` and `module, version` already, and now it's `module-version`? Or it is `module/version` but somehow it makes `v0.9-100` part of the module name... I'll look into it.

kakra, 2022-05-25:
I'm not sure how you got into the situation of having `hid-xpadneo-v0.9-100/g...` in the status line, I could not reproduce it, it always installed as `hid-xpadneo/v0.9-100-g...` for me. Did you fiddle around manually with dkms trying to add the module? In any case, I think I fixed the uninstall case, it works for me now. I'll push an update soon.

MNS26, 2022-05-25:
nope i used the github install.sh to install it or via aur but that one also builds its automatically

medusalix, 2022-05-25:
@kakra This is an Arch bug: https://bugs.archlinux.org/task/73194. It happens on every kernel upgrade...

kakra, 2022-05-25:
@medusalix This patch (greedy vs non-greedy) probably fixes nothing because we have hyphens in both the module name AND the version name. That's exactly what is hitting us here: `git describe` makes `v0.9-100-gCOMMITHASH`, so with non-greedy matching we get `gCOMMITHASH` as the version, with greedy matching we'd get `xpadneo-v0.9-100-gCOMMITHASH`  as the version (and `hid` as the name) - both are bad. The hook should probably rather read `PACKAGE_NAME` and `PACKAGE_VERSION` from the dkms.conf file in that directory. The hook is just dumb, there's no sane way to infer name and version from the directory path. It even already matches for that very file, so why not read it in the first place?

What about
```bash
for path in "$source_tree"/*/dkms.conf; do
   ( source $path && echo "${PACKAGE_NAME}/${PACKAGE_VERSION}" )
fi
```

instead? Similar for the other location (except it's a while loop and should continue filtering for dkms.conf).

kakra, 2022-05-25:
BTW, it works here in an arch container updated to latest versions:
```
[kakra@arch-linux xpadneo]$ sudo ./install.sh
* creating dkms.conf
* adding hid-xpadneo-v0.9-109-gf754f6e folder to /usr/src
* installing module (using DKMS)
Creating symlink /var/lib/dkms/hid-xpadneo/v0.9-109-gf754f6e/source -> /usr/src/hid-xpadneo-v0.9-109-gf754f6e
Error! Your kernel headers for kernel 5.15.41-gentoo cannot be found at /usr/lib/modules/5.15.41-gentoo/build or /usr/lib/modules/5.15.41-gentoo/source.
Please install the linux-headers-5.15.41-gentoo package or use the --kernelsourcedir option to tell DKMS where it's located.
[kakra@arch-linux xpadneo]$ dkms status
hid-xpadneo/v0.9-109-gf754f6e: added
```

So the arch bug seems fixed?

(ignore the kernel sources error)

medusalix, 2022-05-26:
The bug is only triggered by the pacman DKMS hooks (e.g. by upgrading the kernel). Normal installation always works fine.

kakra, 2022-05-26:
Ah okay, containers have no kernel so I cannot reproduce that here. Hopefully, the patch gets merged soon (though, the bug report exists since over a year).

kakra, 2022-05-26:
@MNS26 This report has deviated from the original report, I'm inclined to closing it. Are there any issues left?

MNS26, 2022-05-28:
other thant it refusing to detect as the correct controller on ~~steam~~ my system (steam not running) no not that i know of

MNS26, 2022-05-28:
and the uninstall works now (yay)
[image]

kakra, 2022-05-28:
> other thant it refusing to detect as the correct controller

This is purely a cosmetic issue currently, fixing #286 may fix this in the future. It should have no function consequences because xpadeo itself is detecting your controller correctly.

closed 2022-05-28 by kakra