atar-axis/xpadneo #346: Can't install via DKMS

jaimetavares-code, 2022-04-13:
## Severity / Impact

* [ ] I've read the docs and the bug reporting instructions
* [ ] I've applied the latest firmware update to the controller
* [x] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the Bug

After following the prerequisites and installation commands, I cloned the repository and attempted to run the install command `sudo ./install.sh`, only to see the following output: `sed: can't read lib/../VERSION: No such file or directory`

## Steps to Reproduce

Clone repository;
Access xpadneo folder;
Open terminal in folder;
Type: `sudo ./install.sh` and press enter

## Expected Behavior

xpadneo to install.

## System Information

Pop!_OS 21.10

```console
# uname -a
Linux pop-os 5.16.15-76051615-generic #202203161444~1647964027~21.10~e706226 SMP PREEMPT Tue Mar 22 17 x86_64 x86_64 x86_64 GNU/Linux

```

## Additional Context

Attempted to install xpadneo, today, on Linux Mint 20.3 as well, also had a sed error.

kakra, 2022-04-13:
Which download link did you use?

readtedium, 2022-04-14:
Ditto, used the default download link in the readme.

tescoshoppah, 2022-04-14:
getting this same issue, this is the output when using the verbose option

* verbose mode enabled
+ source lib/installer.sh
++ '[' 0 -ne 0 ']'
+++ git rev-parse --show-toplevel
+++ true
++ GIT_ROOT=
+++ __version
+++ git describe --tags --dirty
++++ dirname lib/installer.sh
+++ sed '-res/v?(.*)/v\1/' lib/../VERSION
sed: can't read lib/../VERSION: No such file or directory
++ VERSION=

Yasand123, 2022-04-15:
`sed: can't read lib/../VERSION: No such file or directory`
Started happening just today after upgrading the system. Using ArchLinux 5.17.3-arch1-1.

List of the upgraded packages (not sure if this is helpful):
```
alsa-card-profiles (1:0.3.49-1 -> 1:0.3.50-1)
amd-ucode (20220309.cd01f85-1 -> 20220411.705f19a-1)
binutils (2.38-3 -> 2.38-4)
cmake (3.23.0-1 -> 3.23.1-1)
vmaf (2.3.0-1 -> 2.3.1-1)
pango (1:1.50.6-1 -> 1:1.50.7-1)
libusb (1.0.25-3 -> 1.0.26-1)
git (2.35.2-1 -> 2.35.3-1)
openssh (8.9p1-1 -> 9.0p1-1)
gvfs (1.50.0-1 -> 1.50.0+r5+g747c7f6e-1)
lib32-pango (1:1.50.6-1 -> 1:1.50.7-1)
libgit2 (1:1.4.2-1 -> 1:1.4.3-1)
linux (5.17.1.arch1-1 -> 5.17.3.arch1-1)
linux-firmware-whence (20220309.cd01f85-1 -> 20220411.705f19a-1)
linux-firmware (20220309.cd01f85-1 -> 20220411.705f19a-1)
linux-headers (5.17.1.arch1-1 -> 5.17.3.arch1-1)
python-setuptools (1:59.4.0-1 -> 1:59.5.0-1)
meson (0.62.0-1 -> 0.62.0-2)
nvidia (510.60.02-11 -> 510.60.02-15)
pipewire (1:0.3.49-1 -> 1:0.3.50-1)
semver (7.3.6-1 -> 7.3.7-1)
xapian-core (1:1.4.19-1 -> 1:1.4.19-2)
```

EwokExile, 2022-04-15:
I had the same thing happen on a recent kernal update on Manjaro unstable, switching to the AUR version allowed me to get things running again for now

medusalix, 2022-04-15:
@kakra This issue is probably caused by the recent Git vulnerability (`CVE-2022-24765`) that has been fixed in `v2.35.2` (see [this](https://github.blog/2022-04-12-git-security-vulnerability-announced/) blog post).
The installer calls `git describe` which fails with `fatal: unsafe repository ('/home/abc/xpadneo' is owned by someone else)` because the script is running as `root` while the repository is owned by your local user. The fallback to `sed` is somehow failing (which should be investigated) but the root cause is basically the Git vulnerability fix.
I'm hitting the same issue with `xone`.

Yasand123, 2022-04-15:
> @kakra This issue is probably caused by the recent Git vulnerability (`CVE-2022-24765`) that has been fixed in `v2.35.2` (see [...]

Can confirm, downgrading to git v2.35.1 allowed me to reinstall just fine.

kakra, 2022-04-15:
Ah okay... Gentoo does not provide this version yet because they investigate how this affects usage in Portage. The package is available but masked from installation. I'll try to setup the sandbox to find an idea how to work around this issue.

kakra, 2022-04-15:
> The fallback to `sed` is somehow failing (which should be investigated)

It fails because the `VERSION` file does not exist. Archive downloads from a tagged version will include this file, so in that case `git` will fail and use the `VERSION` file instead. But if installing from a git checkout, there's no such file and it will be derived from the git checkout itself. So this is expected although it could be documented better.

AerosolSP, 2022-04-16:
> It fails because the `VERSION` file does not exist. Archive downloads from a tagged version will include this file, so in that case `git` will fail and use the `VERSION` file instead. But if [...]

I'm an idiot and can't parse this to figure out what the actual fix is. I just installed Zorin OS fresh today this is the first thing I started fiddling with.

the-questjon, 2022-04-16:
I can confirm this same issue on Pop OS 21.10, Zorin 16, Kubuntu 20.04.4

I opened this issue #348, will close and follow here.

Joomsy, 2022-04-16:
> I'm an idiot and can't parse this to figure out what the actual fix is. I just installed Zorin OS fresh today this is the first thing I started fiddling with.

Not a fix, but a workaround is simple enough. Just drop to a root shell, clone the repo somewhere, and run the install script. To overly simplify the problem, it's a permissions issue. The install script has to be ran as root, but it sources files that aren't owned by root, causing it to fail thanks to new security implemented into Git.

Worked just fine for me on Pop 21.10 with xanmod stable:
```
root@wedge /usr/local/src/xpadneo                                                             [16:13:52]
> # ./install.sh                                                                             [±master ✓]
* creating dkms.conf
* adding hid-xpadneo-v0.9-96-g4fd620c folder to /usr/src
* installing module (using DKMS)

Creating symlink /var/lib/dkms/hid-xpadneo/v0.9-96-g4fd620c/source ->
                 /usr/src/hid-xpadneo-v0.9-96-g4fd620c

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.15.34-xanmod1 -C /lib/modules/5.15.34-xanmod1/build M=/var/lib/dkms/hid-xpadneo/v0.9-96-g4fd620c/build/src VERSION=v0.9-96-g4fd620c modules...
cleaning build area...

DKMS: build completed.

hid-xpadneo.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.34-xanmod1/kernel/drivers/hid/

Running the post_install script:
Not disabling ERTM, kernel version doesn't require it...
Installing modalias database...
Installing udev rules...
Reloading udev...

depmod...

DKMS: install completed.
```

SonGoku90, 2022-04-22:
On linux mint cinnamon 20.3 this workaround does not work.

Joomsy, 2022-04-22:
> On linux mint cinnamon 20.3 this workaround does not work.

You did something wrong then. I just set up a Mint 20.3 VM, and was able to install xpadneo without any problems. Use these commands:
```
sudo su -
cd /usr/local/src
git clone https://github.com/atar-axis/xpadneo.git && cd xpadneo
./install.sh
```

AerosolSP, 2022-04-23:
Worked for me. Cheers.

SonGoku90, 2022-04-23:
> You did something wrong then. I just set up a Mint 20.3 VM, and was able to install xpadneo without any problems. Use these commands:
>
> ```
> sudo su -
> cd /usr/local/src
> git clone [...]

It works thanks, I did it wrong in the beginning.

Paulemeister, 2022-05-04:
I just copied the VERSION file from the latest releases zip archive. Now it works, but it should get tracked, because it used to work without this workaround.

kakra, 2022-05-04:
Well, you can create the version file before running install with `make -C hid-xpadneo ../VERSION`, then run the installer. But the version file won't become cleaned up that way when you update the source tree later, so you'd need to delete it after installation. This file will only be tracked in the stable versioned branches, there's no way to properly track that in the development branch.

If you install from the development branch, you should probably add the git directory as trusted so a different user can safely use the git repository: That's actually what happens during install: The installer switches to root, and then tries to generate the version file but now the git repo is not trustable for the root user.

You can probably run `sudo git config --global --add safe.directory $PWD` once from the repo directory to fix that for future installations: The installer should now be able to create an intermediate version file by inspecting the git status. I cannot currently verify this because Gentoo didn't switch to the new Git version for obvious reasons (it's an integral part of the package manager).

@Paulemeister Please do not use or recommend using version files from a different archive as it will mess up our reports here (reported code and version won't match). If you want to have a version file tracked (which we actually DO have), checkout the stable versioned branches. I thus hide your comment. Thanks.

closed 2022-05-25 by kakra