atar-axis/xpadneo #348: sed: can't read lib/../VERSION: No such file or directory

the-questjon, 2022-04-15:
sed: can't read lib/../VERSION: No such file or directory

'sudo ./install.sh' is outputting 'sed: can't read lib/../VERSION: No such file or directory' on Ubuntu based distros

Joomsy, 2022-04-15:
Glad I'm not the only one. I have to uninstall and reinstall this with every kernel update because my controller won't properly connect until I do. Just updated this morning, and got this same error when trying to reinstall this. I'm on Pop 21.10.

Lordfirespeed, 2022-04-15:
I thought I was just being inept! Also on Pop 21.10. Same issue

kakra, 2022-04-15:
I think this is a duplicate of #346. Can you confirm this and close this issue, following up on the other one?

the-questjon, 2022-04-16:
Sorry. Yes this has been opened in issue #346. Will close and follow on there.

closed 2022-04-16 by the-questjon