atar-axis/xpadneo #349: Does not work with Intel AX210 Bluetooth

Kzimir, 2022-04-23:
## Version of xpadneo
I built the last up-to-date git version of xpadneo.

## Controller Model

* [ ] Xbox One S controller
* [ ] Xbox Elite 2 controller
* [X] Xbox Series X|S controller
* [ ] Other:

### Connection mode

* [X ] Bluetooth connection
* [ ] Xbox Dongle connection (not yet supported)
* [ ] USB cable (not yet supported)

## Installed Software

* [ ] Steam Input (enabled by default via Steam Desktop client)
* [ ] Steam Link (usually via Raspberry Pi or other micro computers)
* [ ] xow (alternative driver using the Xbox dongle)
* [ ] Anti-Micro (may affect button mappings)
* [ ] netstick (shares input devices via network similar to Steam Link)
* [ ] xboxdrv (user-space gamepad driver)

## Severity / Impact

* [X] I've read the docs and the bug reporting instructions
* [X] I've applied the latest firmware update to the controller
* [X] It does not work at all
* [ ] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the Bug
Unable to pair and connect the gamepad to Bluetooth. Each time i try to connect the pad, the controller is connected in gnome-bluetooth, rumble but the led flashes constantly. When i run The Witcher 3, the gamepad does not work in-game.

## Steps to Reproduce
Go to Gnome 42 settings -> Bluetooth. Turn on the pad to discover it and press the button above the led to pair.

## System Information

```console
# uname -a
Linux debian-laptop 5.17.4 #4 SMP PREEMPT Sat Apr 23 21:20:22 CEST 2022 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/8548736/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/8548737/xpadneo-dmesg.txt)

Bus 003 Device 002: ID 8087:0032 Intel Corp. AX210 Bluetooth

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/8548738/xpadneo-lsusb.txt)

kakra, 2022-04-23:
This is known and can be reproduced even in Windows.

Kzimir, 2022-04-24:
Yeah, update the firmware on Win10 fixes the issue :)

kakra, 2022-04-24:
Closing and tagged in https://github.com/atar-axis/xpadneo/projects/11

closed 2022-04-24 by kakra