atar-axis/xpadneo #352: Xbox One S controller doesn't connect automatically consistently.

Skullfurious, 2022-05-01:
## Version of xpadneo
hid-xpadneo-v0.9-96-g4fd620c

## Controller Model
Xbox One S Model 1708

* [ ] Xbox One S controller

### Connection mode

* [ ] Bluetooth connection

## Installed Software

* [ ] Steam Input (enabled by default via Steam Desktop client)
* [ ] (I disabled Xbox Configuration Support)

## Severity / Impact

* [ ] I've read the docs and the bug reporting instructions
* [ ] I've applied the latest firmware update to the controller
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up

## Describe the Bug

Essentially after pairing for the first time the controller, when trying to connect, will keep connecting and disconnecting in an inconsistent way. The white light on the controller tends to keep flashing but the Bluetooth Icon keeps appearing and disappearing on the top bar in Gnome.

You can resolve the connection issues by opening up the bluetooth settings window and it will pair after a few seconds generally, but I've had it where it still refused to connect even when the bluetooth settings window was opened.

## Steps to Reproduce

Install fresh version of Pop!_OS 22.04.
Install Steam (No Flatpak).
Disable Xbox Configuration Support if enabled (It wasn't by default for me)
 ```
sudo su - 
git xpadneo.git... 
cd xpadneo 
./install.sh
```
Restart Computer

## Expected Behavior

I would expect it to just connect automatically and not cut in and out while attempting to connect. I noticed that when it connects it does like 3 or 4 vibrations all at the same time.

## Screenshots / GIFs / Videos

https://user-images.githubusercontent.com/34455502/166160787-5e4b274e-abb8-4224-8c09-fb82c66415c3.mp4

## System Information

```console
# uname -a
Linux skull-pop 5.16.19-76051619-generic #202204081339~1649696161~22.04~091f44b SMP PREEMPT Mon Apr 11 17 x86_64 x86_64 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
No such file or directory
4294967295 0
```

## Controller and Bluetooth Information

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/8599575/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/8599576/xpadneo-lsusb.txt)
[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/8599574/xpadneo-btmon.txt)

## Additional Context

kakra, 2022-05-02:
You seem to use a very early version (probably initial release version) of the Xbox One S controller firmware which is known to have these problems. Can you try upgrading the firmware to 5.13 using an Xbox or the Windows Xbox Accessories app?

Skullfurious, 2022-05-03:
Seems to be working a lot better after an update. Thanks a lot! I could have sworn I already updated this controller previously on Windows to get it stop misbehaving in the past but perhaps it was a different model or different controller as I have a few lying around.

Thanks again. I'll be sure to post an update if anything unexpected happens!

Skullfurious, 2022-05-03:
So one thing I've run into is that the controller will occasionally stop sending inputs to the computer but everything still appears to be connected. If you turn the Bluetooth off and try to turn it back on it wont work unless you remove the bluetooth dongle and plug it back in. 

This was about after 2 hours of gaming and a battery swap. It was long after the battery swap however and after I reconnected everything the controller lasted another hour or so before I called it. I figured it could have been a battery issue but that doesn't appear to be the case.

Any ideas on what could be causing this? The controller is now on the latest 5.13 driver.

kakra, 2022-05-03:
Since the controller now connects via BLE (BT low energy, since firmware 5.x), you may need to check the bluez latency settings which could explain what you are seeing. But I suspect the dongle itself may have driver issues.

- https://github.com/atar-axis/xpadneo/blob/master/docs/TROUBLESHOOTING.md#high-latency-or-lost-button-events-with-bluetooth-le

Skullfurious, 2022-05-03:
Apologies if this is obvious but would the controller hanging for 30 seconds to a minute at a time be related to that latency problem or would it be more intermittent? I didn't clarify how long the controller was losing signal, if you can call it that, for.

kakra, 2022-05-03:
I didn't test what the behavior of the current firmware versions is or if they even send correct latency settings to bluez now, but with some older 5.x firmwares, I saw delayed or prolonged rumble with intermittent loss of input events for at least a few seconds until at some point the controller stopped doing anything at except continue to rumble. After 30-60s the BT connection would then eventually time out, alternatively you could restart the connection.

You can try running `btmon` in parallel but it's very noisy and thus usually useless unless you're able to tell the exact part when it happened and cut off the rest.

Skullfurious, 2022-05-03:
Alright, I appreciate that information regardless. I'll try your latency work around if I can't gather any useful data for you.

Here's the dmesg that contains a few of the connection attempts. 

https://pastebin.com/4M2vFvSb

kakra, 2022-05-03:
There's nothing in dmesg that sticks out but at least the Broadcom chipsets are not very widely tested for xpadneo. I personally use CSR 4.0 dongles (from different vendors, e.g. TP-LINK).

Skullfurious, 2022-05-03:
I'll try the CSR 4.0 TP link adapter, perhaps that'll yield different results.

Skullfurious, 2022-05-03:
Got something interesting this time when I opened up the bluetooth settings menu after it started to hang.

https://pastebin.com/KjLrjuFd

Skullfurious, 2022-05-03:
Actually here's another bonus one when I tried to disable and reenable the bluetooth on my system.

https://pastebin.com/eryw4fcH Bit messy with all those random events firing but there are some errors in there.

Skullfurious, 2022-05-03:
/etc/bluetooth/main.conf
https://pastebin.com/6M8Hv3H4

And I didn't install TLP unless it came with Pop!_OS. When I ran TLP in the terminal it said I could install the package but it is not installed currently.

Skullfurious, 2022-05-04:
I was running btmon for about 45 minutes and was waiting for the issue to come up again.

`> ACL Data RX: Handle 64 flags 0x02 dlen 23          #136549 [hci0] 2382.132494`

Is the last input that was logged and btmon just sat there hanging with no inputs being received. The following lines only showed up after I manually went into the settings and as soon as the bluetooth settings window opened those following error logs showed up.

https://pastebin.com/ZjHPgiy3

The odd thing is that the Bluetooth settings menu shows the Xbox controller is still connected, and the light on the controller is still on. My system started to hang as I had the Bluetooth settings menu open and the controller was bugged out and it only began to work again as I unplugged my Bluetooth dongle.

kakra, 2022-05-04:
> The odd thing is that the Bluetooth settings menu shows the Xbox controller is still connected, and the light on the controller is still on.

Yep, that's what I also saw sometimes but it's some time ago. Didn't see it in a while. If you leave it that way for 1-2 minutes, the controller will eventually disconnect.

> My system started to hang as I had the Bluetooth settings menu open and the controller was bugged out and it only began to work again as I unplugged my Bluetooth dongle.

What do you mean by "system started to hang"? If it hangs or freezes parts of the system, the problem is really at some lower level and should be reported to the distribution issue tracker. This looks like an issue between bluez and the dongle, or even the kernel and the dongle.

> Oh and I realize you're going out of your way to help me I sincerely do appreciate it.

Well, the point is: There's really nothing xpadneo could do, xpadneo as a driver has no physical contact points with Bluetooth, although I wish we could set some quirks and settings from within the driver. I can only help to debug and pinpoint the issue, and we are collecting problems and solutions/work-arounds about Bluetooth problems in the projects tab above. I'm also reporting some issues to the bluez tracker or comment on that.

Does the same dongle work fine in Windows with the controller? I've seen dongles showing similar problems in Windows although they were less prominent there but still could be triggered.

There's a newer kernel available: 5.16.20 but I don't think it will change anything for this issue. Your bluez version is already the current version.

Skullfurious, 2022-06-11:
I ended up switching to a TP-LINK dongle and had no issues afterwards maintaining a connection or connecting automatically.

Manufacturer ‏ : ‎ TP-LINK Canada
ASIN ‏ : ‎ B07V1SZCY6
Item model number ‏ : ‎ UB400

For anyone curious. Was using the Plugable one prior to this.

Thanks again for all your help. Noticing now however that the back button on my Xbox One S controller isn't working for any games. But I doubt it's related to dongle. I'll setup another ticket. Thanks again!

kakra, 2022-06-12:
There was a bug mapping some buttons, please retry with the latest version. Also, reloading the updated module may expose another bug, try from a fresh reboot after updating the driver.

closed 2024-04-03 by github-actions[bot]