atar-axis/xpadneo #356: Xbox one Series X share button does nothing

robzombie91, 2022-05-13:
**Is your feature request related to a problem? Please describe.**
None

**Describe the solution you'd like**
Either take a screenshot using appropriate screen grabbing software or be able to be programmed to do something else based on user input

kakra, 2022-05-13:
It's actually sending events when looking at `evtest` so it's up to user-space to make use of this button. You may be able to bind a screenshot program to the button. Later version of xpadneo will improve that by allowing to customize the key which gets sent. If you use Steam, you may want to set it to `F12` then.

There may be an issue with the Share button being in the consumer control device, user-space may not properly look at those. This is already investigated.

robzombie91, 2022-05-13:
Can you link the relevant issue# or pr#?

robzombie91, 2022-05-13:
or is this the first one to be opened?

kakra, 2022-05-13:
The customization stuff will start after v0.10. More relevant issues are linked in the relevant projects.

Skullfurious, 2022-06-11:
Having issues with the share button as well. Doesn't appear to work in any games.

kakra, 2022-06-12:
There's a patch queued which assigns F12 for the Share button (see commit link above). There's currently no support for sending it as a native gamepad button.

closed 2022-09-15 by kakra