atar-axis/xpadneo #357: add option to disable pressure-dependent effect intensity for the triggers

Etaash-mathamsetty, 2022-05-13:
currently, most games don't expect a pressure-dependent effect, so the trigger rumble is way too strong. I want to disable that feature so the trigger rumble actually works like normal.

kakra, 2022-05-14:
You can use the module parameter `rumble_attenuation=0,50` to attenuate the trigger rumble by 50 percent (or some other value, the first value attenuates all motors, the second one attenuates only the triggers and is optional). If you want to disable it completely, use `trigger_rumble_mode=2` which disables trigger rumble motor programming.

Etaash-mathamsetty, 2022-05-15:
that's only a reduction though right? (I haven't tried it)

kakra, 2022-05-15:
Both `rumble_attenuation=0,100` and `trigger_rumble_mode=2` disable it completely. If you use lower values than 100%, the effect is reduced by that percentage value so you could still adjust it to your liking if you want to use it but it is too strong.

Etaash-mathamsetty, 2022-05-15:
> Both `rumble_attenuation=0,100` and `trigger_rumble_mode=2` disable it completely. If you use lower values than 100%, the effect is reduced by that percentage value so you could still adjust it to your liking if you want to use it but it is too strong.
> 
> > so the trigger rumble actually works like normal
> 
> The kernel does not currently support trigger rumble. So games won't be able to use it natively. I think the patches for SDL or wine aren't there yet, too. SDL can bypass the kernel to directly program the triggers, in this case xpadneo isn't involved at all.

oh ok that makes sense

kakra, 2022-05-15:
Yes, xpadneo's trigger rumble effect is just a way to make use of the trigger rumble while games cannot actually use it at all because the interfaces are missing. So I decided we can create a synthetic effect by using the current maximum amount of both main motors, and then apply this to each trigger based on how much you press the trigger. This works quite well for a lot of games to create some more haptic feedback but depending on the game, it may be too strong.

kakra, 2023-06-14:
Closing due to no feedback. Current behavior can be adjusted via module parameters. This feature will be redesigned when trigger rumble supports lands in kernels, and work properly then.

closed 2023-06-14 by kakra