atar-axis/xpadneo #360: Issue with make modules_install

Kzimir, 2022-05-22:
Hi,

It's not an issue with xpadneo itself but with the git repo.
I tried more times and i still have the same issue. 
After to download the git repo, i run : 
- make modules
```git describe --tags --dirty >../VERSION
make -C /home/kazimir/Projets/Kernel/linux INSTALL_MOD_DIR="kernel/drivers/hid" LD=ld.bfd M=/home/kazimir/Projets/xpadneo/hid-xpadneo/src VERSION="v0.9-102-ga279cc4-dirty" modules
make[1] : on entre dans le répertoire « /home/kazimir/Projets/Kernel/linux »
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Debian 11.3.0-1) 11.3.0
  You are using:           gcc (Debian 11.3.0-3) 11.3.0
  CC [M]  /home/kazimir/Projets/xpadneo/hid-xpadneo/src/xpadneo.o
  CC [M]  /home/kazimir/Projets/xpadneo/hid-xpadneo/src/xpadneo/core.o
  CC [M]  /home/kazimir/Projets/xpadneo/hid-xpadneo/src/xpadneo/consumer.o
  LD [M]  /home/kazimir/Projets/xpadneo/hid-xpadneo/src/hid-xpadneo.o
  MODPOST /home/kazimir/Projets/xpadneo/hid-xpadneo/src/Module.symvers
  CC [M]  /home/kazimir/Projets/xpadneo/hid-xpadneo/src/hid-xpadneo.mod.o
  LD [M]  /home/kazimir/Projets/xpadneo/hid-xpadneo/src/hid-xpadneo.ko
make[1] : on quitte le répertoire « /home/kazimir/Projets/Kernel/linux »
rm ../VERSION
```

The issue is with this last command : 
- make modules_install.

```git describe --tags --dirty >../VERSION
Error: fatal: unsafe repository ('/home/kazimir/Projets/xpadneo' is owned by someone else)
To add an exception for this directory, call:

	git config --global --add safe.directory /home/kazimir/Projets/xpadneo
make: *** [Makefile:9 : ../VERSION] Erreur 128
rm ../VERSION
```

Of course the git config command does not solve the issue.

kakra, 2022-05-22:
Checkout the git repository as root, it should work then. This is due to a security update to git. Duplicate of #346

closed 2022-05-22 by kakra

kakra, 2022-05-25:
> Of course the git config command does not solve the issue.

Well, it does, if you run it with the same permissions as you do for the make command. See https://github.com/atar-axis/xpadneo/commit/d6b35a58f51c75bb2bbd7db0b195edfce52ddaeb