atar-axis/xpadneo #362: Left Stick click and Right Stick click, and Guide not registering properly

tmayoff, 2022-05-28:
## Version of xpadneo
0.9.2

## Controller Model
* [ ] Xbox One S controller
* [X] Xbox Elite 2 controller
* [ ] Xbox Series X|S controller
* [ ] Other:

### Connection mode

* [X] Bluetooth connection
* [ ] Xbox Dongle connection (not yet supported)
* [ ] USB cable (not yet supported)

## Installed Software

* [X] Steam Input (enabled by default via Steam Desktop client)
* [ ] Steam Link (usually via Raspberry Pi or other micro computers)
* [ ] xow (alternative driver using the Xbox dongle)
* [X] Anti-Micro (may affect button mappings)
* [ ] netstick (shares input devices via network similar to Steam Link)
* [ ] xboxdrv (user-space gamepad driver)

## Severity / Impact

* [X] I've read the docs and the bug reporting instructions
* [X] I've applied the latest firmware update to the controller
* [ ] It does not work at all
* [X] It used to work in a previous version
* [ ] It mostly works but sometimes it doesn't
* [ ] I found a work-around
* [ ] I probably didn't figure it all out but it's too early to give up
* [ ] I don't know how to ... 
* [ ] It's too complicated
* [ ] Fantastic work but ... 
* [ ] I can code and I want to help

## Describe the Bug
After a reboot the left stick click registers as a right stick click, and the right stick click registers as Guide, and Guide as nothing. Oddly enough it seemed fine immediately after installing, but stopped after a reboot. I noticed it first in game, then later tested with AntiMicroX

## Steps to Reproduce
Install 0.9.2 and reboot

## Expected Behavior
The left stick to register as left stick, right stick as right, and Guide as Guide

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux malcolm 5.17.11-300.fc36.x86_64 #1 SMP PREEMPT Wed May 25 15:04:05 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 a1 01 0a 85 00  ...%.u..............
000000c8: 15 00 26 ff 00 95 01 75 08 81 02 0a 99 00 15 00 26 ff 00 95  ..&....u........&...
000000dc: 01 75 04 81 02 15 00 25 00 95 01 75 04 81 03 0a 81 00 15 00  .u.....%...u........
000000f0: 26 ff 00 95 01 75 04 81 02 15 00 25 00 95 01 75 04 81 03 c0  &....u.....%...u....
00000104: 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02  ...!........%.u.....
00000118: 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04  ..%.u......p..%du...
0000012c: 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02  ...Pf..U...&..u.....
00000140: 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15  ....&..u.....e.U..|.
00000154: 00 26 ff 00 75 08 95 01 91 02 c0 05 01 09 06 a1 01 85 05 05  .&..u...............
00000168: 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81 02 95 01 75 08 81  ...)...%.u.......u..
0000017c: 03 95 06 75 08 15 00 25 65 05 07 19 00 29 65 81 00 c0 05 0c  ...u...%e....)e.....
00000190: 09 01 85 0c a1 01 0a 9e 00 15 00 26 ff 00 95 01 75 08 81 02  ...........&....u...
000001a4: 0a a1 00 15 00 26 ff 00 95 01 75 08 81 02 0a a2 00 15 00 26  .....&....u........&
000001b8: ff 00 95 01 75 08 81 02 0a a3 00 15 00 26 ff 00 95 01 75 08  ....u........&....u.
000001cc: 81 02 c0 c0                                                  ....
1410450731 2192
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/8791768/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/8791772/xpadneo-dmesg.txt)

## Additional Context

kakra, 2022-05-29:
Please disable Steam Input and/or RGB software like OpenRGB

tmayoff, 2022-05-29:
So with steam not opened, and openrazer not running, I get the same behaviour. Also I tried testing with [this](https://gamepad-tester.com/) and it used to work fine, now the controller is detected but none of the inputs are. AntiMicro was used to test the controller, and determine what was happening as nothing else seemed to work

tmayoff, 2022-05-29:
Also I wrote a small SDL2 application to test as well, and Left Stick click is registering as Right Stick Click, Right Stick click is Guide then Guide is registering as Left Stick Click (the difference being that guide is registering as something)

kakra, 2022-05-29:
> Also I wrote a small SDL2 application to test as well, and Left Stick click is registering as Right Stick Click, Right Stick click is Guide then Guide is registering as Left Stick Click (the difference being that guide is registering as something)

Okay, this effect is most likely due to SDL2 controller mappings not matching what v0.9.2 exposes. Please try `gamepad-tool` and post the text console log, there should be no indicator that it found or used a mapping other than built-in.

xpadneo specifically fakes a single model to select an official SDL gamecontrollerdb entry that matches our mappings:
```
[  711.532859] xpadneo 0005:045E:0B22.000F: pretending XB1S Windows wireless mode (changed PID from 0x0B22 to 0x02E0)
[  711.532862] xpadneo 0005:045E:0B22.000F: working around wrong SDL2 mappings (changed version from 0x00000513 to 0x00000903)
```

See also #286, your buttons are rotated by one position. It looks like this is broken since v0.9.2. This patch seems to work around it but I'd rather like to find a better matching controllerdb entry:
```patch
diff --git a/hid-xpadneo/src/xpadneo.h b/hid-xpadneo/src/xpadneo.h
index 3a248b5..e3859db 100644
--- a/hid-xpadneo/src/xpadneo.h
+++ b/hid-xpadneo/src/xpadneo.h
@@ -35,8 +35,8 @@ do {                                                                  \

 /* button aliases */
 #define BTN_PADDLES(b) (BTN_TRIGGER_HAPPY37+(b))
-#define BTN_SHARE      BTN_TRIGGER_HAPPY1
-#define BTN_XBOX       BTN_MODE
+#define BTN_SHARE      BTN_TRIGGER_HAPPY2
+#define BTN_XBOX       BTN_TRIGGER_HAPPY1

 /* module parameter "trigger_rumble_mode" */
 #define PARAM_TRIGGER_RUMBLE_PRESSURE    0
```

because otherwise we remove `BTN_MODE` which some software may expect. We are currently using entry `050000005e040000e002000003090000` which defines the Guide button as button 10 (counting from 0).

Instead, this patch may actually work, could you try:
```patch
diff --git a/hid-xpadneo/src/hid-xpadneo.c b/hid-xpadneo/src/hid-xpadneo.c
index 75d3be9..43ed7ee 100644
--- a/hid-xpadneo/src/hid-xpadneo.c
+++ b/hid-xpadneo/src/hid-xpadneo.c
@@ -1137,8 +1137,8 @@ static int xpadneo_probe(struct hid_device *hdev, const struct hid_device_id *id
        case 0x0B13:
        case 0x0B20:
        case 0x0B22:
-               hdev->product = 0x02E0;
-               hdev->version = 0x00000903;
+               hdev->product = 0x02FD;
+               hdev->version = 0x00001130;
                break;
        }

```

This will then match controllerdb entry `050000005e040000fd02000030110000`.

If it works, I'll make a proper patch for v0.10, and then backport it to v0.9.x.

tmayoff, 2022-05-29:
This is the output from `gamepad-tool`

```
[LOG] SDL2 Gamepad Tool v1.1.2 by General Arcade (compiled with SDL version 2.0.4, DLL version 2.0.22)
[LOG] Website: http://generealarcade.com/gamepadtool/

[LOG] Searching gamepads...
[LOG] Found 1 gamepad(s):
[LOG] "Xbox One Wireless Controller", 050000005e040000e002000003090000 (mapping available)
[LOG] Environment variable "SDL_GAMECONTROLLERCONFIG" is not defined
[LOG] Checking if new mappings available from github: https://github.com/gabomdq/SDL_GameControllerDB
```

I'll try the patches and get back to you

tmayoff, 2022-05-29:
Both patches do in fact work

closed 2022-05-30 by kakra